the-modem-distro/pinephone_modem_sdk #1: MACHINE=mdm9607 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.

VielLosero, 2021-01-12:
-----------------------------------------------
[v@arcadia pinephone_modem_sdk]$ make kernel                                                                                    [5/4359]
mv /home/v/data/olimex/usb-glink/new_kernel_and_bootloader/pinephone_modem_sdk/yocto/build/conf/local.conf /home/v/data/olimex/usb-glink
/new_kernel_and_bootloader/pinephone_modem_sdk/yocto/build/conf/backup.conf                                                             
cp /home/v/data/olimex/usb-glink/new_kernel_and_bootloader/pinephone_modem_sdk/tools/config/poky/rootfs.conf /home/v/data/olimex/usb-gli
nk/new_kernel_and_bootloader/pinephone_modem_sdk/yocto/build/conf/local.conf                                                            
cd /home/v/data/olimex/usb-glink/new_kernel_and_bootloader/pinephone_modem_sdk/yocto && source /home/v/data/olimex/usb-glink/new_kernel_and_bootloader/pinephone_modem_sdk/yocto/oe-init-build-env && \
bitbake -c menuconfig virtual/kernel

### Shell environment set up for builds. ###

You can now run 'bitbake <target>' 

Common targets are:
    core-image-minimal
    core-image-sato
    meta-toolchain
    meta-ide-support

You can also run generated qemu images with a command like 'runqemu qemux86'

Other commonly useful commands are:
 - 'devtool' and 'recipetool' handle common recipe tasks
 - 'bitbake-layers' handles common layer tasks
 - 'oe-pkgdata-util' handles common target package tasks
WARNING: Host distribution "arch" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=mdm9607 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
-----------------------------------------------------

is there a layer missing? maybe I missed something ...
any help is appreciated thanks

Biktorgj, 2021-01-14:
Don't know what exactly went wrong with the init script, but it looks like it didn't add the meta-qcom layer to your settings, can you check if all layers defined in init.sh have been populated in yocto/build/conf/bblayers.conf?

Recently I started moving to yocto release 3.2, if the clone is not recent, you might have a broken tree. If in doubt, do a git pull, completely erase the yocto folder (if you didn't make any changes to the tree, and rerun init.sh to recreate the structure)

VielLosero, 2021-01-15:
U right seems the firewall blocks something. From a fresh init without firewall it works great. 
I need to disable firewall to make too because yocto download lot of files with no common ports 80 or 443.
Sorry for the inconvenience , and thanks a lot.

Biktorgj, 2021-01-25:
Closing the issue since it worked, if anything else fails please let me know

closed 2021-01-25 by Biktorgj