the-modem-distro/pinephone_modem_sdk #2: [Call Audio] No audio in incoming/outgoing calls

PsychoGame, 2021-04-23:
**[Environment]**
Hardware: PinePhone KDE CE with mainboard rev. 1.2b
OS: Manjaro ARM Phosh Beta 8
Software version: Modem Release 0.2.1 [Latest]
Provider used: Dutch cellular provider (Simpel)
Reproducibility rate: 100%

**[Steps to reproduce]**
Initial situation:
Latest Quectel stock modem with Simpel simcard inserted in PinePhone
1. Update the modem to @Biktorgj his released 0.2.1 firmware release following the exact instructions provided in the FLASHING docs.
2. After the fastboot reboot, the modem recovers perfectly and shows my network carrier "Simpel" next to the signal strength indicator.
3. Although the modem is recovered, telephony complains about no voice capable modem available so I've rebooted the complete system.
4. After reboot I'm able to place a call/answer a call using the Calls appliciation in Manjaro.
5. I placed a call to my wifes telephone and also called my PinePhone from my wifes telephone,

**[Expected Result]**
I would expect to hear a dial tone whenever I make a call to somebody.
When the other end answers the call I would expect them to hear me, and for me to hear them.
Whenever someone calls me and the phone rings, I pick up the call.
Now i would expect them to hear me, and for me to hear anything on the other end of the line.

**[Actual Result]**
Whenever i place the call to someone (in my case my wifes phone) I do not hear any dial tone in my earpiece, while my wifes telephone actually rings. Whenever I pick up the call, I'm not able to talk to the other end, and also am not able to hear anything from the other end. The call is initialized properly, as my phone shows the call timer counting, and also my wifes phone shows the call times counting.

Whenever I receive a call (also in my case my wifes phone), the phone rings properly and i'm able to answer this call. Again the call timers start counting on me and my wifes telephone, so I know the call is initialized. Although the call is initialized I'm not able to hear any audio from the other end, and the other end is not able to hear anything from my side.

**[Severity]**
_High/Critical_: Although everything else on the modem seems to work properly and the modem is in good shape, making proper calls is a important function of a modem. This is why I rate the severity high/critical.

If I can provide you with any more information, please let me know what information i can gather for you to triage the problem on this bug.

Biktorgj, 2021-04-23:
Hi,
I will check with manjaro on my side (I've tested postmarketOS, Mobian and UBPorts, but not the rest), but if possible, I'd like to gather a few logs from your modem:
You'll need adb installed, and I'd need you to run, as root, and after completing a call:
1. adb shell dmesg > dmesg.log
2. adb pull /var/log/openqti.log

With that at least we can figure out if modem audio is enabled, and we can check if it's manjaro what's having problems.

A test you can do on your side, is from Phosh, go to Settings -> Sound, and switch Output and input configuration from "Default" or "HiFi" (depending on the distro it might show with different names), to "Phone call". This will make the Pinephone's audio driver switch to the modem path. Then try to make a call and check sound

Let me know how it goes!

PsychoGame, 2021-04-23:
Thank you @Biktorgj for your suggestion.
I've tried manually switching audio profile from "Default" to "Phone call" as you instructed, but unfortunately this still doesn't do the trick in my case.
I'll try to reflash Manjaro Phosh Beta 8 again, see if situation changes on a completely clean build.
I will also try other distributions like pmOS and Mobian, which you have tested successfully as well.
If it is still not properly working I'll try to reflash the modem once again with completely fresh downloaded package (I downloaded the firmware on a rather crappy cellular connection).
When taking the logs, do you prefer "Debug" build, or is normal build also good enough for this?

Biktorgj, 2021-04-24:
Sorry, I had forgotten to update the releases in the readme. One thing fixed!

Two problems remain then, about the dial tone: Does it happen everytime? If the call happens over VoLTE directly I haven't seen that issue, but I have gotten reports of some clicking when establishing the call if it you were in 4G and then the carrier forced the modem to 3G for the call (it's what seems to be happening with you, VoLTE is supported for data but not for calling). Do you hear no tone at all, even if you let the call ring for a minute unanswered?

The incoming call fix is should be an easy one. There's an application called eg25-manager which sets up some parameters for the modem. The default config of that app works perfectly in stock but needs a one line change to make it work with this firmware. This change has already been merged but so far it didn't get to most of the distros.

So, to fix this:
1. As root, from a terminal, go to /etc/eg25-manager
2. Once in that folder, open the config file for your Pinephone version and find the following line:
`    { cmd = "QURCCFG", subcmd = "urcport", expect = "\"usbat\"" },`
Replace `usbat` with `all` so it looks like this:
`    { cmd = "QURCCFG", subcmd = "urcport", expect = "\"all\"" },`
3. Reboot the phone, and incoming calls should work again.

PsychoGame, 2021-04-24:
I'm happy to tell you that it's only one small issue remaining.
The modem firmware is now reliably usable in receiving/making calls.
Only problem which is remaining at the moment is the calling sound while placing a call.
I will pull the requested logs for you, so you can see what is going on in the modem while making a phone call

Also small question just out of interest:
Is there a reason why the urcport has been changed from usbat to all?
In previous release 0.2.0 this was not yet the case was it? As I was able to receive calls with that release without changing the interface.

Biktorgj, 2021-04-24:
About the first part, (and just thinking out loud here) I should probably begin setting audio params as soon as the call is being downgraded from 4G->UMTS. Right now I'm setting it up when it's originating, receiving or establishing, but if the call indication from one of the states comes too late or comes in reverse order it might not be hitting the correct code path until the other end actually answers the call. I patched it already and we'll see if this fixes it on the next release

About the URC port settings, you actually should have needed it all the time, but depending on the pinephone software it could have been reading the ring notification through QMI instead of the AT interface. I never really got to the bottom of it, but 'usbat' only works when using the stock kernel. The USB AT interface is omitted in the kernel to allow the ADSP to use it directly, and while that passthrough works fine with a newer kernel, it looks like some address changes or some "magic code" somewhere in the stock kernel isn't there in mine, and the URCs don't get passed to the correct port if using usbat. Someone mentioned that usbmodem works in this scenario, but setting it to 'all' will allow both stock and custom kernels to work without further tinkering and doesn't create any problems, so it's safe to keep it as a default

Biktorgj, 2021-04-24:
Please, try this build and let me know how it behaves on your use case:

https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.2-testing

PsychoGame, 2021-04-24:
The dialing sound is present now on this latest test build.
Also tested other functions to see if nothing else regressed which it does not luckily, but audio paths switchover looks a bit improved now. It feels somewhat faster overall. 
When placing a call I timed the time necessary for the dialing tone to appear. 
This is approx. 2 to 3 seconds after the phone starts ringing on the other end.

Biktorgj, 2021-04-24:
Perfect! I will include the fix in the next release then. Please tell me if you find anything else :)

closed 2021-04-24 by Biktorgj