the-modem-distro/pinephone_modem_sdk #4: Suitability for other EG25-G modem implementations?

dagbdagb, 2021-05-16:
Hi.

Slightly off-topic issue, hope I can be forgiven. 

I have been looking at options for desktop integrated (mobile) telephony for Linux. Purely "because we can" and because I thought it might be cool.

Hardwarewise, the pinemodem is a Quectel EG25-G modem. But it is my understanding that in the pinemodem case, call audio is routed via i2s, and not via a 'USB Audio Card' as the Quectel application note calls it. Is this correct, or just an issue of misinterpreting the available information?

So, bottom line, I wonder if a USB device which embeds the EG25-G:
1. can be set up for voice calling from a Linux desktop (given the right software and configuration)
2. will work with your SDK out of the box

I am also trying to figure out exactly what Quectel means by stating that VoLTE is optional for their solutions. Would you happen to have any insight here?

Other showstoppers to consider?

Biktorgj, 2021-05-17:
Hi @dagbdagb!
The answers for your questions would be yes and maybe :)
Yes, the modem can be set up to use audio through USB and it will emulate a USB audio capture card. This is configured with an AT command. You probably need to setup your computer to be able to route your mic and speakers in and out of the modem, so you will probably have to tinker with alsa/pulsedriver/pipewire/whatever to make it actually work. When you use audio over USB you loose some functionality though (I think it was GPS but don't take my word for it)

About 2: Yes and no. The firmware will work just fine, but I don't have Audio over USB working at this point. You could use it for data, but audio would go over i2s which is not connected in USB. It's on the list, but hasn't been a priority since the Pinephone doesn't use it...

On the VoLTE thing, not sure but I guess it's because the modem has a couple of preloaded, certified VoLTE profiles for certain carriers like t-mobile, AT&T etc. but doesn't support all of them. I haven't had a problem with LTE in both unmentioned carriers though, so I guess it will depend on how picky is your carrier with it

dagbdagb, 2021-05-17:
Thank you!  I'll be ordering one to tinker with. I found a model with a TTL header, so should be fairly safe to load shady software on it, if/when you run out of more interesting things to do.  :-) . 

The "VoLTE profiles", what are they exactly? Binary DSP configs? Text files? Are they cryptograhically signed? Built as part of the kernel or a userspace thing on the modem? Your SDK also enables VoLTE, I presume. So whatever they are, they can be copied over from the official firmware, right?

And I believe you are right about the "either voice or GPS" thing, I think I read something about this in the "Application Note" pdf.

dagbdagb, 2021-05-18:
From Quectel: "EG25-G, it supports volte."

Is echo cancellation implemented in your SDK? Does it work in both directions? I am not an audio engineer, so this may be a lame question...

Biktorgj, 2021-05-20:
Hi!
VoLTE profiles are stored in the ADSP firmware itself, so there's nothing to be done from modem's userspace to make them work. If your carrier is supported, it'll work just fine (. 
On the echo cancellation, if you mean Qualcomm's voice calibration (which in theory does some noise reduction and gain adjustments) it isn't working in the open firmware. I started looking into it a bit, but it is totally undocumented, needs some binary files that cannot be modified without Qualcomm software for fine tunning and I couldn't find any audible differences between using them or not. For real noise cancelling in the modem you would need dual microphones and a setup that handles it *before* audio gets in the modem. Each side has to do its own noise reduction by the way :)

Biktorgj, 2021-05-30:
Closing it now, anything I can help just let me know :)

closed 2021-05-30 by Biktorgj

dagbdagb, 2021-05-30:
👍