the-modem-distro/pinephone_modem_sdk #5: Custom Mobian kernel

Chief-Detektor, 2021-05-16:
Hi!
Thanks for your effort on this project!
I'd like to try this but I'm using Manjaro on my pinephone.
So my questions is what patches did you apply to the mobian kernel in order to work with this modem firmware?

Thanks in advance :)

Biktorgj, 2021-05-17:
Hi!
The patch is included in the release :)
https://github.com/Biktorgj/pinephone_modem_sdk/releases/download/0.2.2-testing/ec25.diff

The only thing it does is add an alsa control to the EC25 codec driver to be able to force it to a specific sample rate, the rest is done using the alsa profiles that are also zipped in the same release, and the command that needs to be sent to the modem to enable a specific rate. Let me know if you get it running!

Chief-Detektor, 2021-05-20:
Works perfectly fine!
I had to rename ec25.diff to ec25.patch then add it to the PKGBUILD file with corresponding sha256sum for the kernel to build.
After flashing the firmware I could make a call and it works. Veeeery nice Sound quality! :)

I have another issue I hoped to get resolved by this firmware:
I can't send SMS since 2 or 3 months.. I have no clue what causes this? Do you have any Idea?

Thanks in advance!

Chief-Detektor, 2021-05-21:
I can send SMS! Only on 3G but I can deal with that!

closed 2021-06-07 by Biktorgj