the-modem-distro/pinephone_modem_sdk #6: Incoming calls don't ring [mobian]

kkeijzer, 2021-05-22:
I'm trying the last release of this firmware, and it seems to be working well, mostly.

Outgoing calls and data work fine, but incoming calls do not have any effect on the phone. It doesn't ring; it doesn't wake up, and the other side just hears the dial tone until it goes to voicemail.

kkeijzer, 2021-05-22:
Ah, it looks like a typo in the docs (`:` instead of `=`).

But when I run

```
AT+QURCCFG="urcport","all"

OK

AT+QURCCFG=?

+QURCCFG: "urcport",("usbat","usbmodem","uart1","uart2","all")

OK
```

it still doesn't seem to ring.

kkeijzer, 2021-05-22:
After some testing, it does work properly now. I guess the modem was in a confused state before. However, in order to get this to work properly, it has to be set on every boot.

For that, I changed `/usr/share/eg25-manager/pine64,pinephone-1.2.toml `

It has a line

```
{ cmd = "QURCCFG", subcmd = "urcport", expect = "\"usbat\"" },
```

That should be changed to:

```
 { cmd = "QURCCFG", subcmd = "urcport", expect = "\"all\"" },
 ```

I wonder if this also works with the stock modem firmware, because then it should probably be upstreamed.

Biktorgj, 2021-05-24:
Hi!
The QURCCFG change has been upstreamed to eg25-manager repo, but not all distros have picked it up yet. Once changed in the settings the modem needs to be rebooted so the modem can pick it up, maybe that's why it didn't work for you on the first try?
Anyway, these are the settings I'm using in Mobian to have ring in and audio at 48K: https://github.com/Biktorgj/pinephone_modem_sdk/releases/download/0.2.2-testing/config_files.tar.gz
It includes both eg25-manager settings and alsa rules

kkeijzer, 2021-05-24:
I indeed saw that it's upstreamed in `eg25-manager`, but the Mobian package is still at an older version. I have opened a bug in the Mobian GitLab about that.

Note that your `README.md` still states the following:

> * Ring In: Works correctly when setting the modem to report RING to all interfaces. You can do this by sending the following command to the modem:
>   * AT+QURCCFG:"urcport","all"
 
 It should be `AT+QURCCFG="urcport","all"` instead of `AT+QURCCFG:"urcport","all"`, which caused my initial confusion.

closed 2021-05-25 by kkeijzer