the-modem-distro/pinephone_modem_sdk #7: Call volume is relatively low [mobian]

kkeijzer, 2021-05-25:
Compared to the proprietary firmware, the call volume is relatively low. In noisy environments, it's hard to understand the other side, even though the earpiece is at maximum volume.

I'm not sure if the modem firmware itself has some volume controls that could be tweaked, but I was able to improve the situation by setting ALSA's "DAC" output to 100% during calls, which defaults to 40% normally.

In order to automatically set this during calls, one must edit `/usr/share/alsa/ucm2/PinePhone/VoiceCall.conf` and modify the following line:

```
cset "name='DAC Playback Volume' 160"
```

It must be changed to:

```
cset "name='DAC Playback Volume' 100%"
```

Unfortunately, `alsaucm reload` doesn't seem to work, so the phone must be rebooted.

After changing this file, the call volume is a lot more bearable.

Biktorgj, 2021-05-25:
Thank you!
The proprietary firmware applies some gain and noise reduction to the incoming audio which I'm not doing. There's some volume controls I can set on the modem side, as I'm always quiet while I'm working on this audio levels have always been "ok" for me. I wonder if there will be an audible differrence if setting volume levels from the pinephone or from the modem side. Will do some tests when I get some free time

Question: are you using the default sample rate (8khz) or 48k?

kkeijzer, 2021-05-25:
I'm using the default sample rate, and I'm also using Mobian's stock 5.11 kernel.

Biktorgj, 2021-05-26:
Ok, I rechecked the audio driver and it doesn't have any volume regulator when there's a voice call, all the volume levels are for devices that are not present in this modem implementation (headset etc.), so there's no really any way to compensate the volume in the modem itself

I've added sample alsa config files to the repo with your your volume settings. I think I got it a little bit louder without distortion by setting AIF1 DA0 Playback Volume to 90% too, but not sure (AIF1 is where the modem is connected to the Pinephone). I could use another set of ears. If it sounds good for you too I will put that change in the repo too

Thanks for the help!

kkeijzer, 2021-05-26:
I've tested it (AIF DA0 at 90% and DAC at 100%), and it seems to work very well. Even better than the defaults with the proprietary firmware I'd say.

Would there be a way to cleanly upstream this in `alsa-ucm-conf`? Because maybe it would be too loud for the proprietary firmware this way, or not, because one could always just turn the volume down of course. I'm not sure if there is a way for ALSA to differentiate between the modem firmwares (I guess not).

The downside to manually editing stuff in `/usr/share/` is that any package update would overwrite it.

Biktorgj, 2021-05-26:
I tested these values with a user with stock firmware and there doesn't seem to be any audible distortion (I was hoping it would simply be less work for the audio normalizing functions in the modem and it seems to be the case). I also tested with headphones and it was working nicely. I have to find my bluetooth headset and run some more tests but if it's working fine in all scenarios with both types of firmware I will request the changes to be merged with their respective repos (I'll have to go hunting for all distros since I can't seem to find any declared as "master" from which everyone picks the changes like with eg25-manager)

ALSA won't know what kind of firmware is installed in the modem but both modem-power and eg25-manager could know which one is running from the responses to AT commands. I've been looking at it for quite some time to be able to support the 48K sampling rate without having to resort to a custom kernel with a patched module, but it'll be a while before I get some suitable fix that's not embarrasing to publish :)

PsychoGame, 2021-05-26:
I could apply a merge request with Manjaro if you Luke. I'll keep an eye on this issue whenever you'd like to upstream these changes.

Biktorgj, 2021-05-27:
Thanks @PsychoGame! I asked to the other devs, and @bshah was already thinking of fixing the decentralized state of these files. He has created a new repo at pine64's Gitlab so all distros can use the same set of profiles.
I don't know if it's too soon to use this repo when building a distro image, but here's the repo if you want to keep an eye on it: https://gitlab.com/pine64-org/pine64-alsa-ucm/

PsychoGame, 2021-05-27:
Alright I'll stir up the discussion with Phil an Dan from Manjaro Phosh and Plasma edition. I'll do some tests myself at Manjaro as well.

raphj, 2021-07-29:
These settings seem to work quite well on Mobian, with the default modem-manager version, on the Pinephone Manjaro CE.

                cset "name='DAC Playback Volume' 100%"
                cset "name='AIF1 DA0 Playback Volume' 90%"
                cset "name='AIF2 DAC Playback Volume' 160"

setting `AIF2 DAC Playback Volume` to 100% seem to saturate the sound a lot.

kkeijzer, 2021-08-16:
I'm guessing we can't do a lot about this from the modem firmware, so the issue here should probably be closed. Perhaps README.md can be updated with a suggestion to change the ALSA UCM config file?

Silent-Hunter, 2021-08-29:
I'm curious, if the proprietary firmware can do this, why can't yours? Is it some secret thing that they just haven't documented? Speakerphone also is useless, as my friend hears himself louder than he hears me.

Silent-Hunter, 2021-08-31:
I can't seem to get this to work on PostmarketOS, no matter what I change these volume values to, the output is the same.

Biktorgj, 2021-09-08:
> I'm curious, if the proprietary firmware can do this, why can't yours? Is it some secret thing that they just haven't documented? Speakerphone also is useless, as my friend hears himself louder than he hears me.

Sorry for the delay,
The stock firmware uses, apart from the common mixers, some audio calibration files, that, among other things, adjust the DSP gain, does some equalization and set some filters. The files tools and loaders are all closed. I've started trying to reimplement the loader but been busy fixing stability issues that had more priority. If I'm a little more available I'll get back in this next

Silent-Hunter, 2021-09-12:
Thanks! It's the only major issue with your firmware, that I find. If you end up fixing this, it'll basically be perfect.

sc0w, 2022-01-05:
Hi all, here with pinephone mobian bookworm

With 90% 90%, into a call, the other person (android phone) hear its own voice (echo)

With 80% 80% it seems to work fine

@Biktorgj 
> I've started trying to reimplement the loader but been busy fixing stability issues that had more priority. 

Thanks, can you add it to the TODO list into readme file?

btw, what does exactly `AIF2 DAC Playback Volume` and `AIF2 ADC Capture Volume` in `VoiceCall.conf` file? What are the differences?

CodePhase, 2022-03-02:
Is there any advice for the Pinephone Pro?  I'm looking at the /usr/share/alsa/ucm2/PinePhonePro/VoiceCall.conf file and the settings in this thread are not present at all (mobian).

Biktorgj, 2022-03-03:
I haven't yet started playing around with the UCM profiles in the Pro, but some people reported better audio by lowering the default microphone boost settings.
Look for the parameter `IN1 Boost`. It's set to `8` by default, maybe setting it somewhere between 0 and 3 is good enough :)

About the static, I'm investigating it, it seems it occurs about 1/4 of the times the modem starts audio, but havent' found what exactly causes it yet

If you find a good combination please let me know or do a PR ;)

CodePhase, 2022-03-07:
I tried setting the `IN1 Boost` to 0 and it didn't have any effect.  I guess I can try other ADP versions but I have a question about those.  Can I apply / change that after I install this custom firmware or do I need to revert back to the OEM firmware first?  Is there any way to check the ADP version installed without re-flashing to the OEM firmware?

CodePhase, 2022-03-12:
Just as an FYI, if I go into phosh's settings app and turn the input volume as low as it will go without it going to mute, the audio on the receiver side of a phone call works fine.  I use this as a workaround.

noneucat, 2022-03-18:
> Calls are hit at miss on the Pro with this firmware (0.5.9). The speaker on the phone _sometimes_ has static. The receiver on the other phone of a call also has poor volume or there is a bad echo/distortion.

Also experiencing this; will try lowering the boost as suggested.

IntinteDAO, 2022-03-21:
Just wondering, isn't there a universal config file that would work everywhere?

3hhh, 2022-05-28:
This one worked for me on the PP with mobian bookworm: https://gitlab.com/postmarketOS/pmaports/-/blob/e4366d5f8ad87786f64904e90f8f6b186d0d854a/device/community/device-pine64-pinephone/ucm/VoiceCall.conf

G-P-L, 2022-06-26:
I could solve this issue by modifing the `/usr/share/alsa/ucm2/PinePhone/VoiceCall.conf` file as suggested, but this caused another annoying issue. If I set the values from the default "160" to a percentage one, then I cannot control the audio volume during the call with the volume buttons on the side.

Biktorgj, 2022-07-01:
The RAW values go from 0 to 255, so I `204` instead of `80%` should give you the same result. Does it allow you to change volume that way?

G-P-L, 2022-07-02:
That was the first thing what I tried and things only get worse. But I could solve this issue in the meantime simply by switching from PulseAudio to PipeWire (what I wanted to do anyway). :)

kkeijzer, 2022-07-02:
@G-P-L How well does PipeWire work on Mobian, with output switching (and back) for calls and all? I've been thinking about switching as well, but I wasn't sure about the phone-specific parts being tested on anything other than Pulse.

G-P-L, 2022-07-03:
It works quite well for me on Arch Linux ARM, but I did not test it on other distros.

barbastrejo7, 2022-07-09:
Better on arch than on mobian definitely. I tested both

G-P-L, 2022-07-10:
In the long run it turned out also on Arch it is not working properly all the time. The switching between the profiles and outputs is a mess and I think it is connected to this PipeWire issue: https://gitlab.freedesktop.org/pipewire/pipewire/-/issues/1553#note_1209281

So I switched back to Pulse for now and still experimenting with the `VoiceCall.conf` settings to find an optimal solution.

via, 2022-08-07:
On the PinephonePro with mobian I've had an issue for a while where the mic gain was clearly far too high to the point where nobody can understand me.  The suggested settings for IN1 Boost didn't appear to fix it, but I had an opportunity to experiment with alsa controls while in a call this morning:
For me, the IN1 and IN2 boost levels make no difference, however the "ADC" gain level was at 100 (30dB) gain.  Setting this to 0 dB (value 19 for me) appears to completely resolve the issue. So I have:
```                cset "name='ADC' 19"
```
In VoiceCall.conf's enable sequence.

CodePhase, 2022-08-07:
Looking at this file myself, I don't see a line that reads exactly
`             cset "name='ADC' 19"`
I have two entries, the first of which I'm wondering is the setting that would be the closest match:
`cset "name='ADC Capture Volume' 47"`
`cset "name='Mono ADC Capture Volume' 47"`

I wonder if my setting of 47 for these is not 100 since I have my input volume set to as close to 0 as I could get it in gnome settings

gorbehnare, 2022-08-08:
Changing these values does nothing on my device at all: 
cset "name='ADC Capture Volume' 47"
cset "name='Mono ADC Capture Volume' 47"

I changed them to 10 and 100, and no difference at all. What do these do?

via, 2022-08-08:
This file does not have a completely exhaustive list of mixer controls -- try adding it in the config. I determined the name by experimenting with alsamixer during a call.

CodePhase, 2022-08-15:
I gave it a shot and added the same setting to the enable sequence as @via suggested.  For me, running on a Pinephone Pro with Mobian Phosh (Bookworm) and pipewire (not pulseaudio), this broke my sound output altogether.  No sound on calls, the sound test in the settings app, or on websites.  I wonder if pipewire needs some other setting.

strungar, 2022-08-30:
i've got headphone muted on ppp. tried several params in the config, no go.
could anyone post `sudo amixer`'s output with it working, please?
or does anyone know the parameter name i'm looking for? :)

fuzzy7k, 2022-11-06:
I'm a little late to the party, but I've put together some proof of concept echo cancellation for the OG PinePhone. 
https://gitlab.com/fuzzy7k/pinephoneucm

It needs integrated into a profile manager to make things smoother. Currently, the sink disappears while it is still being used, which may cause issues in PA.

I've also summarized my findings for volume levels, which vary from device to device.

sado1, 2022-11-27:
(Pinephone Pro, Danctnix Arch, Phosh)
I noticed that cycling (turning on then off) the speakerphone mode, makes the microphone volume okay. This makes me think, the mic setting used for the speaker mode, could be used as a workaround for the default (internal earpiece) mode. I think it's just a matter of diffing mic settings in VoiceCall and HiFi configs, to find out the right parameters.

I have my doubts though, if such 'fix' should be implemented by distros - assuming gain will be added in modem firmware in the future, we will have to revert to the current config.

PR #25 (2021-08-20): Add call volume information to README
This adds some information about the call volume to README.md. There is probably not an easy universal solution for the values, so users will have to experiment with what works for them. This way, they will at least know where to look, so this should close #7, as nothing can be done about this from the modem firmware.