the-modem-distro/pinephone_modem_sdk #10: [PlaMo] Ofonoctl modem initialization time

PsychoGame, 2021-05-29:
Hello @Biktorgj 

What is the minimal necessary recommended time for the open firmware modem to be initialized?
When using Plasma Mobile on Manjaro Arm, the ofonoctl.service failed to start using the standart ofonoctl.service.
I already altered the service to restart on failure by setting option "Restart=on-failure".
This makes the modem usable for telephony at least, by retrying to initialize the modem until it succeeds.
I found out on the other hand this will break the wwan part of the modem firmware.
When choosing a big enough sleep timer before starting ofonoctl this initializes modem/wwan properly, problem is I don't want to add a to big sleep timer.
Originally ofonoctl has a 15 second sleep timer before starting the real ofonoctl:
> [Unit]
> Description=Poweron-Modem
> After=ofono.service
>
> [Service]
> Type=simple
> ExecStartPre=/usr/bin/sleep 15
> ExecStart=/usr/bin/ofonoctl online
> Restart=on-failure

> [Install]
> WantedBy=default.target
Was original modem firmware quicker in initializing/starting up?

Biktorgj, 2021-05-29:
From reset to USB ready is about 23 seconds. Never saw any issues in pmOS/plasma not detecting the modem, but I haven't tested with manjaro tbh. Will find me some time to get me latest beta and give it a look. One question though, shouldn't ofono just wait for the modem to appear? It's what ModemManager does...

PsychoGame, 2021-05-29:
To be honest I'm not completely familiar with Ofono.
All I know is that with the standard supplied ofonoctl from PlaMo Beta 4 in Manjaro the modem isn't initialized properly.
Now I believe that pmOS makes use of OpenRC instead of SystemD.
Don't know if these services are handled completely different.
When making the service restart on failure, then telephony etc. works fine.
Now I know that it takes 23 seconds to boot from reset to ready, I set the sleep time in the ofonoctl to 25 seconds.
With this change initializing the modem/wan related stuff works perfectly fine

closed 2021-06-23 by Biktorgj