the-modem-distro/pinephone_modem_sdk #9: Sometimes modem disappears partially [mobian]

kkeijzer, 2021-05-29:
Sometimes the modem disappears partially. `gnome-control-center` will say "No wireless / QMI device found", `ip a` will still show a `wwan0` device, but with no IP addresses or routing tables, and `gnome-calls` will state that there is no voice-capable modem detected.

This can happen after resuming from deep sleep, but sometimes also while the phone is awake. (I sometimes wake up finding the modem "broke" overnight, while the phone was charging and not going to deep sleep.)

The modem is still detected with `lsusb` in this state; no different from normal operation:

```
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 008: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The only way to get the modem to work again, is by running `adb shell reboot` from the PinePhone. (`adb reboot` does not work.)

Attached are [openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6564458/openqti.log) and dmesg of both the [modem](https://github.com/Biktorgj/pinephone_modem_sdk/files/6564459/dmesg-modem.log) and the [PinePhone](https://github.com/Biktorgj/pinephone_modem_sdk/files/6564461/dmesg-pinephone.log).

I can't see anything out of the ordinary. Do you have any suggestions of where I should look the next time this happens?

Biktorgj, 2021-05-29:
There is a race condition between eg25manager amd ModemManager in mobian, amd generally, in all distros using ModemManager that haven't been updated with Dylan's patches for suspend/resume.
The problem is ModemManager disconnects the modem on suspend and reprobes the modem on resume, and that takes about 10-15 seconds. If it takes a bit longer than expected, eg25manager ends up unbinding and rebinding modem's usb port, making modemmanager segfault and restart in the middle of the probe. 
All that would just be annoying, the problem comew from QMI. When ModemManager connects it registers itself as a qmi client. The modem supports, if I recall, about 20-25 concurrent clients. But when modemmanager dies or is unbinded that client is not released from the modem, and it ends up rejecting new clients when it's full, forcing you to restart... I am waiting for mobian to update ModemManager with those patches (Dylan's patches were ready to be merged to ModemManager last time I checked)

DylanVanAssche, 2021-05-29:
@Biktorgj My patches are shipped as `modemmanger-git` package

kkeijzer, 2021-05-30:
@DylanVanAssche Where can I find this package? Is a deb being built? Because I can't find it in the repositories.

Biktorgj, 2021-05-30:
You'll probably need to switch to the unstable branch (https://blog.mobian-project.org/posts/2021/03/15/unstable-distro/) and then simply apt install modemmanager-git

I'm currently testing the package too

kkeijzer, 2021-05-30:
Is it working for you? Voice calls and sms work for me, but data does not.

I keep getting

```
NetworkManager[601]: <warn>  [1622413024.3334] modem-broadband[cdc-wdm0]: failed to connect modem: Couldn't bind mux data port: QMI protocol error
```

when I try to enable the mobile connection.

Biktorgj, 2021-06-02:
Not as far as I can see, and it doesn't work for me either in factory firmware or custom, so I guess it's missing some polishing... (works perfectly fine in pmOS)

PsychoGame, 2021-06-03:
Looks like Manjaro Arm Plasma Mobile is also affected by this. Although I don't know if it's really a modem issue. My data worked fine until ModemManager got updatet. But i'll try stock modem again to see if data works on there, so I can really tell if it's modem related or just upstream package trouble.

Biktorgj, 2021-06-06:
Looks like an issue on my side. Still investigating, though it _looks_, like ModemManager now tries to bind to some port and is unable to... either there has been some change in libqmi or modemmanager now does thing differently. 

- Journal begins at Mon 2021-04-12 18:21:25 UTC, ends at Sun 2021-06-06 06:00:09 UTC. --
jun 06 05:57:45 mobian systemd[1]: Starting Modem Manager...
jun 06 05:57:45 mobian ModemManager[519]: <info>  ModemManager (version 1.17.1) starting in system bus...
jun 06 05:57:46 mobian systemd[1]: Started Modem Manager.
jun 06 05:57:50 mobian ModemManager[519]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/soc/1c19000.usb/musb-hdrc.4.auto': not supported by any plugin
jun 06 05:57:50 mobian ModemManager[519]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/soc/1c10000.mmc/mmc_host/mmc1/mmc1:0001/mmc1:0001:1': not supported by any plugin
jun 06 05:58:09 mobian ModemManager[519]: [/dev/cdc-wdm0] Opening device with flags 'version-info, proxy'...
jun 06 05:58:09 mobian ModemManager[519]: [/dev/cdc-wdm0] created endpoint
jun 06 05:58:09 mobian ModemManager[519]: cannot connect to proxy: Could not connect: Connection refused
jun 06 05:58:09 mobian ModemManager[519]: spawning new qmi-proxy (try 1)...
jun 06 05:58:09 mobian ModemManager[519]: <warn>  [ttyUSB0/probe] failed to parse QCDM version info command result: -7
jun 06 05:58:09 mobian ModemManager[519]: [/dev/cdc-wdm0] Checking version info (45 retries)...
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0] QMI Device supports 30 services:
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    ctl (1.5)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    wds (1.67)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    dms (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    nas (1.25)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    qos (1.12)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    wms (1.10)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    auth (1.3)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    at (1.2)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    voice (2.1)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    cat2 (2.24)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    uim (1.46)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    pbm (1.4)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    test (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    loc (2.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    sar (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    ims (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    ts (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    tmd (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    wda (1.16)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    csvt (1.1)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    imsp (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    imsa (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    coex (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    pdc (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    rfrpe (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    dsd (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    unknown [0x2d] (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    unknown [0x30] (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    unknown [0x32] (1.0)
jun 06 05:58:10 mobian ModemManager[519]: [/dev/cdc-wdm0]    unknown [0x36] (1.0)
jun 06 05:58:16 mobian ModemManager[519]: <info>  [device /sys/devices/platform/soc/1c1b000.usb/usb2/2-1] creating modem with plugin 'quectel' and '6' ports
jun 06 05:58:16 mobian ModemManager[519]: <warn>  [plugin/quectel] could not grab port ttyUSB3: Cannot add port 'tty/ttyUSB3', unhandled port type
jun 06 05:58:16 mobian ModemManager[519]: <info>  [base-manager] modem for device '/sys/devices/platform/soc/1c1b000.usb/usb2/2-1' successfully created
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0] Opening device with flags 'version-info, proxy'...
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0] created endpoint
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0] Checking version info (45 retries)...
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0] QMI Device supports 30 services:
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    ctl (1.5)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    wds (1.67)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    dms (1.0)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    nas (1.25)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    qos (1.12)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    wms (1.10)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    auth (1.3)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    at (1.2)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    voice (2.1)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    cat2 (2.24)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    uim (1.46)
jun 06 05:58:16 mobian ModemManager[519]: [/dev/cdc-wdm0]    pbm (1.4)
[... cut ...]
jun 06 06:00:08 mobian ModemManager[519]: <info>  [modem0] state changed (registered -> connecting)
jun 06 06:00:08 mobian ModemManager[519]: <info>  [modem0/bearer3] net link qmimux0 created (mux id 1)
jun 06 06:00:08 mobian ModemManager[519]: <warn>  [modem0/bearer3] connection attempt #4 failed: Couldn't bind mux data port: QMI protocol error (3): 'Internal'
jun 06 06:00:08 mobian ModemManager[519]: <info>  [modem0] state changed (connecting -> registered)
jun 06 06:00:08 mobian ModemManager[519]: <info>  [modem0/bearer3] connection #4 finished: duration 0s, tx: 0 bytes, rx :0 bytes

Biktorgj, 2021-07-23:
> Sometimes the modem disappears partially. `gnome-control-center` will say "No wireless / QMI device found", `ip a` will still show a `wwan0` device, but with no IP addresses or routing tables, and [...]

I just posted a pre-release that might mitigate the issue described here (if using ModemManager in Mobian -not modemmanager-git- )

If you're still running Mobian you can give it a try: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.4
It won't make the modem recover any faster on resume, but should hopefully make it not need a reboot after a bunch of suspend-resume cycles

kkeijzer, 2021-07-23:
To be honest, this makes things worse. Resuming from suspend, I now never have internet. When using `IPV4V6` as the PDP type, I get two IPv6 addresses on the interface, none of them working. When I use `IP` as the PDP type, I get no IP address at all.

After the modem is woken up, NetworkManager seems to hang on every interaction. Running something like  `nmcli connection up Default` will just time out.

Biktorgj, 2021-07-23:
Interesting, works perfectly fine in postmarketOS. Could you share a full dmesg and the contents of /var/log/openqti.log?

kkeijzer, 2021-07-23:
I'm running modemmanager on Mobian unstable. modemmanager-git does not work at all. Or does it with this release?

Biktorgj, 2021-07-23:
This won't fix anything with modemmanager-git since it's badly broken, uust wanted to know your setup to try to replicate when I get some wifi to get latest mobian again 

If you have a spare sd, it'd be interesting if you could run a live postmarketOS edge/phosh distro to check if your IPV4/6 issues are fixed, since pmOS is bundling a much newer modemmanager version.

In any case, I'd need to see what is going on with the firmware to know where to touch next :)

kkeijzer, 2021-07-23:
[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6871277/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6871278/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6871279/modemmanager.log)

Biktorgj, 2021-07-24:
Okay, that makes sense I guess. I wasn't tracking the release requests from the host, and, as in Mobian ModemManager releases itself prior to suspend, when reattaching to the modem it was kicking it out only to reconnect, which made matter worse.

Let's see if this fixes it... This is only a rootfs image, please flash it on top of your system partition, 
`adb shell reboot; fastboot oem stay; fastboot flash system rootfs-mdm9607.ubi; fastboot reboot` should be enough

Let me know if this works for you
[mobian_test.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6871850/mobian_test.tar.gz)

kkeijzer, 2021-07-24:
My first impression is that this seems to be working. I resumed from suspend and I have a working connection now.

kkeijzer, 2021-07-24:
adb, however, doesn't seem to work any more:

```
kevin@pinephone:~$  adb devices
List of devices attached
```

kkeijzer, 2021-07-24:
Hmm, never mind. Rebooting to fastboot and then rebooting back seems to fix it. Maybe it was just a glitch.

I'll keep on testing. So far it seems to be working well. Let's hope it stays connected without having to be rebooted every now and then.

Biktorgj, 2021-07-24:
Found one more possible bug from ModemManager that I hadn't accounted for:

On resume with that version of ModemManager, if you suspend too quickly after wake up, it can start saturating the modem again.
Here's a newer version with a fix for that. If OpenQTI finds the ADSP is handling more than 32 clients it will kill all of them and start from zero (I saw that when doing a lot of quick suspend-resume cycles client count could go up to more than 200 concurrent connections, it's not normal to behave like that but maybe some users have suspend set to 1 second or something like that)
[mobian_test2.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872103/mobian_test2.tar.gz)

kkeijzer, 2021-07-24:
I suspend after two minutes, so it probably won't be an issue for me, but I flashed it and will test it.

I tried `AT+ADBON`, but that didn't help. Rebooting the phone didn't help either. But going to fastboot and back apparently did help, and got ADB working again. Weird.

I'll see if I can find the time to try another distro to test a newer ModemManager. In my previous ModemManager log, you could also see that IPv4 wasn't working on a dual stack connection. I guess that would be my only remaining issue.

kkeijzer, 2021-07-24:
I still get some issues where I have to reboot the modem or restart ModemManager after resume. The main error I get is that the PPP dialer won't start.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872526/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872527/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872528/modemmanager.log)
[networkmanger.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872529/networkmanger.log)

Biktorgj, 2021-07-24:
That's on me, I messed up in the rmnet proxy thread function I think...
Third time is the charm?

[mobian_test3.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6872792/mobian_test3.tar.gz)

kkeijzer, 2021-07-25:
It seems to be working better. I've been running it since last night, and resuming from suspend works all right now.

However, this afternoon, I saw my network was down while the PinePhone was plugged in (so not sleeping). I checked `ip addr` and saw I had two IPv6 addresses again; none of them working. I had to run `adb shell reboot` before I could get it to work again.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6874123/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6874124/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6874125/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6874126/networkmanager.log)

P.S. I also tested PostmarketOS today, and with the newer ModemManager the IPv4/IPv6 dual stack bug is still present. I only get IPv6 addresses unless I re-register with `IP` as the PDP type and then change it to `IPV4V6` so IPv6 is added.

Biktorgj, 2021-07-26:
Okay, another test build, 0.2.7. 
I'm not going to say this fixes the disappear issues completely until you tell me so, because everytime I think I fixed it another way of crashing appears.

https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.7

Once thing I was thinking about your disconnect-reconnect. If you're custom handling the modem profile and it suspends and resumes and at some point something breaks and ModemManager thinks it is a new modem, you might want to do a test run disabling suspend hooks in ModemManager. It's not great but it will stop trying to detach itself from the modem everytime you suspend, and wil recover more quickly. If a USB disconnect happens inside the modem and the connection dies and it needs to renegotiate, this last firmware "might" (|| should || will?) handle it better than previous ones.

To change it, add the command line arg " --test-no-suspend-resume" to the modemmanager systemd service:
/usr/lib/systemd/system/ModemManager.service

`ExecStart=/usr/sbin/ModemManager --test-no-suspend-resume`

kkeijzer, 2021-07-26:
I haven't yet tested disabling the suspend hooks, but the current firmware does not seem to change much otherwise. In fact, it looks as if it randomly disconnects now. And sometimes when I run `adb shell reboot`, the phone seems to lock up, as if something is generating a lot of I/O.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6880316/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6880319/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6880317/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6880318/networkmanager.log)

Biktorgj, 2021-07-27:
There's something funky with your pinephone at the 520 second mark:
[  520.363640] msm_thermal:do_freq_control Limiting CPU0 max frequency to 1190400. Temp:60

The modem was hitting the temperature throttling threshold, don't know if it was charging or what...

When I close all the connections to the Modem when modemmanager fails to wakeup or suspend, I found it doesn't seem to like to get a ton of packets it didn't request. I'm currently testing draining them before returning control to it, and at least in pmOS it behaves as expected. Flashing Mobian again to a SD to test out, but in case you want to give it a shot:

[mobian_test5.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6882888/mobian_test5.tar.gz)

kkeijzer, 2021-07-27:
It was indeed charging, but it was on my desk, so it's weird that it got so hot.

Anyway, tested the last firmware, but I still have to reboot the modem after resume.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6883964/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6883968/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6883965/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6883967/networkmanager.log)

Biktorgj, 2021-07-27:
Ok, with latest firmware, can you update _only_ the adsp?

Download this: https://github.com/Biktorgj/quectel_eg25_recovery/blob/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi

And flash it from fastboot: 'fastboot flash modem NON-HLOS.ubi'

In this latest firmware WDS seems to work better than the previous one. Maybe the ADSP firmware is less buggy and the userspace side was worse?

kkeijzer, 2021-07-27:
Ok, testing with version 0.2.7 + mobian_test5.tar.gz and NON-HLOS.ubi now. I also added the `--test-no-suspend-resume` parameter to ModemManager.

By the way, it is cleaner to use an override instead of editing stuff in `/lib/systemd/`

So:

`sudo systemctl edit ModemManager.service`

Paste this:

```
[Service]
ExecStart=
ExecStart=/usr/sbin/ModemManager --test-no-suspend-resume
```

And then:

`sudo systemctl daemon-reload`
`sudo systemctl restart ModemManager.service`

kkeijzer, 2021-07-27:
So far it's been working pretty well. I was able to resume from suspend immediately this way. But after a couple of times, when the phone woke up, the modem was gone. Invisible with `lsusb`, so I wasn't able to use ADB. I tried running `systemctl restart eg25-manager.service`, but to no avail. Only rebooting the phone seemed to help.

[dmesg-pinephone.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887522/dmesg-pinephone.txt)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887523/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887524/networkmanager.log)

kkeijzer, 2021-07-27:
Hmm, it just broke again with the 'two IP addresses, none of them working' problem. It wasn't even suspending when this happened.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887953/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887956/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887954/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6887955/networkmanager.log)

Biktorgj, 2021-07-28:
On the disappearing side, I see the kernel failing on suspend with the SDIO driver which I've never seen before... is your SD card okay? :)

I also wonder if your vpn client is forcing everything to go through the tunnel, and if the tunnel is dying on suspend blocking all connections outside, did you check that?

Let's try another quirk, but this time from the kernel. Your logs do not show anything out of the ordinary unless I missed something, and I've spent 16 hours with no resets-no crashes-always recovering with latest firmware+modem ADSP firmware in Mobian unstable (without no-suspend-resume), and some time with no suspend-resume and a script that made the system suspend without giving the phone time to settle in between. I patched the modem kernel so it doesn't send the missed disconnect signals, and I even got ModemManager to crash but it recovered itself and got network again, so I'm now at a point where I can't seem to make it "die".

Please try this modified kernel, no need to flash it, just a one time boot: `adb shell reboot; fastboot oem stay; fastboot boot mdm9607-perf-boot.img`

[modem_kernel_test_no_missed_disconect.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6890228/modem_kernel_test_no_missed_disconect.tar.gz)

And let's see how it goes

Biktorgj, 2021-07-28:
> So far it's been working pretty well. I was able to resume from suspend immediately this way. But after a couple of times, when the phone woke up, the modem was gone. Invisible with `lsusb`, so I [...]

If this happens again, please connect through minicom to the serial port (/dev/ttyS2) and issue an AT+RESETUSB, maybe it can come back this way. There's a bug I'm looking into in the A64 USB stack, where it will start failing probing the modem USB and will just drop messages to the kernel log stating "Maybe USB cable is bad". If that's the case the modem won't come back, but the AT interface will still be working fine since it doesn't depend on USB

kkeijzer, 2021-07-28:
I'm pretty sure my SD card is okay. But my `/` is on the eMMC. The SD card only contains `/home`.

The reason I'm using the VPN is because the dual stack setup is so broken. When I use `IPV4V6` as PDP type, I only get an IPv6 address, because ModemManager for some reason then thinks IPv4 is not allowed by the APN. When I use `IP` as PDP type, I only get an IPv4 address (obviously), but then I don't have IPv6 at all. The VPN can connect over either IPv4 or IPv6 and then do 6in4 or 4in6. NetworkManager starts the VPN as a secondary for the LTE connection, so the VPN is not started at all until the LTE connection is up.

If we could find a way that dual stack would just work, I probably wouldn't be using the VPN at all.

The only way for me to get that to work now, is running this script after every (re)connection:

```
#!/bin/bash

mmcli -m any --command='AT+CGDCONT=1,"IP"'
mmcli -m any --command='AT+CFUN=0'
mmcli -m any --command='AT+CFUN=1'
sleep 1
mmcli -m any --command='AT+CGDCONT=1,"IPV4V6"'
nmcli connection up 'Standaard'

exit 0
```

But I'll try with the modified kernel, `IP` as the PDP type, and no VPN. But not having IPv6 will be really litiming though.

kkeijzer, 2021-07-28:
So far it seems to be going all right with this kernel. I'm using dual stack IPv4+IPv6 using the script above, but the VPN is disabled. I would have to run that script on every (re)connect, but I guess I can live with that for now. I'm also still using `--test-no-suspend-resume`.

Should I eventually flash this kernel?

Biktorgj, 2021-07-28:
If it's working correctly you can flash it :)

I think part of what is messing with the script could be timing, issuing a cfun=0 next to a cfun=1 without some sleep might be too fast for the modem, maybe you can add some sleeps in between and see if it helps

kkeijzer, 2021-07-28:
The script itself is working fine. The problem is that without the script, ModemManager only gets an IPv6 address when the PDP type is `IPV4V6`. For some reason, when I register to the network with `IPV4V6`, ModemManager claims the IPv4 call is not allowed, and only gives me an IPv6 address.

When I register with `IP` as the PDP type, I do get an IPv4 address, but (obviously) no IPv6. But when I change from `IP` to `IPV4V6` *after* registering (so registering with `IP` and then changing to `IPV4V6` and restarting the connection), I do get both an IPv4 address and an IPv6 address. It's really weird..

kkeijzer, 2021-07-28:
I'll add some logs.

Probably not interesting:

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892689/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892694/openqti.log)

When the modem is booted with `IPV4V6` as PDP type, and registers like that:

[modemmanager-before.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892691/modemmanager-before.log)
[networkmanager-before.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892693/networkmanager-before.log)

As you can see, IPv4 fails, and I only get an IPv6 address.

When I run the script, which changes the PDP type to `IP`, registers again to the network, and then changes it to `IPV4V6` immediatelf afterwards:

[modemmanager-after.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892690/modemmanager-after.log)
[networkmanager-after.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6892692/networkmanager-after.log)

So for some reason, registering with `IPV4V6` breaks IPv4, but changing it to `IPV4V6` after registering allows both to work.

The kernel is flashed by the way. I'll continue testing.

Biktorgj, 2021-07-29:
Hey @kkeijzer , is your modem still working or have you needed to reboot?

PsychoGame, 2021-07-29:
Today's update of packages on Manjaro  Phosh  development made everything go bad. After each sleep session I need to reboot to get mobile back online. will try the no-suspend-resume hook to see if this improves the situation

Biktorgj, 2021-07-29:
Get me a dmesg of the PinePhone kernel when it happens next time, please!

PsychoGame, 2021-07-29:
I have the logs for you on my phone already. Will upload them later in the day. Currently I'm in an amusement park with my kids.

kkeijzer, 2021-07-29:
> Hey @kkeijzer , is your modem still working or have you needed to reboot?

So far it's still working. I rebooted the entire phone once for a different reason though. But I travelled to work and back today, and the connection didn't drop along the way. So I'm quite positive for the time being.

kkeijzer, 2021-07-29:
Unfortunately I celebrated too early..

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6902013/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6902016/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6902014/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6902015/networkmanager.log)

Biktorgj, 2021-07-29:
Are you still using the no suspend resume param?

kkeijzer, 2021-07-29:
Yes. And the phone was plugged in (so not going to sleep) when this happened.

Biktorgj, 2021-07-30:
So this is a message I hadn't seen in a long time:
```
[16023.571441] msm_hsusb msm_hsusb: CI13XXX_CONTROLLER_RESUME_EVENT received
[16023.571767] msm_otg 78d9000.usb: Avail curr from USB = 500
[16023.578149] msm_otg 78d9000.usb: Avail curr from USB = 2
[16023.578305] msm_hsusb msm_hsusb: CI13XXX_CONTROLLER_SUSPEND_EVENT received
[16023.595749] udc_irq: USB reset interrupt is delayed
[16023.595862] msm_hsusb msm_hsusb: CI13XXX_CONTROLLER_RESUME_EVENT received
[16023.596080] msm_otg 78d9000.usb: Avail curr from USB = 500
[16023.596309] msm_otg 78d9000.usb: Avail curr from USB = 100
[16023.596453] diag: USB channel diag disconnected
[16023.935561] android_usb gadget: high-speed config #1: 86000c8.android_usb
[16023.941484] diag: USB channel diag connected
[16023.946852] msm_otg 78d9000.usb: Avail curr from USB = 500
[16036.505955] msm_otg 78d9000.usb: Avail curr from USB = 2
```
So at some point the Pinephone issued an USB reset, could you grab the log from eg25manager just in case?
My test is so far 23 hours 47 minutes without a crash, reset, reboot or anything needed in pmOS Edge

Did you both @kkeijzer @PsychoGame  have to reboot the modem or the phone to make it work again or it recovered?

kkeijzer, 2021-07-30:
It didn't recover. I had to run `adb shell reboot`. Attached is yesterday's eg25-manager log.

[eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6905476/eg25-manager.log)

DylanVanAssche, 2021-07-30:
> It didn't recover. I had to run `adb shell reboot`. Attached is yesterday's eg25-manager log.
> 
> [eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6905476/eg25-manager.log)

Ah interesting! the reset from above comes from the eg25-manager... So it is not from the A64 or modem kernel, but the A64 userspace.

PsychoGame, 2021-07-30:
Hello @Biktorgj,

No my modem didn't recover automatically in my case.
I tried restarting eg25-manager and also NetworkManager, but to no avail.
In my case had to reboot the phone to get it working again for a short while.
In my case enabling persistent ADB makes everything more stable. At the moment can't really figure out why.

DylanVanAssche, 2021-07-30:
@PsychoGame enabling ADB may reduce the ability to suspend the USB core (which increases the power consumption). This may be the reason why it seems more stable.

Biktorgj, 2021-07-30:
Next time this happens (there's a lot of conditions the modem cannot handle, like something else removing it from the host like eg25-manager did), try to restart only ModemManager (systemctl restart ModemManager). At least that way we can know if OpenQTI can handle it (it should but as it is obvious, what works for me might not be what works for you :))

DylanVanAssche, 2021-07-30:
Maybe the eg25-manager shouldn't do this reset anymore? If it is stable enough, it cannot be lost anymore...

kkeijzer, 2021-07-30:
It just broke again, while plugged in and doing nothing. I tried restarting ModemManager, but that did not help.

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/6906911/dmesg.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6906914/openqti.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6906912/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6906913/networkmanager.log)

DylanVanAssche, 2021-07-30:
@kkeijzer Do you have the eg25-manager log as well?

kkeijzer, 2021-07-30:
Sure.

[eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6906971/eg25-manager.log)

DylanVanAssche, 2021-07-30:
Hmmmm it is again the eg25-manager who kicks out the modem, maybe that reset logic of the eg25-manager must be revisited with the masking @Biktorgj did regarding disconnect events?
Would be interesting if we can run the eg25-manager but with the reset logic disabled.

Biktorgj, 2021-07-30:
Fastest test would be to boot the phone, let it ask for the PIN, then disable eg25manager service and kill it with -9 so it doesnt shut down the phone. Let the modem live only attached to modemmanager and see how it behaves

kkeijzer, 2021-07-30:
The daemon just respawns when I try that.

ThePosichronic, 2021-08-03:
Hello. I got curious and decided to try to help test this. I'm running biktor's 0.28 firmware with the updated ADSP on Arch Linux ARM Phosh with ModemManager 1.16.8. 

I tried what biktor suggested, simply stopping the eg25-manager service and killing it once the system boots. However, as soon as eg25manager is dead the modem seems to stop communicating in a useful way with the system. It doesn't vanish, but stops being useful for data, texting, calls, etc. I don't have logs for this event, but I can reproduce it if they may be useful.

After that I tried getting the eg25manager source and commenting out the two modem_reset calls (I assume this is what I should be commenting out, I have no knowledge of the eg25manager code). One in udev.c and the other in suspend.c. Unfortunately not long after I rebooted with the newly compiled eg25manager, the modem simply vanished again. Logs for my test are attached.

[2021-08-02-21:21-dmesg.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6920461/2021-08-02-21.21-dmesg.log)
[2021-08-02-21:21-eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6920462/2021-08-02-21.21-eg25-manager.log)
[2021-08-02-21:21-modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6920464/2021-08-02-21.21-modemmanager.log)
[2021-08-02-21:21-networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6920471/2021-08-02-21.21-networkmanager.log)
[2021-08-02-21:21-openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/6920472/2021-08-02-21.21-openqti.log)

Let me know if there's anything else I can do.

[37 more replies]

closed 2021-09-14 by kkeijzer