the-modem-distro/pinephone_modem_sdk #11: Whats the state of the AT commands?

Zapeth, 2021-06-02:
The README states

> Listen to AT Commands and blindly respond OK to everything not implemented (to make userspace happy)

Is there an overview somewhere which AT commands are fully/partially/not implemented? If not I think it would be helpful down the road to have a listing of that somewhere (maybe as a wiki entry?).

Possibly even with more detailed descriptions when a command might deviate from the original implementation, that way Quectel's AT command manual can still be used as a reference.

Biktorgj, 2021-06-02:
It's not really easy to tell :)
The full list of AT commands declared in the custom firmware is here:
https://github.com/Biktorgj/meta-qcom/blob/hardknott/recipes-modem/openqti/files/inc/atfwd.h

The only, partially implemented command (from stock) is AT+QFASTBOOT and AT+QDAI, but QDAI disabled on purpose, since the Pinephone only supports two audio configurations (8k and 16k sampling rates unless you want to play around with master-slave settings in pinephone audio but you'll run into a lot of problems), and those are implemented via new commands to support 8k/16k/48k sampling rates (stock firmware only supports 8/16).

The different commands I have implemented so far are these:
AT+QFASTBOOT: Jump to fastboot mode, like stock
AT+QDAI (but disabled)
AT+PWRDN : Shut down the module
AT+ADBON: Start ADB
AT+ADBOFF: Stop ADB      
AT+RESETUSB: Stop and start USB      
AT+REBOOT_REC: Reboot to recovery      
AT+EN_PCM16K: Set sampling rate to 16KHz
AT+EN_PCM48K: Set sampling rate to 48KHz
AT+EN_PCM8K: Set sampling rate to 8KHz

Now the problem comes here: Most of the other commands are "listened to" on the userspace but are really handled on the ADSP. The original firmware listens to the command "AT+CFUN", but all the work for that command is really handled by the DSP, the userspace is only notified after the fact, probably to reset some internal software state in the userspace. So most of the important commands will still do their thing even if the userspace isn't notified about them (which we don't need because we're leaving modem handling to the Pinephone's software directly most of the time), and the commands that we're strictly userspace (turn BT on, turn web server on or off and stuff like that) we don't even need them because we're not running any of that on the modem (I'd rather not run a web server in the modem itself, bluetooth or wifi modules aren't connected to it etc.)

Anyway, you're right and I will add to my todo list to write a small doc with the commands I've implemented, which ones I did not, and which ones are pending (for the record, if you want to programatically know if you're running this firmware you only need to run AT+PINE64ROCKS, if you get an OK you know you're running this :)

Zapeth, 2021-06-02:
Thanks for the clarifications, I guess I'll leave this issue open until that documentation is written in case other people might have a similar question (but feel free to close it if necessary).

Biktorgj, 2021-06-09:
Just a heads up,

I have begun with the AT Command listing. It will be a ton of work to fix all the descriptions and go one by one to check which of them are "missing cooperation" from the userspace to actually work, but just in case you want to keep an eye on it: https://github.com/Biktorgj/pinephone_modem_sdk/blob/hardknott/docs/AT_INTERFACE.md

dagbdagb, 2021-06-09:
Megi did some reveng on the modem. Is any of that work of use here, @Biktorgj ?   https://xnux.eu/devices/feature/modem-pp-reveng.html

Zapeth, 2021-06-09:
@Biktorgj Thanks, thats already more extensive than I was hoping for.

I know it says its just a placeholder, but assuming the listing itself should be complete I compared the first table with the listing of the Quectel document and found some discrepancies (some possibly caused by your method of extraction?).

It wasn't too much work for me to go through all the commands in the document, so if you want I can do a PR with the commands that are listed in the overview, along with their descriptions and leave placeholders in "Handled by" and "Implemented" that are not yet filled in.

However on the topic of descriptions, while it would be helpful to list them I wonder if its safe to directly copy them, the document is not publicly available after all (though I guess the same goes for the firmware files, and nobody has raised an issue about those so far).

Biktorgj, 2021-06-18:
Not really worried with the descriptions to be honest, most of them are just that, and are mostly the same accross different documents from different vendors (all the AT+C**** commands are standard GSM commands, for example)

Any help is welcome, not only with documentation but with anything you might find broken, missing or that can possibly be optimized or that could add value, I started this project but it's in no way "mine" but of everyone who uses it or wants to help in any way :)

Zapeth, 2021-06-18:
Ok, I'm mostly finished going through all listed commands, just some remaining open points:
1. should the commands be sorted alphanumerically or grouped by their function (i.e. same order as in the document)?
2. some commands with parameters are listed explicitly in the document along with their base command (e.g. `AT+QMBNCFG` and `AT+QMBNCFG="<subcommand>"`), would it be better to only list the base command here or all subcommands as well? (the latter would add a little more bloat to the listing, but they are separate commands technically)
3. should any of the "Other related" commands be listed too (GNSS, MMS, etc), or omitted for now?
4. do you know if all listed commands are only handled by the ADSP in firmware, or are there any exceptions?

Biktorgj, 2021-06-19:
> Ok, I'm mostly finished going through all listed commands, just some remaining open points:
> 
>     1. should the commands be sorted alphanumerically or grouped by their function (i.e. same order as in the document)?
The first table is unsorted because that's the way it was in the document. I did a _really_ dirty parser and that's how it turned out. I assume everyone looking for a specific command will do CTRL+F anyway but feel free to rearrange them if you want 
> 2. some commands with parameters are listed explicitly in the document along with their base command (e.g. `AT+QMBNCFG` and `AT+QMBNCFG="<subcommand>"`), would it be better to only list the base [...]
Only the base commands I think. I already removed tons of variants of each command. QCFG has a lot of options, for example. We would have to split the document to make it readable otherwise 

>     3. should any of the "Other related" commands be listed too (GNSS, MMS, etc), or omitted for now?
These commands come from the AT manual, I haven't started with the GPS/GNSS one, those would be better in diffrerent files but there's enough work with this one already :)
In any case, both GPS and MMS only depend on the ADSP firmware, so they should all work regardless of the userspace you're running. I'd really like to upload the documentation here, but that will probably grant a dmca takedown

>     4. do you know if all listed commands are handled by DSP in stock firmware, or are there any exceptions? (or is that part of your current listing only a placeholder for now?)

First table is all the commands that are on the manual, second table is handled mostly by the userspace but there are some commands that are also run by the ADSP, third one is only userspace and not in stock.

Everything on the second table is handled either directly by a userspace daemon called `atfwd_daemon` or a mix of ADSP + atfwd_daemon. The first table shows all the commands in the manual, and everything that shows up on the first table that isn't listed on the second is handled exclusively by the ADSP.

By the way, thanks for the help :)

Zapeth, 2021-06-22:
> By the way, thanks for the help :)

No problem. I'm interested in trying out the firmware myself, but I have yet to find the time to do the necessary preparations to get it installed/running on my Pinephone (might be able to contribute more once I do).

Also, closing this issue since a dedicated document for AT commands exists now.

closed 2021-06-22 by Zapeth