the-modem-distro/pinephone_modem_sdk #13: Optionally route audio via USB

dagbdagb, 2021-06-05:
Seeing how this is not listed under 'Next steps' in the [README](https://github.com/Biktorgj/pinephone_modem_sdk#readme), I figured I could submit it as a (humble) feature request. If for nothing else, you could shoot it down and let this issue serve as a warning to those following in my steps.  :-)

As you have stated, the pinemodem does not need it, as audio is routed via i2s in that particular implementation. But those of use wanting to give this firmware a spin on other EG25-G implementations (plain USB), may not have that particular option.

Biktorgj, 2021-06-07:
You're right.
Not having my best week (month?) and haven't had much time for anything. Hadn't forgotten though, and now it is in the readme as it should.
Enabling USB audio is really simple, and the kernel is correctly activating it, but I have no idea if modem audio will pass through as-is without further manipulation in the modem's userspace)...

If you want to give it a shot, here's the only thing you need to do:

1. Open a shell through adb (adb shell)
2. Run mount -o remount,rw /
3. vi /etc/init.d/usb
4. Find the line where it says `echo diag,serial,rmnet,ffs > /sys/class/android_usb/android0/functions`
5. Add aduio after ffs, like this: `echo diag,serial,rmnet,ffs,audio > /sys/class/android_usb/android0/functions`
6. Save and exit
7. Reboot

On reboot (be extra careful when writing to that file otherwise usb won't start) you should see a USB audio card. If you need to do something special with that other card I don't know (check out Quectel's guide for that: https://forums.quectel.com/uploads/short-url/79VE3yUZiWsTQ5SQ3GnmxyH3mnW.pdf)

Before doing anything, just remember to run 'fastboot oem stay' before plugging the modem just to check that the recovery method works in case you mess something up :)

dagbdagb, 2021-06-07:
I'll give the manufacturer firmware a spin first, just to get a baseline of what the hardware can do. 

I managed to enable the USB audio function of the modem:
AT+QCFG="USBCFG",0x2C7C,0x0125,1,1,1,1,1,1,1
AT+QPCMV=1,2

Does your userspace handle these commands, or are they just punted to the DSP somehow? Trying to measure the likelyhood of your userspace 'just working'....

I also managed to initiate a call with modemmanager : 
mmcli -m 0 --voice-create-call='number=0043xxxxxx'
mmcli -m 0 --voice-list-calls                                                                                                                          
mmcli -m 0 --start -o <N>      where N is the index of the call, as listed in the output from the previous command.

But as you have stated earlier, I need to pipe the audio between the modem audio card and audio card handling the external mic/headset. I think I have what I need to jury-rig it with some 'arecord | aplay' atrocities.

Biktorgj, 2021-06-08:
USBCFG won't work in this firmware (hence the need to manually edit the usb init script), I think QPCMV might work, but not sure. Most of the commands registered in the userspace are handled directly by the DSP and then passed onto the userspace to finish things up, but hard to say without having the actual source for the atfwd_daemon... When testing a call from the Pinephone I could hear audio still going through i2s but at the same time the kernel was indicating activity in the usb audio driver so I honestly don't know what the outcome will be in your case

If you want to share your aplay atrocity I can give it a shot :)

dagbdagb, 2021-06-09:
Looks like I need to do some resampling for this to work. The modem UAC only allows for 8k sampling. 

Anyways, I established a call and hooked up audacity to the modem. Audacity did not detect any data from the USB sound card when my peer talked. I'll retest soonish. Need to figure out if I can actually set up calls between my twin SIMs...

gregvish, 2021-07-27:
Hi. Great project! I'm using an EG25G attached via USB to a PC. I've set up you FW on it a while ago. The only thing missing is indeed audio.

In your latest release you said "...added a command to enable USB audio". I did try to do the following:
```
AT+EN_USBAUD
AT+QPCMV=1,2
(call my #)
ATA
```

On the PC, I've detected the audio device. Then something like:
`arecord -D hw:2,0 -f S16_LE | hexdump`

Nothing is coming out (except the WAV header in the beginning that's added by arecord). Sending audio in also doesn't come out the other end of the call.

I take it AT+QPCMV is fully handled by the ADSP firmware? Not by your atfwd? 
I did not try to use the original FW to do this, but the AT+QPCMV step is from their "Quectel_LTE_Standard_UAC_Application_Note_V1.0.pdf", and I don't think I missed anything there.

Do you think there should be some kind of forwarding process on the linux?

Biktorgj, 2021-07-27:
That's mainly why I've left this issue on hold. I believe I will need to use something like arec/aplay in the userspace to pipe real and virtual modem audio cards, but have bigger fires right now that are more urgent. Could you try the stock firmware and tell me the exact commands you used? I can build you an image with arec and aplay preinstalled in the firmware so you can play a bit inside the modem itself if you want

dagbdagb, 2021-07-27:
@gregvish  Cool! Another USB-user. While your Hayes commands may be sufficient for a regular analogue  modem, it may be insufficient for a cellular modem. @Biktorgj  should know, I think.  :-)

Have you tried using modemmanager for setting up a call and see if this results in audio being routed to the modem audio port?

And yes, you definitely need something to forward audio between the modem usb audio devices and your regular audio card, if you want to hear the audio. It can possibly be done purely with a custom ~/.asoundrc on your host computer. Or check out alsapipe. But for testing if audio from the other end is coming through, you can of course just use arecord as you have done.

gregvish, 2021-07-27:
Ok, I've dug into this a bit.
On the original firmware, exactly what I did above works (it is what was described in the PDF after all).
I got data from `arecord` on my PC, and successfully piped it into my headphones. Works fine, albeit the sound is quiet.
Before and after the call, I ran the following on the EG25 via adb:
```
# ls -la /proc/*/fd | grep pcm
ls: /proc/1336/fd/3: cannot read link: No such file or directory
ls: /proc/self/fd/3: cannot read link: No such file or directory
ls: /proc/thread-self/fd/3: cannot read link: No such file or directory
```
And **during** the call:
```
# ls -la /proc/*/fd/ | grep pcm
ls: /proc/1241/fd/3: cannot read link: No such file or directory
lrwx------    1 root     root            64 Jul 27 20:29 24 -> /dev/snd/pcmC0D2c
lrwx------    1 root     root            64 Jul 27 20:29 25 -> /dev/snd/pcmC0D2p
ls: /proc/self/fd/3: cannot read link: No such file or directory
ls: /proc/thread-self/fd/3: cannot read link: No such file or directory
```
The process that opened that fd is `quectel_daemon`. Killing the daemon prevents the sound from going out to USB.
So the correct device is `/dev/snd/pcmC0D2c`. I compiled a static arm binary of the `tinycap` utility from https://github.com/tinyalsa/tinyalsa/ . This is supposed to be equivalent to `arecord`. However:
```
/ # /tmp/tinycap /proc/self/fd/1 -D0 -d2 | hexdump
Unable to open PCM device ()
```
It fails like this regardless of the call state (during or not). Of course the `quectel_daemon` has to be dead for it to even manage to `open()` the device. Maybe this is just the `tinycap` utility being incompatible? 

Note: annoyingly, the setting AT+QPCMV=1,2 does not survive reboot. Without doing it, everything seems to work, but there is no data coming of the USB sound device on the PC (I guess no data is routed from the DSP to the pcmC0D2c device).

I guess the next step would be indeed to try the full ALSA utility to try to capture from the device? If you have static binaries (or a whole image I guess) I'd like to try.

dagbdagb, 2021-07-27:
@gregvish 
Did you also fire AT+QCFG="USBCFG",0x2C7C,0x0125,1,1,1,1,1,1,1       prior to     AT+QPCMV=1,2     ?

Biktorgj, 2021-07-27:
PCM0C2 is the internal adsp audio card (c/p) when the call is being routed through 2g/3g
PCM0C4 is used when the call goes through VoLTE. The issue I have is I don't know how to route it to the android usb gadget audio driver in the kernel. Closed blobs have a ripped off arec/aplay programs that pipe audio from the internal card to the usb one. I can do that from OpenQTI to, but I don't know where to send them. When I looked, I didn't find a new pcm device when I enabled USB audio... 

OpenQTI already handles internal audio, and it would be relatively simple to spawn a separate thread to do that if we can find the "usb" side of it, but as I said, havent had much time to look how to interface with the audio part of the android usb gadget. With that you wouldn't need to send any command on each boot, and I could make it persistent too

gregvish, 2021-07-27:
> @gregvish
> Did you also fire AT+QCFG="USBCFG",0x2C7C,0x0125,1,1,1,1,1,1,1 prior to AT+QPCMV=1,2 ?

Oh yes, I forgot. That's indeed essential. That's sort of equivalent to AT+EN_USBAUD in @Biktorgj's firmware.

> PCM0C2 is the internal adsp audio card (c/p) when the call is being routed through 2g/3g
> PCM0C4 is used when the call goes through VoLTE. The issue I have is I don't know how to route it to the [...]

Ok, I don't know about the USB sound system. I'll be fine just having the PCM piped over one of the TTYs :) I'll look into the code of OpenQTI to try to do this.

Biktorgj, 2021-07-28:
Hi, 
In this firmware you won't need to run any AT commands from the Quectel manual. What's missing is basically what AT+QPCMV does I think.
Just uploaded a new release with included alsa-utils and support for UAC2 from OpenQTI. This makes the modem know there are two sound cards. If you can find the correct commands to pipe arec+aplay from one card to the other, we'll probably be able to make this work and I can then try to implement this automatically.

Grab the release here:
https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.8

Remember PCM device 2 in card 0 for normal calls, 4 for VoLTE :)

gregvish, 2021-07-28:
Hi. Thanks so much for the new firmware!
After playing around for a bit, I'm able to send sound in/out the gadget soundcard on the EG25, like this:
`cat /dev/urandom | arecord -r44100 -c1 -D hw:1,0 -f S16_LE `
`aplay -r 44100 -c1 -D hw:1,0 -f | hexdump`
And I can see it on my PC (or send it from the PC). The card refuses for any rate other than 44100 (on the PC side too). This wasn't the case with the original firmware (PC set the rate to 8khz and the card accepted).
However I was unable to use the CS-voice soundcard. All I get is something like:
```
sh-5.1# arecord -c1 -D hw:0,2 -f S16_LE
Recording WAVE 'stdin' : Signed 16 bit Little Endian, Rate 8000 Hz, Mono
arecord: set_params:1407: Unable to install hw params:
ACCESS:  RW_INTERLEAVED
FORMAT:  S16_LE
SUBFORMAT:  STD
SAMPLE_BITS: 16
FRAME_BITS: 16
CHANNELS: 1
RATE: 8000
PERIOD_TIME: 128000
PERIOD_SIZE: 1024
PERIOD_BYTES: 2048
PERIODS: (3 4)
BUFFER_TIME: 500000
BUFFER_SIZE: 4000
BUFFER_BYTES: 8000
TICK_TIME: 0
```
And in dmesg:
```
[ 2880.535263]  Circuit-Switch Voice: ASoC: no backend DAIs enabled for Circuit-Switch Voice
[ 2880.572448] voc_end_voice_call: Error: End voice called in state 0
```
Doesn't matter if a call is going on or not.

Biktorgj, 2021-07-28:
The problem is that I don't know how they made it work :)

The error you see in dmesg should be normal if there's no active call or you didn't set the appropriate mixer controls, that's one of the things openqti manages automatically.

Audio in the modem is supported in 8k/16k/48k. I guess the best approach will be to either stick to 8k or 48k and align everything from the modem side. I rather use 48k so you can have VoLTE HD if supported by your carrier.

So, next step should be:
In OpenQTI, when USB audio is ON, set sampling rate to 48K, check if UAC2 supports that sample rate, and set them both to same params
When there's a call in progress, fork a thread that passes buffers between soundcards and hopefully that should get us audio
Once that's done I can make usb audio persistent like with ADB

dagbdagb, 2021-07-28:
So, just as a datapoint, with **original firmware** and no AT commands issued after a fresh reboot of both host and modem, I see:

~ $ cat /proc/asound/cards
 [......]
 3 [EG25G          ]: USB-Audio - EG25-G
                      Quectel EG25-G at usb-0000:02:00.0-2, high speed

(I have issued the two commands above previously) 
So far, so good.

I can also initiate an outbound call, either by Hayes commands, or with mmcli. The call arrives at my cellphone. I can answer it. 
Recording provides audio **if** I enter AT+QPCMV=1,2 prior to the call. So yeah, does not look like this command survives a reboot here either.

arecord -c1 -D hw:3,0 -f S16_LE mycall.wav  (on host)

gregvish, 2021-07-28:
Another datapoint. When strace'ing openqti, I see no events arriving (or being handled) when I'm setting up the call using `ATD`. Therefore the code in `handle_call_pkt` in openqti doesn't run, and the mixer and all that aren't being setup.
I believe you intend on "catching" the voice call indication when it is coming as a QMI packet from the DSP, but I couldn't find a way to make the DSP send one. Probably need to somehow register to receive this event.
I'm controlling the modem with QMI using `qmicli` on the host, but that doesn't have options to start voice calls.

dagbdagb, 2021-07-28:
Is it `--voice-create-call` which is flaky or the subsequent `--start -o N` ?  I notice that `-o` is only documented under common options, which is kind of odd.

gregvish, 2021-07-28:
> Is it `--voice-create-call` which is flaky or the subsequent `--start -o N` ? I notice that `-o` is only documented under common options, which is kind of odd.

It's the ModemManager in general. Probably on it's first interaction with the modem. If I get this again I'll share, but in the strace it looked like some kind of loop of QMI messages with the modem.

gregvish, 2021-07-28:
Just to add data, here are the QMI messages that I can see in the strace of `openqti` after a successful call to `--voice-create-call` and `--start -o`:
```
'0152008003010401002400460001060001010102010511040003030405120500a9010200001503000105001d0200eba61e0400767c490026080003000000030000002802000a01290500a9010200002a010000',
'01380080030104020024002c0001060001010102010511040003030405120c00a90102000743656c6c636f6d2802000a01290500a901020000',
'010c00000301000c0020000000',
'011800800301020c0020000c0002040000000000010200a405',
'01310080030104030024002500010600010101020108110200010b1503000108001d0200feff1e040020823300250200dc02',
'01290080030104040024001d00010600010101020108110200010b1b0100011c0800e507071c1008200c'
```
In `handle_call_pkt`, the byte at offset 4 is verified to be 9 (QMI_VOICE_SERVICE in the qmi.h headers), however here all of the messages have this field set to QMI_NAS_SERVICE (3). Don't have a parser for QMI yet, so these hexdumps are it for now.

dagbdagb, 2021-07-28:
@gregvish care to ELI5? Modemmanager does not use libqmi to place calls? Is that it?

Biktorgj, 2021-07-28:
> Another datapoint. When strace'ing openqti, I see no events arriving (or being handled) when I'm setting up the call using `ATD`. Therefore the code in `handle_call_pkt` in openqti doesn't run, and the mixer and all that aren't being setup.
> I believe you intend on "catching" the voice call indication when it is coming as a QMI packet from the DSP, but I couldn't find a way to make the DSP send one. Probably need to somehow register to receive this event.
> I'm controlling the modem with QMI using `qmicli` on the host, but that doesn't have options to start voice calls.
> 
> Edit: using `mmcli -m 0 --voice-create-call` is flaky for me (with an ubuntu ModemManager), but in the one time it did work, I'm pretty sure the code in `handle_call_pkt` still didn't run judging from the strace output.

Yes, this will only work through a QMI client on the other side. I could try to add support for ATD commands by subscribing to the command from the at forwarding thread, but I don't know if the modem spits out the call type to be able to guess which pcm device to use. All my tests are done with ModemManager and oFono

You have a debugging option in openqti, which will spit out all the packets between the host and the adsp, just add " -- -l" to the init script and it will dump everything to openqti.log

gregvish, 2021-07-29:
Ok. So to keep playing with this I've made a modified version of `qmicli` to which I added an ability to call the `qmi_client_voice_indication_register` API so that voice QMI notifications will arrive from the modem. This works.

`openqti` intercepts these, and sets up the mixer and all that now. 
However, apparently this is not enough. Trying to `read()` the pcm fd blocks and then fails. I also tried the SNDRV_PCM_IOCTL_READI_FRAMES ioctl since I've seen this used in `tinyalsa` to get data from the pcm fd, this fails too. Using `arecord` instead of opening the PCM fd in `openqti` (with a modified version of openqti) also fails (in the same way).

So I went to the original FW to analyze further. It seems that there is some "magic" that is configured at boot by a process called "alsaucm_test". Killing the process changes nothing (the sound still works), but preventing it from running at boot *does* break the ability of the original FW to transfer sound. I'm guessing that this "magic" is some additional mixer configuration of some sort that directs the modem PCM output to the USB output. `strace` looks quite complicated there.

On the other hand, and I guess you've already seen this, the original `quectel_daemon` does open the PCM devices the same way as `openqti`, but does not do anything with them (no read/write thread as far as I can tell).

Before I keep looking, just wondering if you've already looked into this `alsaucm_test` process while researching?

gregvish, 2021-07-29:
Ok, so the stuff I mentioned above was a red herring. After properly reading about what `set_mixer_ctl` acutaly does, I managed to make the voice over USB to work with this minimal change to `openqti`:
```diff
   switch (type) {
   case 1:
     logger(MSG_DEBUG, "Call in progress: Circuit Switch\n");
-    set_mixer_ctl(mixer, TXCTL_VOICE, 1); // Playback
-    set_mixer_ctl(mixer, RXCTL_VOICE, 1); // Capture
+    set_mixer_ctl(mixer, "Voice_Tx Mixer AFE_PCM_TX_Voice", 1); // Playback
+    set_mixer_ctl(mixer, "AFE_PCM_RX_Voice Mixer CSVoice", 1); // Capture
     strncpy(pcm_device, PCM_DEV_VOCS, sizeof(PCM_DEV_VOCS));
     break;
   case 2:
     logger(MSG_DEBUG, "Call in progress: VoLTE\n");
-    set_mixer_ctl(mixer, TXCTL_VOLTE, 1); // Playback
-    set_mixer_ctl(mixer, RXCTL_VOLTE, 1); // Capture
+    set_mixer_ctl(mixer, "VoLTE_Tx Mixer AFE_PCM_TX_VoLTE", 1); // Playback
+    set_mixer_ctl(mixer, "AFE_PCM_RX_Voice Mixer VoLTE", 1); // Capture
```
The "magic" is that this AFE mixer is connected to the USB soundcard "internally". This was tested on the older image, not the new one where you upgraded to UAC2.
Currently, I only tested the sound one way (RX), but it'll probably work the other way too (will test soon).

gregvish, 2021-07-30:
Hi. I tested this with a full call, and it's now working. I don't have VoLTE here though. I'll make a PR to your `meta-qcom` repo, that adds the proper `if`s and sets up the mixers according the USB mode.
The `AT+QPCMV=1,2` command is not necessary. Basically, you just set up the EN_USBAUD mode, and then make calls over QMI normally (with ModemManager or whatever). In order to support calling via AT commands only, maybe `openqti` will have to look at `/dev/diag` instead of sniffing the QMI messages.

Biktorgj, 2021-07-30:
QPCMV probably just tells it what you're manually doing

The idea would then be to enable the AFE loop when usb audio is enabled, and shut it down otherwise, then look for another enpty segment in the misc partition to store and retrieve the persistency mark.

There's another AFE loop mixer control for VoLTE, I think I have the names somewhere

gregvish, 2021-07-31:
I'll link the PR I opened here for reference: https://github.com/Biktorgj/meta-qcom/pull/6

Biktorgj, 2021-08-21:
So... I ended up redoing the thing. USB audio should work, persistence too, and it should automatically enable audio on boot

Can you give it a try?

[0.3.1_test1.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/7026223/0.3.1_test1.tar.gz)

gregvish, 2021-08-22:
Can confirm that your version is working. Thanks

closed 2021-09-08 by Biktorgj