the-modem-distro/pinephone_modem_sdk #16: openqti takes 100% CPU on the modem

raphj, 2021-07-04:
Hello,

I flashed the latest release (https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.3) this afternoon. It seems to works well.. Congratulations and thank you for this work. This is an important project.

However, I noticed the phone stays warm when plugged and the battery charges very slowly while the modem is on. I'm not sure whether the firmware is at fault here, but connecting to the modem using `adb shell` and running top shows that `openqti` constantly uses 60 to 100% of the modem's CPU.

Back to the stock firmware, I don't see this CPU usage in the modem.

Is this expected?
If no, how can I help / debug this? is there an easy way to install gdb or perf and attach to the process to see what's happening?

I'm using Mobian (up to date).

Zapeth, 2021-07-05:
Looking at the [source files for openqti](https://github.com/Biktorgj/meta-qcom/tree/hardknott/recipes-modem/openqti/files) it seems like there should be a log file written to at `/var/log/openqti.log`, that might already give a clue what the issue is.

If not you could also try increasing the log level to include debug messages, though I'm not sure what the recommended procedure for restarting the service with the required argument is.

raphj, 2021-07-05:
Thanks, maybe the right incantation of `ps` / files in `/proc/$PID` can give the right parameters to pass to openqti if restarting it manually works.

I'll look into it.
I notice my modem stays warm even with the stock firmware so there may be something happening here.

Biktorgj, 2021-07-07:
Yes, you have a log file at `/var/log/openqti.log`. By default it only logs a few things but you can enable more verbose logging if needed. OpenQTI shouldn't be using that much of the CPU, but I have seen that myself once or twice when hitting a ModemManager bug

First things we should know: 
1. What distro are you running? 
2. Were you downloading a lot of stuff? 
3. Which phone edition you have? 
4. Can we get a copy of 'openqti.log' and a modem kernel log? That will make it all a lot easier to understand.

Notice the modem's temperature will be affected by three things: if there's an active data session (in 2G/3G/4G), if there's an active call, or if the A64 in the other side of the board is working a lot and heating up the motherboard. If Modem's temperature goes above 50ºC you should see a message stored in the kernel log

raphj, 2021-07-07:
> * What distro are you running?

Mobian, up to date.

> * Were you downloading a lot of stuff?

No

> * Which phone edition you have?

The Manjaro Community Edition

> * Can we get a copy of 'openqti.log' and a modem kernel log? That will make it all a lot easier to understand.

Sure! I'll produce it when I get the chance to reflash the firmware. Maybe tonight. I flashed the stock firmware back, but I noticed the modem is still warm with it actually.

> if there's an active data session (in 2G/3G/4G)

Mobile data where enabled but not actively used.

This is a bit irrelevant to this issue, but In the meantime, I took those measures to keep the phone cold:
 - make the phone sleep after 30 seconds of inactivity, both on battery and while charging
 - disable mobile data (but since the phone goes to sleep anyway)
 - shut down two CPU cores, and bluetooth, and wifi if I can.

Now the phone has incredible battery life. The one measure that helps is the first, but the modem gets warm fast as soon as the phone is awake (on the stock firmware, same thing on the custom one).  Battery percents fly away.

PsychoGame, 2021-07-09:
I've checked my PinePhone as well today, to see if my device would have the same problem as well.
Minimum CPU load I've seen is 68%, max cpu load around 98% in top.
Toggling cellular connection to "off" in Phosh doesn't seem to make a difference.
I'm also running Manjaro ARM Phosh Dev Image, updated to 08-07-2021.
Modem is not actively used, just connected to my Phone carrier and using Wi-Fi to write this message actually.
The version of ModemManager in use on this image is: 1.30.2.
I'll try to enable more verbose logging in modem, and provide you with dmesg dump and openqti.log
If I can be of any more assistance just let me know.

Biktorgj, 2021-07-09:
Interestin, I'm in Mobian and here it is stuck at 17%, which is much more reasonable. Can you attach dmesg and openqti logs?

PsychoGame, 2021-07-09:
Hello @Biktorgj,

I still have to produce the logs, cause I was busy to find a way of reproducing the bug each time.
After fresh reboot everything with openqti is perfectly fine, also after one hour of running.
The problem of openqti filling up cpu usage arises as soon as a usb reset seems to happen.
When using this firmware I always disable ADB with the AT+ADBOFF command as it should save a running process on the modem CPU.
This command triggers a modem reset and then comes back online.
Now when re-enabling adb with AT+ADBON and then issue "top" in the adb shell, you'll notice the high CPU load on the openqti process.
Also command AT+RESETUSB produces the same result.
I can imagine this high cpu load puts some extra drain on the battery life.
As USB resets also appear occasional during normal usage in some cases, I imagine this issue affects everybody eventually.
I have to go to sleep at the moment, so i'll provide you with the logs tomorrow.

Biktorgj, 2021-07-10:
Okay... that would make perfect sense!

On start, ModemManager / oFono register themselves to the ADSP via QMI, and they need to deregister themselves. When you issue a +resetusb / +adbxx, OpenQTI cycles usb off and on but ModemManager/ oFono don't have a chance to deregister themselves, for them it just vanished and they need to do the setup again.
When they connect again, the ADSP gives them new client IDs, but the older ones are stuck there, so it starts sending all the notifications twice (network strength, call indications, GPS activity etc.), so it starts saturating OpenQTI. Do it enough times and you're filling USB with duplicated connections (openqti's main job is to act like a proxy between the adsp firmware and the exposed usb to the pinephone).

I could do two things to fix this:
1. When turning USB off, send a signal to the ADSP to disconnect all services from everywhere.
2. Make ADB settings persistent

Fastest thing to implement would be to make ADB persistent by using the data or system partition, but cleaner one would be to be able to kick Pinephone's connections to the modem when needed, that way we could fix the scenarios where ModemManager or oFono crash and leave stale connections open...

Zapeth, 2021-07-10:
> When they connect again, the ADSP gives them new client IDs, but the older ones are stuck there, so it starts sending all the notifications twice (network strength, call indications, GPS activity etc.), so it starts saturating OpenQTI.

Wouldn't it (also) make sense for ModemManager/oFono to release the old client IDs when they receive a new ID, or are there cases where it might be desirable/necessary for them to keep multiple IDs active? If not it might be worth opening an issue about this on their end too.

> 2. Make ADB settings persistent

I'm unsure what you mean with this, what exactly would be kept persistent in order to circumvent this issue? Either way, I agree that 1. sounds like a cleaner solution (might be desirable to have that functionality for other use cases anyway?)

> that way we could fix the scenarios where ModemManager or oFono crash and leave stale connections open...

Is there a heartbeat communication with openqti, or how are you able to register if a ModemManager/oFono crash ocurred, on openqti side?

PsychoGame, 2021-07-10:
> Okay... that would make perfect sense!
>
> On start, ModemManager / oFono register themselves to the ADSP via QMI, and they need to deregister themselves. When you issue a +resetusb / +adbxx, OpenQTI [...]

Just my 2 cents:

In my opinion the first solution would be the most clean one, as the second solution only covers the ADB specific reset. The modem gets reset in some more cases than only toggling off/on the adb daemon on the modem, so this solution would work in most cases a usb reset is triggered. I don't know how the official Quectel firmware handles these cases? Maybe in the sources/firmware images some hints can be found?

Then the ADB persistent settings case:
Having ADB in the modem firmware is very handy for developers, but I like the option to get it toggled on/off.
When daily driving I don't need the ADB daemon continuously running, as it's an extra process running in the firmware which, although very less resource using is still using some resources from CPU. Also I think having easy access to the modem OS by adb poses some security risks, as no password is really necessary to enter the adb shell.

Maybe it's possible to generate two images at build time?
1. A "production" ready one with ADBD standard toggled off, for normal users/usage.
2. A developer image which standard has ADBD running on boot
Don't know how much work that would be to be honest, and also what is your opinion on this?

I really need to dive in the sources on your github to get to understand how everything is build/working and linked. Cause I would be glad to help you in advancing this project.

Greetings Jasper

Biktorgj, 2021-07-10:
Openqti acts as a proxy, and relays everything that goes between modemmanager and the adsp. I can track when modemmanager/oFono registers and if I get a new connection attempt when it was already registered I could release all the previous connections and that would solve this problem *and*  #9 

For the ADB part, I think what I'm going to do is store the ADB setting in the misc partition. Start with default off, and if you turn it on it will be persistent accross updates without needing to use the data partition (and wasting 2 seconds on boot mounting it). 

I've started looking into the required command sequence to release qmi client IDs, will keep this issue open and post a test build when I have something working. Sounds good?

PsychoGame, 2021-07-10:
Sounds good to me. I'll keep an eye on this issue and will test it for you as soon as it comes available.
Thank you for your continuing effort, and your great support.

PsychoGame, 2021-07-14:
I think I already have some understanding on how this ADB is wired in your builds.<br>
You have the usb-normalboot and usb-adb in the meta-qcom, and if I'm correct only usb-normalboot is used currently.<br>
If you wish I will take the enable the ADB part upon me?<br>
My implementation idea is as follows,if you agree of course, otherwise I will try to implement it as you like:<br>
Build new modem images with adb-normalboot/usb-noadb side by side.<br>
Create a file in /persist which holds the adb enable/disable toggle at boottime.<br>
Create new custom AT command for example AT+TOGGLE_ADBBOOT, which triggers script to set the variable in /persist.<br>
After this, a script will run which links the appropriate usb init file, usb-normalboot or usb-noadb to /etc/rcS.d/S97usb<br>
Is this a solution you would be happy with<br>

PsychoGame, 2021-07-14:
My first rough experimental build is working already.<br>
Will put everything on my github and make a pull request as soon as I have it ready.<br>
Then you can review it to see if you like the implementation.<br>

Biktorgj, 2021-07-14:
> I think I already have some understanding on how this ADB is wired in your builds.
> 
> You have the usb-normalboot and usb-adb in the meta-qcom, and if I'm correct only usb-normalboot is used currently.
> 
> If you wish I will take the enable the ADB part upon me?
> 
> My implementation idea is as follows,if you agree of course, otherwise I will try to implement it as you like:
> 
> Build new modem images with adb-normalboot/usb-noadb side by side.
> 
> Create a file in /persist which holds the adb enable/disable toggle at boottime.
> 
> Create new custom AT command for example AT+TOGGLE_ADBBOOT, which triggers script to set the variable in /persist.
> 
> After this, a script will run which links the appropriate usb init file, usb-normalboot or usb-noadb to /etc/rcS.d/S97usb
> 
> Is this a solution you would be happy with

I have half of it implemented in openqti, but maybe we can use some of mine and some of yours. My issue with /persist is that I don't usually mount it since each partition mounted on boot delays everything for about 2 seconds and I'm trying to get all fitted into a ramdisk to be able to just use a boot image instead of all those partitions. Setting AT+ADBON/ADBOFF already takes care of storing  the setting in an unused block in the misc partition, but I haven't started with the usb script part where it reads it on boot. It should be as simple as a small program which reads in the specific block of the misc partition and if it is set, change the functions and transports in usb sysfs accordingly. 
All this might not be futureproof once Konrad gets mainline running but should be more than enough for now. If you want I can push the pieces I have so you can see what I was doing :)

PsychoGame, 2021-07-14:
Hello @Biktorgj,

I would be happy to check your part of the implementation.
I could write up the script for changing the parameters at boot time.
My implementation is actually quite simple.
In my modem firmware build I make use of the already available packages usb-normalboot and usb-noadb.
This method is quite effective.
I'll push my commits tonight to a fork of this project so you can check it out.

Biktorgj, 2021-07-15:
Hi @PsychoGame, just pushed some changes to the meta-qcom repo that add the store/retrieve the persistent adb setting in the misc partition: https://github.com/Biktorgj/meta-qcom/commit/6cf10ddb155d98acbb0631641f697f95a863816a

The idea is to use adbcheck into the usb script and set USB functions accordingly. 
It's missing some stuff in openqti to read and store these settings on boot so it also knows if adb was enabled from boot or not, but more than enough to begin with.
adbcheck will return:
-EINVAL if it cannot read the misc partition
0 if adb was disabled
1 if adb was enabled

If adbcheck returns EINVAL, maybe the most sensible approach would be to actually start ADB since that would mean something went really wrong, so if adbcheck returns 0 we don't enable ADB and in any other case we do start it. We will probalbly also need to either chainload adb from the usb init script or add a check in its script to account for this

Don't pay much attention to the QMI register/release notification code, it's half done and doesn't work ;)

the01player, 2021-07-21:
I'll be installing the firmware after this issue gets fixed. Excited!

Thank you for all of your hard work.

Biktorgj, 2021-07-23:
Hi everyone, I think I fixed both the CPU hogging problem and #9 .
The CPU hogging problem came from a closed file descriptor in the GPS proxy thread after a sudden USB reset/disconnect
As far as I can see on the logs, the forced QMI client release is also working. OpenQTI will wait for 240 seconds after the first register request from mm/ofono, and if after that it reads a new register request it will kick all the registered clients from the ADSP. 

With some more time I'll find the services registered by ModemManager *after* entering the PIN code to account for them and not need that big of a grace period.

Testers welcome: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.4
If you see, in Gnome Calls the message "Can't place calls: No voice-capable modem available", it's because USB started before the modem was fully booted. You can either send a "AT+RESETUSB", restart the ModemManager service, or reboot the phone. Any of them should fix the problem (still trying to find why ModemManager thinks that sometimes)

raphj, 2021-07-23:
Thank you very much. I've been quite busy and have been following this discussion without participating (sorry for that) but will try to find some time to try these fixes as soon as possible.

PsychoGame, 2021-07-23:
Hello @Biktorgj,

I tried to purposely break OpenQTI, to see if the memory hogging would appear again.
As far as I can see this issue has been resolved.
Will test some more to see if this release is not experiencing any other regressions.

Greetings Jasper

Biktorgj, 2021-07-24:
Hi everyone, I made some additional fixes to yesterday's test build to handle some more cases (specifically things I found on Mobian)

I haven't had the chance to test these builds in anything oFono, so if someone's using UBPorts/KDE some insight would be appreciated
Link to release:
https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.5

PsychoGame, 2021-07-28:
From my tests this problem can be marked as solved.

raphj, 2021-07-29:
I just flashed version 0.2.8 of this firmware. The phone is able to charge quite rapidly and `top` in the modem reports 0% CPU usage when not using  something related to the modem. Can be marked as solved for me too. Should I close the issue?

Biktorgj, 2021-07-29:
I'd say let's give it a day and if nobody has more issues related to this I will post a non-prerelease package with all the changes and close it.

Is that okay for everyone?

PsychoGame, 2021-07-29:
That's fine by me. Bit off topic, but did you see my follow up comment in the PR about the fixup_mounts script? Found something in the init.c of mdm9640 which might give a entry point for a solution to use a separate fs than ubifs on the root partition, without breaking the compatibility with stock.

Izaic, 2021-08-01:
> Hi everyone, I made some additional fixes to yesterday's test build to handle some more cases (specifically things I found on Mobian)
>
> I haven't had the chance to test these builds in anything [...]

Plasma Mobile is switching to MM anyways.

Biktorgj, 2021-08-02:
Yes, but other OS like UBPorts or Salifish also use oFono, and I wouldn't want to neglect them either :)

The idea behind this firmware is to be a drop-in replacement for the stock one with as little glitches as possible. If it only works correctly with half of the telephony stack all this one year effort is worth nothing, since I cannot expect a user to flash one modem firmware or another depending on the running distro, just to avoid doing that little extra work

I got a report that in fact something related to this is breaking oFono (#22 ), hopefully I can find an easy fix!

PsychoGame, 2021-08-02:
The build you provided in here v0.2.5 in working fine.
Also 0.2.6 is working perfecrly, but 0.2.7 breaks things.
Probably related to the ModemManager specific hook 0x1a which got introduced between 0.2.6 and 0.2.7.

Biktorgj, 2021-08-07:
New test build is up: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.9
Given that it's all mostly working as it should, closing this issue now. If anybody faces more issues please let me know!

closed 2021-08-07 by Biktorgj

vmaurin, 2022-04-26:
Digging a bit for this issue, but I have noticed that with 0.6.1, I have quite often the same symptom (so warm phone + non charging battery), but when I try to access with adb, it actually stops, so I am not able to see which process is making the CPU run. With 0.5.9, it is happening less often