the-modem-distro/pinephone_modem_sdk #17: How can mobian accomodate your firmware

spaetz, 2021-07-06:
Hi, I am somewhat involved with Mobian, and wanted to ask, what we can incorporate, so that mobian works better with your firmware. Obviously we want still work with the default firmware, but if there are kernel patches, config changes etc that we can include that makes using your firmware easier without breaking the default firmware, we would be happy to do so.

Biktorgj, 2021-07-08:
Hi @spaetz! 

I'm a Mobian user myself, so I would be thankful for some fixes that don't even relate to the custom firmware at all, but would benefit all mobian users :)

We have two ModemManager versions available via apt in Mobian:
1. The shipped version of Modem Manager, which doesn't contain the suspend-resume fixes and makes the modem sometimes disappear, slow to recover from suspend, and in my firmware's case, whenever you hit somewhere between 20-30 suspend cycles, the modem just kills all connections to the PinePhone (this is due to a problem between ModemManager and EG25-Manager when reprobing takes too much time). 
2. ModemManager-git: This seems to incorporate @DylanVanAssche's fixes which makes the modem stay alive between suspend-resumes. The problem with this (I don't know if it's built from master ModemManager branch) is that it also incorporates a bug which makes data unusable, at least with my firmware (I had mixed results with stock firmware)

So it would be great if we could get ModemManager running with suspend fixed (https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/merge_requests/540) without the other issues (https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/issues/367), that would improve the modem situation a lot for all users (though I'm unsure if it would also need a newer kernel since I'm running a custom 5.12 version myself on top of mobian)

For the rest, this firmware works as a drop-in replacement and Mobian already incorporated the required configuration changes to allow ring-in to work in eg25-manager, so the only thing we could add is 48KHz sampling rate support. That would be great, but my patches are so far ugly and doesn't handle bluetooth HFP voice use case, so better to not push that for now (even if HFP doesn't yet work in Mobian).

DylanVanAssche, 2021-07-08:
@Biktorgj The mobile data issue is resolved in master, if `modemmanager-git` is updated to the latest commit it should be fine.
HFP works apparently, but needs configuring, I haven't got it working either.

Biktorgj, 2021-07-08:
The only repo I can find is this:
https://gitlab.com/mobian1/packages/modemmanager

On the phone so a bit tedious to check but it seems newest branch is at least 3 months old, so if Modemmanager-git is also based on this repo it would need some updating...

spaetz, 2021-07-08:
Hi, thanks for your reply. It is good to know that *most* stuff is ready. As to modemmanager vs modemmanager-git, I know that @a-wai did not want to bump to an unrelease git snapshot in our current "stable" release, especially as it includes patched git versions of like 3 other libraries too. As soon as we switch from bullseye to the new bookworm branch of Debian, we hope to be able to bump the current stable release to newer versions.

As to the repository of modemmanager-git, I'll let others comment, as I am not sure myself where the public repo with those patches is.

kkeijzer, 2021-07-08:
I'm also using this firmware with Mobian, and basically the only problem is that the `modemmanager` package is kind of broken, and the `modemmanager-git` package is really broken. Currently `modemmanager` is the only thing working, but it breaks occasionally. `modemmanager-git` could fix that breakage, but it doesn't work for data at all. So if that package could be updated to reflect the current master branch of ModemManager, I think everything would work pretty well.

spaetz, 2021-07-09:
Hi, Data has been fixed for me with modemmanager-git for quite some time (working for me). a-wai has said he plans to update to current master and publish the mm-git repository.

I agree that mm-git is so much better than mm. It is a pity that Debian is basically still on hold before the release. Even Debian Experimental only has 1.16.6...
-- 
Sent from mobile device.

kkeijzer, 2021-07-09:
I just tried `modemmanager-git` again, but it's still not working. Well, calls and sms are, but data is not.

[image]

Biktorgj, 2021-07-09:
There's a part of the problem that I haven't been able to figure out yet with modemmanager-git, which is that my firmware, for whatever reason, doesn't bring up/expose Qualcomm CSD service. It might not even be related at all and it's an issue only with certain carriers or configurations, but with stock firmware in mobian I have very slow data with modemmanager-git, but with my firmware ModemManager just brings up and down wwan after 0 secons of establishing the connection.
There are so many moving pieces in the interaction between the modem, carrier and the pinephone that certain things get hard to debug....

DylanVanAssche, 2021-07-10:
@Biktorgj You have to check if MM-git has this patch: https://gitlab.freedesktop.org/mobile-broadband/ModemManager/-/commit/a357777d39c87da89c285d496a48b53f294f1e4b

kkeijzer, 2021-07-13:
So would this mean that when `modemmanager-git` gets promoted to be `modemmanager` this firmware will stop working?

There are some other bugs that I have found, like IPv4 not working when IPv6 is available, unless you set the PDP type to `IP`, then re-register, and then set it back to `IPV4V6`. But I'm not sure if this is a problem with this modem firmware or with the old `modemmanager` package.

I would like to test this further, but I'm not sure where to start, as I'm not really sure how Mobian builds its `modemmanager` packages.

Biktorgj, 2021-07-14:
At least until next update. This firmware seems to work perfectly fine in pmOS Edge phosh, manjaro etc. Which I believe are bundling newer modemmanager versions where the mentioned bug is not present... But I'm already more than 40Gb into my dataplan so I can't start downloading all distros to recheck :( (no wifi until next month)

One thing you can check about ipv4/6 is if behaviour is the same with stock firmware (my carrier doesn't seem to work in ipv6).

kkeijzer, 2021-07-17:
@Biktorgj Yeah, I'm kind of reluctant to flash the blob firmware when I'm not even sure if the bug is not in ModemManager instead of the firmware. With Mobian shipping such an old version it makes more sense to me to first wait for an update.

Anyway, in short the problem is that when I first connect, I only get an IPv6 /64 address and no IPv4. It will say the IPv4 call failed with reason `[internal] pdn-ipv4-call-disallowed`. Unfortunately a lot of things are still IPv4-only, so not having IPv4 is inconvenient. (Even though I hate IPv4 with the burning force of a thousand suns.)

When I then run these commands (re-register with `IP` PDP type and after that change it to `IPV4V6`), I get a proper dual stack connection:

```
mmcli -m any --command='AT+CGDCONT=1,"IP"'
mmcli -m any --command='AT+CFUN=0'
mmcli -m any --command='AT+CFUN=1'
sleep 1
mmcli -m any --command='AT+CGDCONT=1,"IPV4V6"'
nmcli connection up 'Default'
```

@spaetz So apparently this firmware is working properly with a newer ModemManager in PostmarketOS and Manjaro. I would guess that Mobian's `modemmanager-git` is also too old then. Do you know who within Mobian is the maintainer of that package, and if so, could you ask them to rebuild it against a newer git version? Especially checking for the commit that @DylanVanAssche mentioned.

Biktorgj, 2021-07-17:
Those commands go straight to the ADSP firmware, so it doesn't really matter much which "userspace" firmware you're running in the modem. ModemManager 1.18, whenever gets released fixes _a lot_ of stuff, maybe that's fixed too?

Packaging newer ModemManager for Mobian is a bit complicated because it depends on various libraries that are not-that-often updated in Debian, so when rebuilding it for mobian, all their dependencies need to be upgraded as well... and it's easy to end up in a dependency hell :(

kkeijzer, 2021-07-17:
I can see that, but on the other hand, isn't ModemManager basically the core component of Mobian? I mean, it's supposed to be Debian for mobile devices, and then you can't really say "well, the mobile part isn't working that well, because we'd have to ship newer libraries".

spaetz, 2021-07-18:
It is Debian for mobile devices with the goal to deviate as little from upstream Debian as possible. Given that these packages are all in upstream Debian, forking them is not attractive for us.
We are currently a bit slowed down by Debian's freeze right before their next release, and I expect the next modemmanager versions to come in shortly after the release.

Alexmitter, 2021-07-27:
Hello, I have the same issue here. Mobian mm-git, up to date biktor firmware and data is not working.
It works fine in the same constellation but with the stock firmware.

Is there anything I could do to help diagnose the issue?

kkeijzer, 2021-07-27:
`modemmanager-git` is just completely broken in Mobian and has to be updated, which will probably not happen until it's rebased on bookworm.

Another thing Mobian could do, however, to improve this firmware's usability, is updating the `alsa-ucm-conf` package, changing two values in `/usr/share/alsa/ucm2/PinePhone/VoiceCall.conf`:

```
cset "name='AIF1 DA0 Playback Volume' 90%"
cset "name='AIF2 DAC Playback Volume' 90%"
```

See also #7.

spaetz, 2021-07-29:
If it were only that easy. Others are calling for reducing volume levels to 50%. It seems that various editions, firmware and perhaps even individual device are quite different when it comes to appropriate volume. We are looking forward to transition to Debian bookworm soon and make use of pipewire.

Biktorgj, 2021-08-14:
Just a heads up, Mobian unstable now includes ModemManager 1.18rc1. I've been running it for about 12 with zero issues (only one problem where sound is muted for a couple of seconds on call start but that's my fault). That update with the latest firmware seems to be working great (fingers crossed!)

spaetz, 2021-08-15:
Great to hear that it works way better than 1.14.x!
Looking forward to flashing your modem firmware on mine (been a bit 
hesitant so far :-))

P.S. Feel free to close this issue. It was designed to get a 
conversation started, and you can always contact us when mobian can help 
to support the FOSS firmware better.

Biktorgj, 2021-08-16:
Sounds good to me, I will still keep #9 open for a while until we make sure everything is working properly but given Mobian is now up to date (and passed the 24 hours-no-crashes test) I'll close this one.

Thanks a lot for the work you guys are doing!

closed 2021-08-16 by Biktorgj