the-modem-distro/pinephone_modem_sdk #19: ADB: Switch to online after fresh persistant off reboot

PsychoGame, 2021-07-24:
Currently trying to work out this issue.
I'm reporting this in the issue tracker for good measure, so other people don't need to file a separate bug report on this.

Currently after a fresh reboot, with persistent ADB disabled, you are not able to switch ADB straight to on using the AT+ADBON command.

Current workaround is one of the following two: 
1: Reboot the system after issueing the AT+ADBON command to switch on ADB at boot.
2: First issue AT+ADBOFF, causes a usb reset. Issue AT+ADBON after modem has come back online. This resets usb briefly again, but ADB should now be started.

Probably has something to do with OpenQTI not reading correct USB state or something like that. Have to investigate some more.

Progress can be tracked in here.

Biktorgj, 2021-07-25:
I think this is working in latest test build. Is it working correctly for you?

PsychoGame, 2021-07-25:
Closing this issue now. Released 0.26 build includes fix for this.

closed 2021-07-25 by PsychoGame