the-modem-distro/pinephone_modem_sdk #20: Significantly longer modem boot time compare to stock fw

dogtopus, 2021-07-27:
After flashing this firmware, the boot time of the modem is quite noticeably longer compare to stock. I guess it could be due to the hardcoded powersave cpufreq scaling governor. If this is the case it might be sensible to apply the cpufreq scaling governor settings after the modem is fully up.

(Actually is the application processor ever used in a CPU-bound way other than probably during booting?)

PsychoGame, 2021-07-27:
If you skim through the available sourcecode, you can find that during boottime the kernel has been hardcoded to the performance governor.
Only after the system is up/running it will switch over to powersave.
Nonetheless this is still an issue of course, so it's a valid issue report.
Thank you for thinking about possible solutions.

Biktorgj, 2021-07-27:
It's a couple of things actually.

From poweroff to ready it is about 25 seconds on older builds, from which:
1. 0.5 seconds on bootloader start
2. 2 seconds wait time in fastboot in case you need to recover from something, update etc 
3. About 5 seconds of total kernel boot time, including partition mounting and firmware loading
4. 10 seconds of ADSP firmware boot (this is unavoidable, and goes through system init while this happens)
5. 10 seconds of artificial wait time in USB so ModemManager cannot query the modem before it is fully booted.

In latest test builds the usb wait time has been removed and in testing state, leaving the modem from poweroff to ready in about 16-18 seconds (you can more or less measure from kernel boot to usb ready by running `adb logcat ;adb shell dmesg` )

Kernel always srarts in performance mode and steps back to powersave after the main daemon is ready (the ACPU only acts as a proxy so no need to have a lot of cpu power)

I could remove one more second by only using a ramdisk, and I could possibly trim another second from fastboot, but there's not much more to optimize in that without sacrificing the fastboot recovery options

Stock firmware gets USB ready at the 16 second mark approx, but without the fastboot wait, so if something goes wrong, user needs to open the phone and short the EDL pins, which is much more inconvenient I think

dogtopus, 2021-07-27:
Thanks for the clarification.

kkeijzer, 2021-07-27:
Definitely do not remove the fastboot wait. That would be highly inconvenient in case anything goes wrong when flashing updates.

dogtopus, 2021-07-27:
It's also only 2 seconds long which is quite insignificant.

max1220, 2021-07-28:
The "ramdisk"-approach still might be interesting:
First of, it makes testing new modem firmware easier, as you probably can just `fastboot boot` the initrd+kernel then, without the need to flash partitions at all.
Second, it should make the modem firmware more robust to failures, because (ideally) there is no filesystem to mount at all.

PsychoGame, 2021-08-02:
I've been doing some experiments with ubi fastmapping.
The first boot is just as fast as the normal boot, because it still needs to convert the ubi images, but each following boot makes use of this fastmapping.
My MTD11 partition now attaches in 0.3 sec approx.
I've uploaded the complete patchset to my Github today.
Also there's a PR on your kernel repo to pull it in when you think it's good to go.
I can share a test boot.img with you if you like.

PsychoGame, 2021-08-04:
Also I would like to do some experimenting in using SystemD as init manager. Currently we use SysinitV, which initializes each process after another, while SystemD makes use of parrallelization of different processes. This might provide us with some more optimalization in boot time.

PsychoGame, 2021-08-04:
Thank you for your detailed explanation about the different boot stages the modem firmware goes through.
Working the code slowly gives me more and more understanding how everything is wired.
Learning everyday, and eager to learn more to make usefull contributions.
Do you want me to share the fastmap boot.img which I'm using already for the last few days?
Or will you be building a new image with the Ofono fixes included?
In the meantime I'll try to continue the project to replace the android gadget driver from the code with the mainline gadget driver.
Already found some good info on the QMI replacement outside the Android gadget driver, and getting adb to work on mainline gadget driver should be no problem either.

Biktorgj, 2021-08-05:
Just a heads up, after adding back the top frequency for the cpu and with @PsychoGame 's recommended addition of the UBI fastmap feature the modem can boot in 13 seconds after exiting fastboot (usb ready and finishing audio initialization at that point). These changes will be in the next update.

dogtopus, 2021-08-05:
Great work! That's even faster than stock.

Biktorgj, 2021-08-07:
Seems that was just a one off thing or something, now I'm stuck again in 15-16 seconds... anyway, new build up:  https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.9

It at least doesn't need 23 seconds to start, will try to trim all that a bit more but give it a shot

Biktorgj, 2021-08-08:
So, after a few more days of testing, I'm going to have to rollback the UBI Fastmap feature. 
The modem might be come quite more unstable than it should when you reboot it after mounting it as read write:
```
[    3.244367] CPU: 0 PID: 1 Comm: swapper Not tainted 3.18.140 #1
[    3.249971] [<c0013554>] (unwind_backtrace) from [<c001181c>] (show_stack+0x10/0x14)
[    3.257771] [<c001181c>] (show_stack) from [<c02cb6a8>] (ubi_io_read_vid_hdr+0x48/0x1d8)
[    3.266414] [<c02cb6a8>] (ubi_io_read_vid_hdr) from [<c02c8c70>] (check_mapping.part.0+0x48/0x13c)
[    3.274850] [<c02c8c70>] (check_mapping.part.0) from [<c02c8f40>] (ubi_eba_read_leb+0xd0/0x3a4)
[    3.283403] [<c02c8f40>] (ubi_eba_read_leb) from [<c02c7ee4>] (ubi_leb_read+0x70/0xb8)
[    3.291297] [<c02c7ee4>] (ubi_leb_read) from [<c0147018>] (ubifs_leb_read+0x2c/0x94)
[    3.298978] [<c0147018>] (ubifs_leb_read) from [<c0148a70>] (ubifs_read_node+0x13c/0x2ec)
[    3.307183] [<c0148a70>] (ubifs_read_node) from [<c0145ef8>] (ubifs_read_sb_node+0x50/0x70)
[    3.315508] [<c0145ef8>] (ubifs_read_sb_node) from [<c0146710>] (ubifs_read_superblock+0x7b8/0xe20)
[    3.324536] [<c0146710>] (ubifs_read_superblock) from [<c0145278>] (ubifs_mount+0x9dc/0x1594)
[    3.333092] [<c0145278>] (ubifs_mount) from [<c00df034>] (mount_fs+0x58/0xe0)
[    3.340171] [<c00df034>] (mount_fs) from [<c00f8300>] (vfs_kern_mount+0x88/0x148)
[    3.347592] [<c00f8300>] (vfs_kern_mount) from [<c00fab0c>] (do_mount+0xaa4/0xbd0)
[    3.355182] [<c00fab0c>] (do_mount) from [<c00fae5c>] (SyS_mount+0x70/0x9c)
[    3.362135] [<c00fae5c>] (SyS_mount) from [<c090b0a8>] (mount_block_root+0x158/0x29c)
[    3.369892] [<c090b0a8>] (mount_block_root) from [<c090b3b8>] (prepare_namespace+0x84/0x1b4)
[    3.378362] [<c090b3b8>] (prepare_namespace) from [<c090ad68>] (kernel_init_freeable+0x14c/0x1a8)
[    3.387215] [<c090ad68>] (kernel_init_freeable) from [<c0629408>] (kernel_init+0x8/0xec)
[    3.395286] [<c0629408>] (kernel_init) from [<c000dbb8>] (ret_from_fork+0x14/0x3c)
[    3.402843] ------------[ cut here ]------------
[    3.407380] Kernel BUG at c02cb6a8 [verbose debug info unavailable]
```
After pushing a file, syncing, remounting read only, and rebooting, 8/10 the filesystem gets corrupt, so I guess I'll be back on track to try to move the filesystems to something else I guess...

I'll leave this open until this gets resolved

PsychoGame, 2021-08-08:
Hello @Biktorgj,

That's unfortunate, but a good catch nonetheless.
I didn't test this use case to be honest, and I don't think the regular user would actually hit this bug.
It looks to be related to the fact that UBI images/filesystems usually are used as read only FS's and Fastmap doesn't cope well with read/write actions.
For developers tinkering with the modem image I can imagine this is quite annoying of course.

Looks like we would need to alter the bootloader then.
Figure out if we could create some kind of logic to detect the filesystem which is used by rootfs and make conditional choices based on this.
So for instance if ext4 is used and detected it will mount the filesystem with ext4, and if no ext4 is detected fallback to ubifs mount procedure to prevent breaking compatibility with stock firmware.

Currently I still went ahead with SystemD init, unrelated to boot speed but due to the fact I like it's robustness/error handling of services more than sysinitv.
And could make my life a little easier in trying to get the mainline gadget driver working.
But in the meantime I will also look a little more into the bootloader and see if we can re-use some code which I found in the mdm9640 init.c sourcecode.
In this sourcecode it uses ext4 when emmc is used, and ubifs is used when mtd partitions are detected.
So if we can alter the logic slightly, to not detect emmc or mtd, but try and figure out the filesystem in use we can re-use a lot of the conditional logic from here.

PsychoGame, 2021-08-17:
@Biktorgj, I see you are experimenting some with ext4 if I'm correct.
I'm very curious if you got some trickery up your sleeve.
When investigating the possibilities for using other filesystems on these flash mtd partitions I only found support for jffs/yaffs and ubifs to be honest.
I'm still working on this issue as well, and as soon as I got something working properly and reliably I'll share it with you as well.

Biktorgj, 2021-08-17:
I'm doing some experiments, yes :)
I've patched the bootloader to allow flashing non ubi filesystems and added building ext root images in yocto. So far flashing ok, but not correctly mounting the rootfs on boot (I think it is from some issue with the command line because I dirty changed it but not sure

I managed to replicate the lower boot times without fast map too, just by using a smaller ubi filesystem in the recovery partition slot (that gets the 13 seconds boot time). 

Then I got distracted with reading the wakeup, sleep and dtr pins for more power saving :)
Hopefully I'll have something to show in a few days!

Biktorgj, 2021-08-20:
So, time for experiment resutls:

Fastest boot times, by far, is not having a filesystem at all, but booting from a ramdisk. 2.4 Seconds kernel boot time, 4 seconds to usb ready when not booting the ADSP, 12.6 seconds with ADSP firmware.

Problems: I still need to mount the ADSP firmware partition to be able to make it work as a modem, and it can be booted but cannot be flashed.

Ext4: Messes up the ecc blocks in the nand and seems impossible to correctly align it from yocto and ends up crashing
JFFS2: Yocto builds an unflashable image, maybe comes from the bootloader being broken at some points
RAMDISK: Cannot be flashed to the boot slot because it doesn't fit. A full image of kernel+rootfs on ramdisk leaves me with a 13.9Mb image. It can boot (doesn't overlap bootloader kernel jump address), but can't fit the space in flash. I could repurpose another partition for that but don't think it's worth it

Besides the completely messed up command line argument parsing from Quectel, someone changed the spare size of the NAND to "1". I don't know the reason for that, (and I don't know if I want to know it anyway). In any case, I still need to mount the ADSP firmware partition when booting, so even if rootfs is booting by the 2.4 second mark, I still need to mount another partition wasting more time since I can't boot a 45Mb ramdisk with the binary firmware included.

Instead of wasting more time with this, what I think I could to do is try to limit the rootfs filesystem size, so yocto has to deal with less scanning on boot. Modem partition needs 0.3 seconds to attach, and rootfs needs more than double than that. Since total rootfs is about 12Mb and the rest is freespace, if I can let yocto grow the partition only up to 32Mb it would cut that time in half...

PsychoGame, 2021-08-20:
Hello @Biktorgj,

I was doing some experimenting with Ramdisk as wel.
For me I was trying to get poky-tiny up and running instead of the standard poky distro which we currently use.
Also I shrunk the kernel some more by disabling some unneccasary stuff.
But trying to get the firmware in there as well would require a hell of a lot of more free space.

Biktorgj, 2021-09-09:
So, this is the best we can get I think. I removed the serial debugging console from the command line, and now USB comes online at 11 seconds, modem is fully ready at 13.8 seconds-14.2 max.

Since we cannot really remove the 9 second time for the ADSP to load its firmware, boot and become online, there's not much more we can do about it, but I think it's enough:

Please give it a shot: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.3.6

Biktorgj, 2021-09-13:
Closing this now, as we at least have now boot speed parity with stock (+2 seconds from fastboot)

@PsychoGame If you want to play around a bit more with the partitions, I pushed a premade recipe for you in case you want to push the firmware directly inside the rootfs partition: https://github.com/Biktorgj/meta-qcom/tree/hardknott/recipes-proprietary/adsp-firmware (only modem* and mba* files needed, all the rest can go)

If you get something interesting let me know!

closed 2021-09-13 by Biktorgj