the-modem-distro/pinephone_modem_sdk #22: [Ofono] Modem breaks between pre-release 0.26 & 0.27

PsychoGame, 2021-08-01:
Hello @Biktorgj ,

You were hoping for some more user reports on modem usage on Ofono.
Today I flashed Plasma Mobile Development image, with 0.28 modem which was already running on my Phosh install.
To my surprise I found out the modem was in a broken state.
I then went back to the last known working modem (for me), which was 0.23.
I incrementally upgraded modem FW to 0.27, which broke the modem with ofono again.
Went back to version 0.26, which made it working again.
It seems the choice to only listen to service 0x1a works perfectly for Phosh related buids, but breaks everything for Ofono (Tested Plasma Mobile / Sailfish OS [Both Ofono])

Biktorgj, 2021-08-02:
Turns out the first service requested by oFono is 0x02 and in ModemManager is 0x1a.
I'm going to use this to refactor it a bit so it has a bit more order but it's easy to fix

Thanks for the report!

PsychoGame, 2021-08-02:
Very nice to hear that it's quite an easy fix.
I'll be happy to test it out for you on different OS's when it lands.

Biktorgj, 2021-08-04:
Okay, this does _not_ work for me since it doesn't let me place calls or have working data with oFono in postmarketOS Edge KDE, *but*, stock doesn't work either :)

So if you can give it a shot it would be amazing...

At least network coverage shows and remains between suspend and resume events so at least part of it should be working...

[test_ofono.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/6933386/test_ofono.tar.gz)

PsychoGame, 2021-08-05:
Thank you for the test image.
I currenty flashed it, and will do some extensive testing.
First impression is indeed that signal strength/network name and that kind of things are showing again in contrary to the 0.2.7 and 0.2.8 builds.
Haven't had the time yet to make a phone call, but tonight I'll try to make a call with it, and report back about my experience with it.
Currently I'm running Manjaro Plasma Mobile developer image (fully up-to-date).
What I have experienced though is when issueing a usb reset, the modem disappears (No hardware info showing in the Cellular Network -> Modem menu) and needs to be re-initialized by issueing 'systemctl restart ofonoctl'. (lsusb is still showing the modem though)
But this might also be the case for stock FW I believe, I will check that later as well and report back.

Biktorgj, 2021-08-07:
New test build is up containing the fix: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.2.9
Any other issue please let me know

closed 2021-08-07 by Biktorgj