the-modem-distro/pinephone_modem_sdk #23: the newest version 0.2.8 is incompatible with plasma mobile

Idcrafter, 2021-08-06:
it shows up and i even tried downgrading to 0.2.7 and then reinstalling plasma mobile but it only shows this 
[image]

and this 
[image]
and if i call the number the phone screen just turns on

Biktorgj, 2021-08-06:
I know, currently testing some patches for it, see issue #22 
There's a test build in there you might want to flash on top of latest prerelease version to try it out, but so far it's a WIP (I don't loose network info but I can't call in pmOS edge)

closed 2021-08-06 by Biktorgj

PsychoGame, 2021-08-07:
@Biktorgj, just a heads up in case you missed it in my edited report in my issue report #22.
On Plasma Mobile Manjaro I'm able to make and receive calls fine with the test build you provided.
Maybe check that one out as well, at least then you're not chasing a non existing bug in modem FW.
Maybe it's just a problem in pmOS itself?

Idcrafter, 2021-08-12:
the newest release has fixed it thanks