the-modem-distro/pinephone_modem_sdk #30: VoLTE not working on outgoing call

Silent-Hunter, 2021-08-29:
When I make a call for the first time after booting my Pinephone, it's 3G. I hear a regular ringback tone instead of the more boop sounding one. My friend called me back, and it was VoLTE, and then I made another call, and it was VoLTE.

Silent-Hunter, 2021-08-30:
I'm really sorry for all these erroneous bug reports. I needed to enable it with an AT command.

https://wiki.postmarketos.org/wiki/PINE64_PinePhone_(pine64-pinephone)#VoLTE

Silent-Hunter, 2021-08-31:
Actually, on your firmware this solution doesn't work. How do I enable VoLTE?

philmmanjaro, 2021-08-31:
Does it work with the regular stock firmwares? Also does your cell tower support 4G at the location you are at? Sometimes they drop down to 3G for phone calls.

Silent-Hunter, 2021-08-31:
Yes it works on stock firmware, and yes my tower supports it. And if I check modem info, it says it's got the VoLTE-ATT carrier config loaded.

Biktorgj, 2021-09-01:
Hi @Silent-Hunter,

What stock version are you trying in stock? Trying to determine if it's an ADSP firmware change or a problem on my side (I never had to configure anything volte related with stock or custom firmware, everything "just worked")

Could you grab a copy of openqti's log? (adb pull /var/log/openqti.log) so I can take a look?

Silent-Hunter, 2021-09-01:
I couldn't get ADB working, as AT+ADBKEY? always just returned OK and nothing else. Also, and I know I'm dumb for doing this but the Pinephone is now my only phone, so I can't leave less functional firmware on it for too long. I will say, VoLTE works in Mobian, but not PostmarketOS. I can reflash your firmware and try it again if you want, but I never could get ADB working.

Biktorgj, 2021-09-01:
ADB is enabled by default in my firmware, if it doesn't show up you can use "AT+ADBON" to enable it and "AT+ADBOFF" to disable it (no need to unlock anything)
There are special settings for AT&T and Verizon in the modem, but those should be transparent (i.e. nothing to do with all of this) to the userspace unless I'm missing something (which I probably am).

When you say "works on mobian but not postmarketOS" do you mean with stock or with any firmware?

Whenever you have some spare time where you can play around, it'd be great to get some logs of what's going on when you try to call with the firmware so I know where to look

Silent-Hunter, 2021-09-12:
On 0.3.6 of your firmware, VoLTE now seems to work. Although, there's no ringback tone. The official firmware makes this booping noise. It's a minor thing, but it'd be nice to have that back.

Biktorgj, 2021-09-13:
All my tests are primarily in VoLTE too, never heard any noise besides the dial tone, could it be an AT&T thingy or special setting?  Could you record it? 
Stock firmware does have some ring sounds but always assumed there were there in case speakers were connected directly to the modem...

Silent-Hunter, 2021-09-14:
Is it safe to keep flashing back and forth? I don't want to break my modem. But I can try to do that and record it, yes, as long as it won't wear it out too much.
‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

Biktorgj, 2021-09-15:
I must have flashed the entire modem at least 400 times already without issues :)