the-modem-distro/pinephone_modem_sdk #31: No AT+CMUT cmd available

philmmanjaro, 2021-08-31:
With the **stock image** you have the option `AT+CMUT` with switches 0 and 1 to mute the microphone. This workaround is used by Manjaro with this [patch](https://gitlab.manjaro.org/manjaro-arm/packages/community/phosh/calls/-/blob/master/0001-use-at-cmds-to-mute.patch). Currently it seems not to be implemented as shown with this terminal output:

```
echo 'AT+PINE64ROCKS' | sudo atinout - /dev/ttyUSB2 -
[sudo] password for manjaro: 

OK
[manjaro@manjaro-arm ~]$ echo 'AT+CMUT=1' | sudo atinout - /dev/ttyUSB2 -

ERROR
[manjaro@manjaro-arm ~]$ echo 'AT+CMUT=0' | sudo atinout - /dev/ttyUSB2 -

ERROR
```

philmmanjaro, 2021-08-31:
Might be got removed with `EG25GGBR07A08M2G_01.003.01.003`. I tested ADSP 01.001 and 01.002, which seems to work just fine with CMUT cmds. Have to double check with 01.003 full firmware ...

philmmanjaro, 2021-09-02:
Seems this fixes it somehow: https://gitlab.com/mobian1/callaudiod/-/merge_requests/10

philmmanjaro, 2021-09-12:
seems to be readded here: https://github.com/Biktorgj/pinephone_modem_sdk/commit/c168a08a9752c654188c648a64b66717a556fd01

Biktorgj, 2021-09-12:
Hi Philip, I thought you were muting the audio directly from the sunxi mixer via ALSA, do you think it's worth emulating AT+CMUT in the modem side?

philmmanjaro, 2021-09-13:
Well, not everyone is happy about AT cmds. With the 001.02 stock firmware it worked just fine. With 001.03 it seems the stock image lost this function. The **callaudiod** MR10 is not yet perfect and creates a ticking sound on mute and unmute. Will have to test the Plasma Mobile solution again by changing the volume to 0 and saving the previous volume to restore it via pulseaudio again.

So if your firmware can emulate it modem-wise it really depends who will use it then ...

Biktorgj, 2021-09-14:
I'm reimplementing this in userspace, so reopening this issue until I publish the changes 

If nobody uses it it won't do any harm, but it's just a mixer control that needs to be touched after all, and doesn't take much time to implement

Biktorgj, 2021-09-15:
Hi again, 
Just published release 0.3.7 with the re-implemented AT+CMUT command in userspace. What it basically does is disable/enable the capture mixer on demand while there's a call in progress.
https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.3.7

I tested it manually and seems to work as expected, please give it a look if you have some time

philmmanjaro, 2021-09-22:
Ya, it works also here. I just have to see on how to use it again in **callaudiod** for example. Sending AT cmds in a call works. Tested with **v0.4.0** firmware.

Biktorgj, 2021-09-24:
Perfect, if there's something else missing let me know :)

closed 2021-09-24 by Biktorgj