the-modem-distro/pinephone_modem_sdk #35: Can't recover modem after failed update

nwlyoc, 2021-09-19:
After having tried previous versions of this firmware successfully, I tried to flash 0.4.0 today, but the process failed at flashing the system partition (something about the device not existing, unfortunately I don't remember the exact error). MAYBE this happened because I had a systemd timer running that checks every minute the presence of `/dev/ttyUSB2` and restarts eg25-manager if it's not there to bring the modem back up when it disappeared (happened a lot recently). Stupidly I forgot to disable the timer before flashing, so maybe this is related. Since then the modem is gone and I can't get it back. It still shows with `lsusb`, but other than that I can't get it back up. There is no `/dev/ttyUSB2` and ModemManager complains that `/dev/ttyUSB0` can't be handled. What I've tried so far:
- Flashing again: doesn't work. Fastboot is waiting for devices, nothing there apparently.
[the following steps according to [this very helpful post](https://github.com/Biktorgj/quectel_eg25_recovery/issues/4)]
- `echo -ne "AT+QFASTBOOT\r" > /dev/ttyS2`: nothing happens, no changes
- `echo -ne "AT+ADBOFF\r" > /dev/ttyS2`, `echo -ne "AT+ADBON\r" > /dev/ttyS2` and `echo -ne "AT+RESETUSB\r" > /dev/ttyS2`: First time I tried it, `/dev/ttyUSB2` showed up, but other than that nothing worked. The next times I tried it, not even that happened.
- Running [this script](https://github.com/Biktorgj/pinephone_modem_sdk/blob/hardknott/tools/helpers/gotofastboot) from Mobian (I usually use Manjaro Phosh): The script doesn't work (anymore?) since the files it's trying to write to don't exist (`/sys/class/gpio/gpio$n`).
- Shorting the EDL pins and running qfirehose: After I finally managed to short the pins (had to solder a wire to one of them because I just couldn't get it to work otherwise) I ran `./qfirehose -f ./` from the [recovery repo](https://github.com/Biktorgj/quectel_eg25_recovery), but that didn't do anything. The command immediately finished without any output and nothing else happened.

Is there anything else I can try? Or something I misunderstood? It would be really nice to have a working modem again :D 
Also thank you for this amazing work here!

Biktorgj, 2021-09-19:
You were running an older version of this very same firmware, then?

It was a lot easier than that!
1. Flip the modem Killswitch to off
2. On a root terminal, run "fastboot oem stay &"
3. Flip the kill switch to on
4. On that same terminal where you left fastboot running in background, restart the eg25manager service (systemctl restart eg25-manager)
5. Rerun the flashall script

Let me know if it works for you!
Btw, did you manage to find the troubleshooting doc? 
https://github.com/Biktorgj/pinephone_modem_sdk/blob/hardknott/docs/TROUBLESHOOTING.md

nwlyoc, 2021-09-19:
Yeees that worked, thank you so much, also for the fast reply! I was indeed running a previous version of your firmware, but didn't expect it to be that easy. It was just a bit confusing as fastboot outputs to the terminal screen (and thus removes the command prompt) even when it's running in the background, but it worked on the second try.
I did not see the troubleshooting doc (probably would have eventually while trying to get deeper into the matter if you hadn't replied so fast), but that would have saved me some slight panicking :) Maybe consider linking it in the readme after "Having issues?" to make it more visible?
Thanks again!

closed 2021-09-19 by nwlyoc