the-modem-distro/pinephone_modem_sdk #36: [mobian][bookworm] Data connection drops and modem disappears ocassionally

fernando-c-m, 2021-10-01:
Hello,
First of all a big thank you for your hard work providing us with an open source userspace for the pinephone modem.
I am facing some problems with r0.4.1, mobian, bookworm stable and 5.14 kernel
- Data disconnects and won't come back
- Modem disappears (but with 5.14 seems to be coming back after a while)
Then everything is OK.
I am aware that this could be also a modemmanager or networkmanager problem, but I need your expertise to  assess that.
All in all with 5.14 the whole system behavior seems to be better than with 5.13 but I have been testing this only for some hours until now. : )
Let me know if you need any additional information.
[modem.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7265241/modem.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7265252/openqti.log)
[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7265255/dmesg.txt)

Biktorgj, 2021-10-04:
Hi,
This is the last remaining reset event that we're tracking down, it's still not fully known if the issue is on the modem side, the Pinephone side or both. 

In any case, ModemManager should reconnect afterwards, but I see the connection getting stuck. Did you flash latest ADSP firmware? It's linked in the release but not bundled in the installation package to avoid mixing closed source stuff with the rest: https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi

That's exactly one of the things it fixes... you can check if you have it already installed by sending the following AT command to the Modem: AT+QGMR

fernando-c-m, 2021-10-04:
Ok. I am getting as version 01.003.00.000
This is  what I (re)did:
`root@mobian:/home/mobian# wget https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi
--2021-10-04 09:07:51--  https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/Biktorgj/quectel_eg25_recovery/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi [following]
--2021-10-04 09:07:52--  https://raw.githubusercontent.com/Biktorgj/quectel_eg25_recovery/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.109.133, 185.199.111.133, 185.199.108.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 41025536 (39M) [application/octet-stream]
Saving to: ‘NON-HLOS.ubi’

NON-HLOS.ubi                           100%[==========================================================================>]  39.12M  9.59MB/s    in 4.5s    

2021-10-04 09:07:56 (8.78 MB/s) - ‘NON-HLOS.ubi’ saved [41025536/41025536]

root@mobian:/home/mobian# echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2
root@mobian:/home/mobian# lsusb
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 015: ID 18d1:d00d Google Inc. Xiaomi Mi/Redmi 2 (fastboot)
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@mobian:/home/mobian# fastboot flash modem NON-HLOS.ubi 
Sending 'modem' (40064 KB)                         OKAY [  1.343s]
Writing 'modem'                                    OKAY [ 12.337s]
Finished. Total time: 13.833s
root@mobian:/home/mobian# fastboot reboot
Rebooting                                          OKAY [  0.004s]
`
but after this  I still get:
`AT+QGMR

EG25GGBR07A08M2G_01.003.00.000

OK
`
There is some step which I am missing?

LD1DL, 2021-10-06:
Hello, 
Thank you for the great job accomplished.
The data connection drop seemed to come with revision 0.4.1 of the firmware. It's not Mobian specific, I get the same with Manjaro.
The stability of the modem was greatly improved with the revisions of last month but sometimes the modem still disappear and doesn't come back. Then the result of "lsusb" tells it went in fastboot mode (Google Inc. Xiaomi Mi/Redmi 2 (fastboot)) but it doesn't react to command "fastboot reboot" and I have to reboot the phone. Is it the last remaining reset event that you are tracking down or I have another issue? ADSP version: 01.003.01.003.
BR

pizdjuk, 2021-10-30:
Just another hang: the modem doesnt appear on the USB bus, the bus number increases iteratively: 2,3,....,127,2,3,4,...,127,2,3.....

[11230.616030] usb 4-1: device not accepting address 14, error -62
[11230.808065] usb 4-1: new full-speed USB device number 15 using ohci-platform
[11231.224044] usb 4-1: device not accepting address 15, error -62
[11231.230157] usb usb4-port1: unable to enumerate USB device
[11232.612115] usb 4-1: new full-speed USB device number 16 using ohci-platform
[11232.804042] usb 4-1: device descriptor read/64, error -62
[11233.108030] usb 4-1: device descriptor read/64, error -62
[11233.408037] usb 4-1: new full-speed USB device number 17 using ohci-platform
[11233.600054] usb 4-1: device descriptor read/64, error -62
[11233.904055] usb 4-1: device descriptor read/64, error -62
[11234.016131] usb usb4-port1: attempt power cycle
[11234.232029] usb 4-1: new full-speed USB device number 18 using ohci-platform
[11234.648239] usb 4-1: device not accepting address 18, error -62
[11234.840068] usb 4-1: new full-speed USB device number 19 using ohci-platform
[11235.256039] usb 4-1: device not accepting address 19, error -62
[11235.262145] usb usb4-port1: unable to enumerate USB device
[11235.544065] usb 4-1: new full-speed USB device number 20 using ohci-platform
[11235.736033] usb 4-1: device descriptor read/64, error -62
[11236.040524] usb 4-1: device descriptor read/64, error -62
[11236.340479] usb 4-1: new full-speed USB device number 21 using ohci-platform
[11236.532113] usb 4-1: device descriptor read/64, error -62

B0bbyTheBuilder, 2021-10-31:
Hello @Biktorgj

I recently took a leap of faith and flashed my daily driver, a PinePhone pmOS CE edition running latest Edge version of postmarketOS.
So far, so good. I haven't had any issues with Calls, SMS nor MMS.

However I'm experiencing data issue where the network interface wwan0 end up loosing it's IP after a while:
```
wwan0     Link encap:Ethernet  HWaddr FE:27:7D:XX:XX:XX  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:12 errors:0 dropped:0 overruns:0 frame:0
          TX packets:317 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:962 (962.0 B)  TX bytes:100806 (98.4 KiB)
```
I have the latest ADSP:
```
AT+QGMR
EG25GGBR07A08M2G_01.003.01.003

OK
```
And I used the flashall script from version 0.4.2 to flash your firmware.
The only command that produced an error was this one:

`fastboot oem stay`

I don't have the exact error but I recall it said something like unknown command and it did proceed with the rest of the script.

So, yesterday evening I restarted the phone and let it sit on the charger overnight. I checked I had data connection right after restarting.
This morning the IP was gone. I was able to call and send SMS though.

I've added the modem's logs here:

[dmesg.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7449004/dmesg.txt)
[messages.0.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7449005/messages.0.txt)
[messages.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7449006/messages.txt)
[openqti.log.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7449007/openqti.log.txt)

Thanks for taking the time to look into it and for the amazing work you and other involved are doing here.

LD1DL, 2021-11-10:
Hello all,

I did the test to come back to firmware version 0.4.0 and I confirm data drop is a problem with firmware versions > 0.4.0. With version 0.4.0 the data connection is stable (thought the modem disappears more frequently and comes back after a few tens of seconds). With v. 0.4.2 the modem is more stable but data drops frequently and I have to reboot the modem to get data back.

BR

eugenr, 2021-11-21:
Hi,

I'm getting this issue also with 0.4.5 @Biktorgj's modem firmware.
- Pinephone Mobian Edition
- upgraded  to mobian/bookworm testing
- Linux kernel 5.13-sunxi64
- ModemManager 1.18.2-0.2
- AT+QGMR:    EG25GGBR07A08M2G_01.003.00.000

Phosh announced (with the icon) the data connection lost at around 'Nov 21 15:51' time. This log time is UTC+2:
[pp-journalctl-e.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576622/pp-journalctl-e.log)

Two hours later I managed to also get all the modem's logs. All this time I've prevented the phone to suspend.
[modem-boot.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576632/modem-boot.log)
[modem-dmesg.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576633/modem-dmesg.log)
[modem-messages.0.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576634/modem-messages.0.log)
[modem-messages.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576635/modem-messages.log)
[modem-thermal.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576636/modem-thermal.log)

[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7576642/openqti.log)
I've truncated this one. It contained further only lines with "[..]USB is suspended." making the original file at 78MB.

eugenr, 2021-11-22:
On 0.4.5 data connection got lost as a side effect of debug messages flooding openqti.log file until the tmpfs was left with no space anymore.

Those debug messages were removed in 0.4.5-2, and using this version I could not reproduce this issue anymore. Yay! It works great!
Thanks!

B0bbyTheBuilder, 2021-11-24:
I second @eugenr , 0.4.5-2 works like a charm. I haven't lost data in more than 48 hours :D

fernando-c-m, 2021-11-25:
0.4.5-2 installled on Sunday
Today it happened again.
See attached logs
[dmesg_modem.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7601380/dmesg_modem.txt)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7601381/openqti.log)
[dmesg_211125_mobian.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/7601382/dmesg_211125_mobian.txt)
:

tiol11, 2022-01-23:
Hi,
I am experiencing a similar issue. Modem does not appear in lsusb answer. There is no answer to `sudo sh -c 'echo -ne "AT+ADBON\r" > /dev/ttyUSB2'`. Logs which I was able to extract are attached.
Modem gets back online by switching off its kill switch, then switch back on and `sudo systemctl restart eg25-manager`. Connection is back but modem does not answer AT commands (no AT port available to run command).
Phone is PinePhone 3GB with up-to-date Mobian Bookworm,
kernel 5.15-sunxi64 from Mobian repo.
Network manager 1.34.0-1 from Debian repo.
Modem firmware 0.5.3.
ADSP version 01.003.01.003 (note 7 in [https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/AT_INTERFACE.md](url) suggests the answer to "AT+QGMR" is the fw version instead of the ADSP, but ADSP section in [https://github.com/Biktorgj/pinephone_modem_sdk/releases](url) release announcements tells the opposite)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7921487/networkmanager.log)
[ModemManager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7921488/ModemManager.log)
[eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7921489/eg25-manager.log)
[dmesg_pinephone.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7921490/dmesg_pinephone.log)

Hope this helps in tracing this issue, thank you for maintaining this firmware.

tiol11, 2022-02-05:
Upgraded modem FW to 0.5.5 last weekend, had only one major issue throughout the whole week using my Pinephone as a daily driver. When this major issue happened, whole Pinephone was behaving strangely, then I am not even sure that it was a modem FW related issue: I will try to reproduce this and then open a dedicated issue in the future with more details.
Modem stability is really good with 0.5.5, thank you!

fernando-c-m, 2022-04-07:
This has been so greatly mitigated, that the issue can now be closed. Thank you for your hard work!
Current setup is:
kernel: 5.15.32-sunxi64
pinephone modem sdk: 0.6.1
modem FW ADSP: EG25GGBR07A08M2G_01.003.00.000
network-manager: 1.36.4-1
modemmanager: 1.18.6-2mobian1
eg25-manager 0.4.3.1

closed 2022-04-07 by fernando-c-m