the-modem-distro/pinephone_modem_sdk #37: Cannot make phone calls after update to 0.4.2

arkeo-ca, 2021-10-06:
I upgraded this afternoon to 0.4.2 from 0.3.0, and upgraded my ADSP firmware as suggested. Since this update, I cannot make or receive phone calls. Data and SMS work flawlessly.

I am using Arch with all packages up to date. 

Whenever I try to place a phone call, the call window appears for a split second and then disappears. The call is logged in the call log from gnome-calls, but nothing happens. The following extract from journalctl shows 2 attempted outgoing phone calls and an attempted incoming phone call.

```
Oct 06 15:53:41 danctnix ModemManager[3080]: <info>  [modem0/call2] user request to start call
Oct 06 15:53:41 danctnix ModemManager[3080]: <info>  [modem0/call2] call state changed: unknown -> dialing (outgoing-started)
Oct 06 15:53:41 danctnix gsd-media-keys[3331]: Unable to get default sink
Oct 06 15:53:41 danctnix gsd-media-keys[3331]: Unable to get default source
Oct 06 15:53:41 danctnix ModemManager[3080]: <warn>  [modem0/call2] couldn't start call: Couldn't create call: QMI protocol error (48): 'InvalidArgument'
Oct 06 15:53:41 danctnix ModemManager[3080]: <info>  [modem0/call2] call state changed: dialing -> terminated (unknown)
Oct 06 15:53:41 danctnix gnome-calls[3495]: Error starting outgoing call ModemManager call to `PHONENUMBER': GDBus.Error:org.freedesktop.libqmi.Error.Protocol.InvalidArgument: Couldn't create call: QMI protocol error (48): 'InvalidArgument'
Oct 06 15:53:41 danctnix gsd-media-keys[3331]: Unable to get default sink
Oct 06 15:53:41 danctnix gsd-media-keys[3331]: Unable to get default source
Oct 06 15:54:12 danctnix ModemManager[3080]: <info>  [modem0/call3] user request to start call
Oct 06 15:54:12 danctnix ModemManager[3080]: <info>  [modem0/call3] call state changed: unknown -> dialing (outgoing-started)
Oct 06 15:54:12 danctnix gsd-media-keys[3331]: Unable to get default sink
Oct 06 15:54:12 danctnix gsd-media-keys[3331]: Unable to get default source
Oct 06 15:54:12 danctnix ModemManager[3080]: <warn>  [modem0/call3] couldn't start call: Couldn't create call: QMI protocol error (48): 'InvalidArgument'
Oct 06 15:54:12 danctnix ModemManager[3080]: <info>  [modem0/call3] call state changed: dialing -> terminated (unknown)
Oct 06 15:54:12 danctnix gnome-calls[3495]: Error starting outgoing call ModemManager call to `PHONENUMBER': GDBus.Error:org.freedesktop.libqmi.Error.Protocol.InvalidArgument: Couldn't create call: QMI protocol error (48): 'InvalidArgument'
Oct 06 15:54:12 danctnix gsd-media-keys[3331]: Unable to get default sink
Oct 06 15:54:12 danctnix gsd-media-keys[3331]: Unable to get default source
Oct 06 15:54:27 danctnix ModemManager[3080]: <warn>  [modem0] network reject indication received
Oct 06 15:54:27 danctnix ModemManager[3080]: <warn>  [modem0]   service domain: cs
Oct 06 15:54:27 danctnix ModemManager[3080]: <warn>  [modem0]   radio interface: umts
Oct 06 15:54:27 danctnix ModemManager[3080]: <warn>  [modem0]   reject cause: network-failure
```

Biktorgj, 2021-10-07:
Hi!

First time I see this :(
You can use whatever ADSP firmware version you need  with this firmware, the reason to recommend v003 is because it recovers better when a data session is unexpectedly closed or orphaned. Never seen calls rejected by that change. Out of curiosity, could you tell me the name of your carrier? And does it do the same when running in LTE or 3G?

arkeo-ca, 2021-10-07:
I am with Public Mobile, a budget division of Telus (Canada). As far as I understand, everything is LTE, other than phone calls, which seem to downgrade to 3G while I make them.
I haven't been able to get VoLTE to work (haven't really tried to be honest).

nesnera, 2021-10-09:
I seem to have encountered [the same problem](https://gitlab.manjaro.org/manjaro-arm/issues/pinephone/phosh/-/issues/205) (note that my carrier has discontinued 3G support).  I'll add a few more extracts from the journalctl (EG25GGBR07A08M2G_01.003.00.000)
- <details>
  <summary>Network mode 4G Only > 3G, 4G (Preffered)</summary>
  
  ```
  říj 09 11:38:08 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] verbose call end reason (3,1028): [cm] emm-detached
  říj 09 11:38:08 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (connected -> registered)
  říj 09 11:38:08 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] connection #5 finished: duration 389s, tx: 732 bytes, rx: 974 bytes
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect started...
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (5/8): register
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (6/8): bearer
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (7/8): connect
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (registered -> connecting)
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] couldn't start network: QMI protocol error (14): 'CallFailed'
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] verbose call end reason (3,2001): [cm] no-service
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] couldn't start network: QMI protocol error (14): 'CallFailed'
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] verbose call end reason (3,2001): [cm] no-service
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <warn>  [modem0/bearer1] connection attempt #6 failed: Call failed: cm error: no-service
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (connecting -> registered)
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer1] connection #6 finished: duration 0s, tx: 0 bytes, rx: 0 bytes
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect started...
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (5/8): register
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (6/8): bearer
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (7/8): connect
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (registered -> connecting)
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer2] couldn't start network: QMI protocol error (14): 'CallFailed'
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer2] verbose call end reason (3,2001): [cm] no-service
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <warn>  [modem0/bearer2] connection attempt #2 failed: Call failed: cm error: no-service
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (connecting -> registered)
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0/bearer2] connection #2 finished: duration 0s, tx: 0 bytes, rx: 0 bytes
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect started...
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (5/8): register
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (6/8): bearer
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (7/8): connect
  říj 09 11:38:09 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (registered -> connecting)
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0/bearer3] couldn't start network: QMI protocol error (14): 'CallFailed'
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0/bearer3] verbose call end reason (2,210): [internal] pdn-ipv6-call-disallowed
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <warn>  [modem0/bearer3] connection attempt #2 failed: Call failed: internal error: pdn-ipv6-call-disallowed
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (connecting -> registered)
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0/bearer3] connection #2 finished: duration 0s, tx: 0 bytes, rx: 0 bytes
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect started...
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (4/8): wait to get fully enabled
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (5/8): register
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (6/8): bearer
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] simple connect state (7/8): connect
  říj 09 11:38:10 manjaro-arm ModemManager[428]: <info>  [modem0] state changed (registered -> connecting)
[... cut ...]
  říj 09 12:10:08 manjaro-arm ModemManager[428]: <info>  [modem1/call5] user request to start call
  říj 09 12:10:08 manjaro-arm ModemManager[428]: <info>  [modem1/call5] call state changed: unknown -> dialing (outgoing-started)
  říj 09 12:10:09 manjaro-arm ModemManager[428]: <warn>  [modem1/call5] couldn't start call: Couldn't create call: QMI protocol error (13): 'NoNetworkFound'
  říj 09 12:10:09 manjaro-arm ModemManager[428]: <info>  [modem1/call5] call state changed: dialing -> terminated (unknown)
  ```
  </details>
(Let me ask. Any idea why I can't install firmware 0.4.2? 🥴)

nesnera, 2021-10-12:
In my experience the ADSP fm version has no effect.
But I accidentally switched the mode to "2G, 3G, 4G (Preferred)" and both calls and SMS (Manjaro/Phosh) work in that mode. The "2G, 3G, 4G" mode cannot be activated - it returns the error "Error: GDBus.Error:org.freedesktop.ModemManager1.Error.Core.Unsupported: The given combination of allowed and preferred modes is not supported"
Carrier: Vodafone Czech Republic

Anyway, I would very much like to be able to easily find out which version of custom firmware is installed. I will create a ticket.

nesnera, 2022-01-03:
With the settings [described earlier](https://github.com/Biktorgj/pinephone_modem_sdk/issues/37#issuecomment-940817689), my calls work and the modem is much more stable 🎉 (I have been using PP intensively as a WiFi router for more than a month to connect to the internet).
From my point of view this issue can be closed.

closed 2022-03-25 by Biktorgj