the-modem-distro/pinephone_modem_sdk #42: boot custom firmware with kexec

ollieparanoid, 2021-11-03:
I've been looking for a way to boot into the custom firmware without modifying the stock firmware.

### Motivation

This would allow to:
* avoid the risk of potentially (soft) bricking the modem while flashing the custom lk
* tie the modem firmware version to the currently booted distribution, making it deterministic
* provide a nice unified way for users to use this firmware, that works accross distros. I would put the logic in eg25-manager (if that's fine with @a-wai et al.), which is widely used already among distros.
* have upgrading the modem firmware as easy as installing a new package and rebooting/restarting eg25-manager

### Research

To my surprise, it seems that the kernel of the stock firmware was built with CONFIG_KEXEC.

<details>

```
/ # echo 0 > /proc/sys/kernel/dmesg_restrict 
/ # echo 0 > /proc/sys/kernel/kptr_restrict
/ # cat /proc/kallsyms | grep kexec
c0035c48 W compat_sys_kexec_load
c0035c48 W sys_kexec_file_load
c0035c48 W sys_kexec_load
c0a62c44 t setup_crash_kexec_post_notifiers
c0aa01e2 t __setup_str_setup_crash_kexec_post_notifiers
c0aa2b64 t __setup_setup_crash_kexec_post_notifiers
c0b357f4 b crash_kexec_post_notifiers
```
</details>

rw tmpfs mounts:

<details>

```
/ # mount
rootfs on / type rootfs (rw)
ubi0:rootfs on / type ubifs (ro,relatime,bulk_read,chk_data_crc)
ubi1:modem on /firmware type ubifs (ro,relatime,bulk_read,chk_data_crc)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime,size=64k,mode=755)
devpts on /dev/pts type devpts (rw,relatime,gid=5,mode=620)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /var/volatile type tmpfs (rw,relatime)
tmpfs on /var/lib type tmpfs (rw,relatime)
/dev/ubi2_0 on /cache type ubifs (rw,relatime,bulk_read,chk_data_crc)
/dev/ubi2_0 on /data type ubifs (rw,relatime,bulk_read,chk_data_crc)
none on /sys/kernel/config type configfs (rw,relatime)
adb on /dev/usb-ffs/adb type functionfs (rw,relatime)
```
</details>

\> 70 MiB of RAM available in tmpfs:

<details>

```
/ # df -h
Filesystem                Size      Used Available Use% Mounted on
ubi0:rootfs              57.9M     51.7M      6.2M  89% /
ubi1:modem               41.1M     33.7M      7.4M  82% /firmware
tmpfs                    64.0K      4.0K     60.0K   6% /dev
tmpfs                    77.8M     28.0K     77.8M   0% /run
tmpfs                    77.8M    112.0K     77.7M   0% /var/volatile
tmpfs                    77.8M    112.0K     77.7M   0% /var/lib
/dev/ubi2_0              48.3M      1.1M     47.2M   2% /cache
/dev/ubi2_0              48.3M      1.1M     47.2M   2% /data
```
</details>

### Concept

So I'm assuming it would work like this:
* automatize getting adb shell access via AT command
* do sanity checks via AT / adb shell (i.e. did we boot into the custom fw already?)
* `adb push` files to `/var/lib` tmpfs:
  * kexec static binary
  * new kernel
  * initramfs file
    * which contains all of the custom firmware userspace code
    * according to @megous this should be doable in [200 KiB](https://xnux.eu/log/#028)? Given that we have > 70 MiB in tmpfs, this seems feasible even if it is a lot bigger.
* shutdown running userspace services via `adb shell`
* use kexec to boot into custom kernel + initramfs
* wait until it is booted
* done, modem can be used now

A potential problem could be, that kexec boot itself takes a long time. Personally I think the advantages would be worth it, even if this means that it takes 20s or so extra after boot until the modem is fully functional. That's just a random guess, I didn't try this concept out yet.

### Next steps

The next step to trying this out is, creating a minimal initramfs that allows communication via serial and building the userspace kexec program statically. To check whether kexec works at all, and to test how long this boot method takes. From there on, we could look into building this awesome firmware into such an initramfs and then potentially automate the whole setup and put it into eg25-manager.

I'll probably not continue investigating this any time soon due to other tasks, but I wanted to write the idea down. If somebody wants to evaluate if this is feasible, please go ahead.

### Thoughts?

What do you think, does it sound feasible or am I overlooking something?

Biktorgj, 2021-11-03:
Not sure if that wouldn't bring even more issues than what we already have :)

The worst problem in stock (apart from the stability issues) is that if power unexpectedly goes out sometimes the entire filesystem might get corrupted. If that happens you'd loose both firmwares. I also don't know how bad can the ADSP firmware will go if it's already booted and you kill all the daemons connected to it

I can imagine an alternate solution thought, which is something I do a lot while testing different things in firmware: use the recovery partitions as an alternate firmware. 

In stock, the recovery firmware is used to flash FOTA updates, but those need to be pushed by the user (Quectel doesn't have an update server publicly available). Since this firmware needs about 12mb of flash space it fits there perfectly fine

There's already a way in the custom firmware to trigger the modem into fastboot via GPIO. We could flash the OpenSource firmware into the recovery slots and allow eg25-manager to trigger booting one firmware or the other via another GPIO. Though you'd still have to flash a different bootloader, which I guess was like the whole reason to look into kexec :)

ollieparanoid, 2021-11-06:
> Though you'd still have to flash a different bootloader, which I guess was like the whole reason to look into kexec :)

Yes :smile:  Thanks a lot for the detailed reply.

I've been brainstorming with @dylanvanassche based on that and we'll try to hook this firmware up with fwupd probably, to find another solution that should work across distros but doesn't require re-inventing the wheel as much. Without kexec boot etc., just regular flashing.

closed 2021-11-06 by ollieparanoid