the-modem-distro/pinephone_modem_sdk #43: [Calling] No dial-tone in phone call

PsychoGame, 2021-11-16:
Hello @Biktorgj,
<p>
On Manjaro ARM Phosh Developer image (updated daily) there's an issue of not having a dial-tone when placing a phone call.<br>
I don't know if this issue is happening on other distro's as well, I suspect so because of this bug report [Callaudio Issue 17]https://gitlab.com/mobian1/callaudiod/-/issues/17)<br>
Firmware version which I was using is v0.42.<br>
When restoring stock firmware the dial-tone appears on all placed phone calls again.<br>
I tried this 3 times, to make sure it was no coincidence, but really a issue with the open firmware<br>
What can I do to help you out in trying to figure out what might cause this issue<br>
<p>
Greeings Jasper

PsychoGame, 2021-11-21:
I see a new update just came out.<br>
I'll retest things with this new modem fimware to see if this situation changes, if not I'll report back.<br>

Biktorgj, 2021-11-24:
Can't reproduce it on my side :(
I've tried with Mobian bookworm, Danct's Arch and pmOS Edge Phosh (this last one without updating from the https://images.postmarketos.org/bpo/edge/pine64-pinephone/phosh/20211117-0505/20211117-0505-postmarketOS-edge-phosh-12-pine64-pinephone-installer.img.xz image) and when I attempt to do a call first ring sounds perfectly fine.

Can you describe a bit more the issue? First of all, does it happen in 2G/3G only? or VoLTE calls are affected too? My carrier forces LTE for calling by default

You can also sort-of measure the time the modem reports the incoming or outgoing call with openqti, `adb shell tail -f /var/log/openqti.log` should allow to see exactly when the call actually starts or the first ring in comes. Might be nice if you can attach it too?

PsychoGame, 2021-11-24:
That's unfortunate and lucky at the same time :P, means that it's problem is not with all users.
For me I haven't tried other distributions myself, only Manjaro, in the 2 different flavours (Phosh & Plasma).
I'll try to describe what's happening as detailed as possible, and later on during the day will try to pick some logs as well.
The problem might actually lie in the switchover from 4G towards 3G.
When my phone is idle connected to the network, it shows that it's having a 4G signal.
Although I'm connected to 4G, it always switches back to 3G whenever I'm making a phonecall (in/out), so I can't provide you on any details about this problem on 4G profiles.
When calling out (from my phone to another phone), as soon as hitting the "Dial" button in Calls it switches over from 4G to 3G. Then it stays silent for as long as it takes for the other end to pick-up and I start to hear the voice conversation.
It doesn't matter if I let the other end ring or not, in my case it doesn't look like it's a delayed ringer-tone, as it simply doesn't appear, also not after 15+ seconds.

I have a phone plan with a Dutch telcom provider called Simpel.
This provider doesn't have it's own network, but makes use of the network of one of the bigger providers in the Netherlands.
In my case this is T-Mobile.
I'll try to provide you with the logs you would like to have later on during the day:)

Biktorgj, 2021-11-24:
So the issue is probably in the call downgrade and retry, then when it's established it comes back... Makes sense. Whenever you have some time, do a test call (best would be one outgoing and one incoming) and attach OpenQTI.log so I can check which call states are reached on each case

Thanks!

nwlyoc, 2021-11-24:
I have the same issue (never had a dial tone with the custom firmware) with the very same symptoms, but my calls work on 4G (at least the 4G icon doesn't change during calls). I will also attach a log when I get to it.

Biktorgj, 2021-11-25:
> I have the same issue (never had a dial tone with the custom firmware) with the very same symptoms, but my calls work on 4G (at least the 4G icon doesn't change during calls). I will also attach a log when I get to it.
> 
> Edit: I now attached logs. I first called a number, the other person let it ring for a bit and then picked up. Then that person called me back and I picked up. There was no dial tone on my side while the other person's phone rang. I'm using Manjaro Phosh (dev), but also tried the recent weekly of Mobian Phosh, it's the same there. [openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7595375/openqti.log)

It's interesting that the log starts with an unknow call type:
[620.7013] W handle_call_pkt: Call direction: outgoing 
[620.7019] E handle_call_pkt: Unknown call type 
[620.7024] I handle_call_pkt: Setting up audio for mode 1 
[621.7152] I handle_call_pkt: Dir: 0x01 Sta: 0x04 Typ: 0x06, Mode: 0x01 
[621.7182] W handle_call_pkt: Call direction: outgoing 
[621.7187] I handle_call_pkt: Call type: VoLTE 
[621.7192] I handle_call_pkt: Setting up audio for mode 2 

Are you Manjaro too? Can any of you try if it does the same with other distros? I'm still up for another 12 hours of postmarketOS testing, so cannot reboot right now, but will do so tomorrow. If you can post here a link to the image you're running it would be great, as to check things as close as what you're running

PsychoGame, 2021-11-25:
If I read correct @nwlyoc has also tried on one of the Mobian weeklies.
I'm using Manjaro Dev image, today downloaded the daily dev build of 25-11-2011.
For some reason though it looks like the bug has solved on my case.
The only thing I've done is running a custom kernel which is very closely based om Megi's default configuration on his PinePhone.
For completeness I'll include the logs, but I doubt if you can find any pointers in there at the moment.
[openqti.log](https://pastebin.com/G9seSLip)
If you like I can also share the kernel build which I'm running, so you can test as well if default Manjaro/ my custom kernel show any differences.

Biktorgj, 2021-11-25:
A little more investigation has lead me to belive the problem comes with the network, not with the modem or the distro.

Some carriers on some configurations dont send a ring tone back when there's an outgoing call and the call state is in "alerting" state (the other side is already ringing). In that case, stock enters a fake tone through the soundcard. So, if what I'm looking at is correct, I have to track another packet type for audio to check if there's a generated tone on the carrier side, and if there's not, inject it myself.
Since I can't reproduce it myself and have an ongoing test right now, I'll do a few checks on my side to try to track down the exact data I need, but I might need some logs on your side too to compare against mine (will need to have openqti in debug mode and that dumps the numbers you're calling, so this will not be the best place to attach those logs).

Will write back when I can do some testing on my side

nwlyoc, 2021-11-25:
Yes, I'm on Manjaro with the dev image from [November 5th](https://github.com/manjaro-pinephone/phosh-dev/releases/tag/202111050248) and the latest updates. I also attached the log from an outgoing call on Mobian for which I used the weekly image from November 21st. That call was not picked up. I didn't hear a dial tone like on Manjaro. The unknown call type message appeared on Mobian as well. I'm on the German Telekom network if that's of any interest (through another smaller carrier though).
[openqti_mobian.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/7605033/openqti_mobian.log)

Let us know if you need more logs! I can also just strip the phone numbers, that shouldn't be the problem.

PsychoGame, 2021-11-25:
@nwlyoc, If you like I can share you my Kernel build as well tomorrow, since you are using Manjaro as well.
Then we can make sure  it's not a kernel configuration issue.
The kernel which I'm running is a kernel configuration which Megi uses on his devices as well, adapted a littlebit to Manjaro, and have all modules to run the basic phone included in the kernel (so no modules for the phone specific part).
This makes it a highly optimized kernel to run on the phone.
Currently I'm testing the hell out of it, and hopefully Manjaro will allow me to upstream it if it confirms to their standard.
Just let me know if you're interested.

nwlyoc, 2021-11-25:
Sounds good, thanks, I'll happily try it! Is there any documentation on the differences regarding the PinePhone compared to Manjaro's current kernel?

PsychoGame, 2021-11-26:
Hello @nwlyoc, hereby I provide you with a share-link to my nextcloud instance;
[kernel-5.15.5-megi](https://cloud.dekroon.duckdns.org/s/eSrpFnGnkd4TMnr)
If no dynamically built kernel modules are used, you only need to download the linux-pinephone-5.15.5-2-aarch64.pkg.tar.zst file.
To install it just execute command `pacman -U linux-pinephone-5.15.5-2-aarch64.pkg.tar.zst` inside the folder where you downloaded the package.
To go back to stock is basically sudo pacman -S linux-pinephone
There is no official documentation on my kernel, but I try to explain the difference as detailed as possible over here;
Included on my Nextcloud instance is the kernel config for my running kernel called; megi-config
The kernel is build according to PKGBUILD which came from the latest linux-pinephone (Manjaro stock kernel), to be found over here;
[linux-pinephone](https://gitlab.manjaro.org/manjaro-arm/packages/core/linux-pinephone/-/tree/5.15-megi)
The kernel configuration which I'm using is mainly based on Megi's own running kernel, and can be found on his website;
[pp.tar.gz](https://xff.cz/kernels/5.15/pp.tar.gz)
Above is a link to his prebuilt kernel, which includes his personal kernel config called: linux.config
Difference between his, and my kernel build is that all peripherals which are normally loaded as kernel modules during boot-up, are built-in in my kernel build.
This means that upon loading the kernel everything is instantly initiated, so no waiting for kernel loading at boot-up.
All other modules which give extra functions to the PinePhone and it's OS are still build as modules.
P.S.: Sorry @Biktorgj for this off-topic. If you'll try Manjaro yourself feel free to test this kernel build as well.

nwlyoc, 2021-11-26:
Thanks a lot for the build and the explanation! I just installed it, but behaviour regarding the dial tone is the same as before (i. e. still nothing). Also apart from that I haven't noticed anything else except that it took some time to connect to WiFi upon the first boot with the new kernel, but that could have just been a coincidence (I'm very happy that the sometimes very long times it took to reconnect to WiFi after suspend have been fixed recently, so far this seems to be the case with your build as well).

Biktorgj, 2021-12-27:
Fixed it in 0.5.0
https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.5.0
You'll have to enable custom alert tones using the AT+EN_CAT command (disable it with AT+DIS_CAT)
Closing the issue now, if you have some issue feel free to reopen it!

closed 2021-12-27 by Biktorgj

PsychoGame, 2021-12-28:
Hello @Biktorgj, The default behaviour is off right?

Biktorgj, 2021-12-28:
Yes, default is whatever the carrier sends (if any). You can enable the "local" mode by issuing AT+EN_CAT, and it should be persistent across reboots