the-modem-distro/pinephone_modem_sdk #44: Writing boot failed

feklee, 2021-11-26:
See error messages on last lines below. The modem is gone now, even after a reboot. *How can I recover from this?*

 1. Checked version of ADSP firmware:

        $ mmcli -m any --command="AT+QGMR"
        response: 'EG25GGBR07A08M2G_01.001.01.001'

 2. Upgraded to ADSP version 01.003.01.003:

        # cd /tmp
        # wget https://github.com/Biktorgj/quectel_eg25_recovery/raw/EG25GGBR07A08M2G_01.003.01.003/update/NON-HLOS.ubi
        # echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2
        # fastboot flash modem NON-HLOS.ubi && fastboot reboot

 3. Flashed the Biktorgj firmware:

        $ cd ~/src
        $ wget https://github.com/Biktorgj/pinephone_modem_sdk/releases/download/0.4.2/package.tar.gz
        $ mkdir firmware
        $ tar xzf package.tar.gz -C firmware
        $ cd firmware
        $ sudo ./flashall
        Sending AT+QFASTBOOT...
        FAILED (remote: 'unknown command')
        fastboot: error: Command failed
        Sending 'aboot' (486 KB)                           OKAY [  0.024s]
        Writing 'aboot'                                    OKAY [  0.209s]
        Finished. Total time: 0.244s
        Rebooting                                          OKAY [  0.018s]
        Finished. Total time: 0.068s
        < waiting for any device >
        OKAY [  0.002s]
        Finished. Total time: 0.003s
        Sending 'boot' (4566 KB)                           OKAY [  0.150s]
        Writing 'boot'                                     FAILED (Status read failed (No such device))
        fastboot: error: Command failed
        ./flashall: exit: line 11: Illegal number: Failed to flash main kernel

Biktorgj, 2021-11-26:
Take a look at the recovery section in troubleshooting:
https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/TROUBLESHOOTING.md#my-modem-wont-boot-anymore

Since the bootloader was correctly written I think something might have killed the modem before finishing the update. Try stopping ModemManager and then try to flash again

feklee, 2021-11-26:
*Wow, thanks for the super fast reply!* Maybe the link to the troubleshooting section could be added to the [installation  instructions][1]? Something like: "If the modem won't boot anymore, don't panic, and follow the [instructions for recovery][2]."

I'm on postmarketOS (SXMO), and to restart the eg25-manager service, I did:

    rc-service eg25-manager restart

Flashing went smoothly on this second attempt.

[1]: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.4.5-2
[2]: https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/TROUBLESHOOTING.md#my-modem-wont-boot-anymore

Biktorgj, 2021-11-28:
Included in the latest test release already (https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.4.7), will stay there from now on

Thank you!

closed 2021-11-28 by Biktorgj