the-modem-distro/pinephone_modem_sdk #47: No more GNSS signal after v4.2

carlosgonz0, 2021-12-31:
On Sailfish the GNSS worked very  well  on version 0.4.2 then after upgrade the firmware the GNSS not work anymore either v0.5.0

Biktorgj, 2021-12-31:
Any logs you can get?
There have been no changes with regards to GPS in a long time, and as far as I can see it reports location data through the port the same way it has always done (you can enable GPS with AT+QGPS=1 and then monitor the output of /dev/ttyUSB1 for NMEA messages)

nesnera, 2022-01-04:
Instead of version 0.5.0 I tried v0.5.1, but I [had to go back to v0.4.2 immediately](https://github.com/Biktorgj/pinephone_modem_sdk/issues/51). In the situation (v0.4.2) where:
- [GNOME Maps](https://wiki.gnome.org/Apps/Maps) location works
- <details>
  <summary># mmcli -m any --location-get [1]</summary>
  
  ```
  --------------------------
    GPS  |               nmea: $GPGSA,A,3,01,08,10,14,21,27,,,,,,,1.2,0.9,0.8,1*26
         |                     $GNGNS,005524.00,4926.533828,N,01732.431995,E,AAN,11,0.9,295.0,43.0,,,V*60
         |                     $GPRMC,005524.00,A,4926.533828,N,01732.431995,E,0.0,73.2,040122,2.1,E,A,V*73
         |                     $GNGSA,A,3,,,,,,,,,,,,,1.2,0.9,0.8,3*31
         |                     $GPGSV,3,1,10,01,28,284,23,08,68,213,25,10,42,056,22,14,12,322,21,1*60
         |                     $GPGSV,3,2,10,21,58,291,23,24,01,037,16,27,43,163,22,03,01,229,,1*68
         |                     $GPGSV,3,3,10,23,09,053,,32,39,118,,1*68
         |                     $GLGSV,2,1,07,74,10,279,16,82,22,115,16,84,40,314,17,67,36,037,23,1*74
         |                     $GLGSV,2,2,07,68,69,127,20,69,27,187,,83,66,085,,1*4A
         |                     $GPVTG,73.2,T,71.1,M,0.0,N,0.0,K,A*22
         |                     $GPGGA,005524.00,4926.533828,N,01732.431995,E,1,06,0.9,295.0,M,43.0,M,,*66
  ```
  </details>
- <details>
  <summary>$ echo at+qgpsgnmea=\"gsv\" | sudo atinout - /dev/EG25.AT - [2]</summary>
  
  ```
  +QGPSGNMEA: $GPGSV,3,1,10,01,29,284,27,08,67,212,25,10,41,056,21,14,12,322,25,1*6F
  +QGPSGNMEA: $GPGSV,3,2,10,21,58,292,27,27,42,163,22,03,02,229,,23,08,053,,1*66
  +QGPSGNMEA: $GPGSV,3,3,10,24,01,036,,32,40,116,,1*64
  ```
  </details>
- <details>
  <summary>$ echo at+qgpsgnmea=\"gga\" | sudo atinout - /dev/EG25.AT -</summary>
  
  ```
  +QGPSGNMEA: $GPGGA,005644.00,4926.533926,N,01732.432013,E,1,06,0.9,295.2,M,43.0,M,,*6A
  ```
  </details>
`$ echo at+qgpsloc | sudo atinout - /dev/EG25.AT -` reports `+CME ERROR: 502`, which, according to [this description](https://forum.pine64.org/showthread.php?tid=14114), it shouldn't. @carlosgonz0, how does it behave for you?

[1] [Mobian wiki - location](https://wiki.mobian-project.org/doku.php?id=location)
[2] [Quectel EG25-G GNSS Configuration and location tracking](https://forum.pine64.org/showthread.php?tid=14114)

Biktorgj, 2022-01-04:
@nesnera
I enabled location via mmcli but it seems to be working for me. Before running --location-get I had to run `mmcli -m any --location-enable-gps-nmea` , but after that I get NMEA data correctly. There should be no need to explicitly set the modes via QGPSNMEA, it's enabling them by default.

Just as a test, disable runtime suspend of the usb port:
- Edit `/usr/lib/udev/rules.d/80-modem-eg25.rules `
- On the line where "power" is set to "auto", replace "auto" with "on", then reboot. Is there any change in stability?

Unrelated, but while testing this I've seen the modem tends to crash if you suspend it for long enough with GPS on, so added that to my todo list

nesnera, 2022-01-09:
The `# mmcli -m any --location-get` works for me (as seen above) and I use them exactly the same way = I enable NMEA by `# mmcli -m any --location-enable-gps-nmea`  first.
Surprisingly I can't say the same for `# echo at+qgpsloc | sudo atinout - /dev/EG25.AT -` You?

Biktorgj, 2022-01-10:
AT+QGPSLOC only works if you previously enabled GPS via AT+QGPS=1 *and* it has a fix on the satellites, do you see if it is reporting correct data in ttyUSB1?

nesnera, 2022-01-21:
I'm glad we have the same results, even if they don't match [expectations](https://forum.pine64.org/showthread.php?tid=14114). As far as practice goes, the [browser positioning](https://developer.mozilla.org/en-US/docs/Web/API/Geolocation/getCurrentPosition) works. I have several routes recorded on PP using [osmin](https://github.com/janbar/osmin) as well.

carlosgonz0, 2022-01-27:
OK i already testing firmodem 0.5.5 and it has same issues no more gnss detecting satellites. what happening?

nesnera, 2022-01-30:
@carlosgonz0: I have also updated to v0.5.5 and I have no problem with satellite signal reception. Geolocation in the [browser](https://wiki.gnome.org/Apps/Web), [GNOME Maps](https://wiki.gnome.org/Apps/Maps), [osmin](https://github.com/janbar/osmin) and the `mmcli -m any --location-get` command works as described [above](https://github.com/Biktorgj/pinephone_modem_sdk/issues/47#issuecomment-1008442132). | PP/Manjaro/[Phosh-dev](https://github.com/manjaro-pinephone/phosh-dev)

carlosgonz0, 2022-04-06:
GNSS worked again on firmodem v0.6.2  at the same time i fixed a bug on gpsd too to Sailfish OS. 
I closing this.

carlosgonz0, 2022-04-11:
Thank you @nesnera for your troubleshooting on this issue.
I am sorry @Biktorgj  because i do not give the log that you wanted i am too busy thank you @Biktorgj

closed 2022-04-11 by carlosgonz0