the-modem-distro/pinephone_modem_sdk #49: QGMR doesn't return ADSP version but OpenQTI version

Ferenc-, 2022-01-01:
As of `0.5.1`, `AT+QGMR` doesn't return what it used to, the more detailed version of `AT+GMR`, but instead the same response as `AT+GETSWREV`.  
I believe this is a bug, but if it is not, then a documentation update would be needed.

Biktorgj, 2022-01-01:
This is on purpose :)
To ease the work done by Dylan (https://dylanvanassche.be/blog/2022/pinephone-modem-upgrade/) I have repurposed AT+QGMR so fwupd doesn't need different commands to identify the firmware depending which firmware is installed.
QGMR +ATFWBRANCH are used to identify if you're running stock or custom and which version of it.

AT+GETSWREV will probably disappear because of its redundancy, but haven't taken a decision on what to do for reporting the ADSP version (ideas welcome, right now it's logged to openqti's log but it's only partially reported by GETFWBRANCH)

Ferenc-, 2022-01-01:
Makes sense. Then I guess eventually `AT_INTERFACE.md` could be changed to reflect this.

Biktorgj, 2022-01-02:
Just updated it: https://github.com/Biktorgj/pinephone_modem_sdk/commit/44aa1ae04444ea77ee8b2a160dd2a3717ed36478

closed 2022-01-02 by Biktorgj