the-modem-distro/pinephone_modem_sdk #53: Enable location services without a connection to a mobile operator

nesnera, 2022-01-23:
I'm in the same situation as [bcnaz](https://forum.pine64.org/showthread.php?tid=8105&pid=51300#pid51300) - I need to have GNSS (GPS) enabled and the mobile should not try to make a GSM connection at the same time. Turning off "Mobile Network" (Phosh) shuts down the whole modem including the location receiving.
[image]
How to reproduce:
- run # mmcli -m any --location-status
- turn off Mobile Network
- run again # mmcli -m any --location-status
If it is not possible to turn off the GSM modem while keeping the location signal receiving running, at least a mode where GSM does not try to connect would help. It would also be a sympathetic step in [terms of privacy](https://www.nytimes.com/interactive/2019/12/19/opinion/location-tracking-cell-phone.html).

Biktorgj, 2022-01-23:
I think this request belongs in ModemManager, maybe it's best to ask here: https://gitlab.freedesktop.org/mobile-broadband/ModemManager

The modem is perfectly capable of enabling GPS without a SIM (you can try removing the SIM and enabling it via AT+QGPS=1 and then monitoring NMEA output via /dev/ttyUSB1), but ModemManager won't let any communication until everything is "active".

If you run ` mmcli -m 0 --location-enable-gps-nmea` without a SIM you'll get an error saying "modem not enabled yet", though there are some functions that could be available.

Not sure exactly what does the "Mobile Network" switch do, it doesn't fully shutdown the modem, but it might be telling ModemManager to put it offline, thus disallowing you from using GPS

nesnera, 2022-01-25:
Thanks for the response. Yes, I also suspect ModemManager, but I wanted to make sure of the lowlevel status first.

I can confirm - the "Mobile Network" switch has no effect on the output in `/dev/EG25.NMEA`, unlike the `AT+QGPS=1/AT+QGPSEND`. How this manifests itself in practice (browser location, eighths) I have yet to find out in the extravilan.

spaetz, 2022-03-16:
There was already an ooen issue in modem manager to enable location tracking when there is no SIM card innserted. Which is slightly different from disabling cell connectivity but enabling location services. Still deep in modem-manager land...

Biktorgj, 2022-03-25:
In any case I'm afraid this is out of the scope for this firmware. Retrieving location works in the firmware even without a SIM after all, so closing this issue now. If anyone finds of some work in ModemManager's part to get this working feel free to reopen though, I'm sure we'd all want to know about it

closed 2022-03-25 by Biktorgj

nesnera, 2022-03-28:
I'd like to add - in Phosh, location services are enabled in Settings > Privacy > Location Services
[image]

magdesign, 2022-06-14:
On Manjaro/Plasma/modem 0.6.6  
I am able to get GPS with an old, inactive SIM Card.
However, after deep sleep GPS seems to be lost... need someone to confirm this.

Biktorgj, 2022-06-22:
I think that's expected behaviour from ModemManager, when suspending it will stop GPS to avoid wasting power (there's no one on the other side to get the data)