the-modem-distro/pinephone_modem_sdk #55: Pinephone Pro no call audio after flash 0.5.5

fbmoose48, 2022-01-28:
Seems I have no call audio after flash. Arch - no audio/no mic, Mobian audio works but mic does not. Was working until flash, is this firmware related?

carlosgonz0, 2022-01-28:
Call and  SMS work good here with mobian-ppp and firmodem0.5.5

Biktorgj, 2022-02-20:
There was a lingering bug with regards to audio, where audio would stop working after suspend, could that be the issue you were facing?
In any case, that should be fixed in latest release, if you could try it I'd love to know if it works for you or doesn't

GertVanKruiningen, 2022-03-05:
Just wondering if mic quality has anything to do with modem fw. I have realy bad, noisy, squiky tones, calls. Both Telegram and cellphone calls.
I can hear other side perfectly fine, but they hear lots of noise and hearsh squiky tones in the background.
When I use SoundRecorder I have isues recording mic sound. Tried all mic sound settings, here in ManjaroPhosh "Make a phone call" and "Play HiFi quality music", low mic, high mic volume......
So question, could any if thus be related to modem?

Ooh yes, there's this terminal command to get list of VoLTE, no idea what it is, but none show a NewZealand carrier if that makes sence. And the last 5 are missing for me.
I am on your latest 0.5.9
Cheers
Gert

fbmoose48, 2022-03-18:
I'd like to test the new FW out on the PPP and see if the issue is resolved. But before I potentially take my daily driver offline I want to successfully flash my Braveheart first. Working through that soon hopefully

CodePhase, 2022-03-24:
Seems like this is an issue on 0.6.1 also: #72

GertVanKruiningen, 2022-03-25:
I am on 0.6.1, and it looks like its a lot better. No more bad background noises, but still a big echo.

Biktorgj, 2022-03-25:
The echo is probably caused by the microphone volume being set too high, you can try two things:
1. While in a call, go to Phosh Settings -> Sound, and move the microphone slider to the left
2. As root, edit `/usr/share/alsa/ucm2/PinePhonePro/VoiceCall.conf`, look for `cset "name='IN1 Boost' 8"` and set it to something lower, in the range of 0-3.

Biktorgj, 2022-04-16:
If anyone is still having audio issues, please move to issue #81 where there's more up to date info so we can all get everything in one place

closed 2022-04-16 by Biktorgj