the-modem-distro/pinephone_modem_sdk #57: Documentation

bfra2373, 2022-01-30:
Hi,
I would like to help you build a better doc because, well I'm kinda confuse :P.
So in order to do that I would need a bit of support.
Is their a way to get in touch?

Biktorgj, 2022-02-17:
Sure thing!
Something in particular that is confusing or would need improving?

bfra2373, 2022-02-17:
Yes indeed something confusing. 
Try to bare with me has I'm not an expert but I'll try to make this as clear as possible.

So I try to flash the modem using the script from the [Flashing this firmware](https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/FLASHING.md) page.
So I first installed adb and fastboot (btw I was using MANJARO PHOSH at the time) and then just tryed to do the install using the script you wrote.
At first, it was unclear if it worked or not has there where no confirmation. I had an error message in the first line(probably because Manjaro user are not in the TTY group by default). Then, I tryed to do it manually with no success accessing fastboot...

I tryed to read again and then found this on the [Unlock ADB access](https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-accessl) page . Should I do an unlock ADB access before running the script? Quiet not sure at this point because there his no mention of this on the [Flashing this firmware](https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/FLASHING.md) page.

I finally tried to reinstall version two of the original driver with no success with the line `echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2`   returning an error ( if I remember correctly can't access ttyUSB2 ). Maybe again be linked to ADB access?

So to summarize, 
A : Should I run the [Unlock ADB access](https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-accessl) script before trying anything else?
B : Can your script be use with the original driver?

Right now my phone is working but it doesn't seem to understand witch driver it's using :P

I'm sorry if it is a bit long but would be glad to had some information to the installation information after I cleared this :)

Biktorgj, 2022-02-20:
There is no need to unlock or enable adb to flash the firmware.
You would just need to
1. Unpack the package.tar.gz file
2. Run the flash all script (depending on the distro, you might have to run it as root)
That should be it

If you want to go back to stock, I added a similar script called "flashstock" in the recovery repo, so you don't need to use qfirehose or anything like that.

If you're getting an error when accesing ttyUSB2 it could be a permission error, or the modem maybe wasn't fully booted when you run it?

Biktorgj, 2022-04-16:
Closing due to inactivity, if you still have issues feel free to reopen the issue!

closed 2022-04-16 by Biktorgj