the-modem-distro/pinephone_modem_sdk #59: Poor call quality

alistair23, 2022-02-11:
Using the latest 0.5.5 version of the firmware with the ADSP 004 I have very poor call quality.

Every few minutes during a call all sound drops, I can't hear the other side and they can't hear me. As well as that the other side thinks that I have poor reception as the quality isn't great.

This all occurs with good phone reception. I'm using pmOS Edge. From memory I think I saw the same issue with ADSP 003 as well.

Biktorgj, 2022-02-20:
Max call duration I've tested was about 20 minutes, and as long as the Pinephone doesn't suspend, audio didn't stop working. Does that still happen with 0.5.9?

alistair23, 2022-02-21:
Hmm... Do you think the phone is suspending during the call?

Besides the drop outs though the quality is still really bad. I don't have VoLTE as it doesn't seem to work, I wonder if that is the issue.

I haven't tried 0.5.9, I'll try that and let you know

alistair23, 2024-06-26:
Closing as this is probably stale

closed 2024-06-26 by alistair23