the-modem-distro/pinephone_modem_sdk #64: USB Reset - A Possible Clue

ndf-zz, 2022-02-20:
Is is possible there is some sort of watchdog or maybe a buffer problem with the AT command port? I was having persisitent USB reset drama, but the following command completely mitigated the problem:

  # while true ; do echo ATI | atinout - /dev/EG25.AT - ; sleep 2 ; done

Simply listening to the output (eg):

  # hd /dev/EG25.AT

was not enough, but the ATI in a loop (embarrassingly) works really well.

Biktorgj, 2022-02-20:
You will probably have a similar result just by disabling runtime suspend in the modem USB port:
Open `/usr/lib/udev/rules.d/80-modem-eg25.rules` and, in the first line, change "power/control" from `auto` to `on`

That way the Pinephone won't try to suspend the usb port while the phone is awake, should remove most of the problems, without having to constantly send stuff to the AT port :)

ndf-zz, 2022-02-21:
Confirmed: setting "power/control" to "on" fixed the USB reset problems I was having. Thanks!

closed 2022-02-21 by ndf-zz