the-modem-distro/pinephone_modem_sdk #67: Interference between call and internet connection

nesnera, 2022-03-02:
I often use PP as a WiFi hotspot to connect my laptop to the Internet. I have noticed that during a call (even a ringtone is sufficient) the internet is unavailable. After the call/ring the connection is restored. Is this a firmware (0.5.5) bug?

IntinteDAO, 2022-03-04:
LTE doesn't allow call communication by default, and I don't know if this firmware supports Vo-LTE.

nesnera, 2022-03-07:
@IntinteDAO Thanks for the 100% valid help. I had to explicitly enable VoLTE with my carrier. Now calls work even when I have "Network Modde" set to "4G Only". @Biktorgj  is to be commended.👍
The problem of simultaneous calls and fast internet connection is thus gone, I'm closing the bug.

closed 2022-03-07 by nesnera