the-modem-distro/pinephone_modem_sdk #69: Modem sleep and periodic wake

SpookySkeletons, 2022-03-12:
Does the onboard chip come with an S3 mode so we can opportunistically wake and sleep the modem to save power? Anymore wiggle room for power saving yet? Is this already performed?

DylanVanAssche, 2022-03-12:
@SpookySkeletons What is S3 mode? Do you have some docs about it?

SpookySkeletons, 2022-03-12:
Suspend to RAM like the pinephone itself or a x86 computer, not sure what the onboard CPU supports.

Biktorgj, 2022-03-13:
Hi,
The modem _does_ have support for suspend to memory, but it's currently not being used in the firmware. I actually started playing around with that two days ago, but I'm unsure of the unintended consequences it might have when enabled. The stock firmware does enable suspend when the host disables USB, but that makes it a bit slower to respond when the PinePhone is waking up, adding more problems to ModemManager when reconnecting.

So far I haven't seen it stay suspended for more than 40 seconds at a time, but it's so far doing OK in the PinePhone Pro. If I see it doesn't introduce more issues, I will enable it by default (with a toggle to turn it off, as always)

In any case, the difference in power consumption should be minimal (ADSP will stay on to maintain the network connection and power draw in idle is about 20mA), so don't expect much of an improvement out of this

Biktorgj, 2022-03-25:
The USB port is what seems to trigger it, if you shut down the port *and* don't have ADB enabled in the stock firmware, the modem enters suspend, and wakes up again when it detects that the port has been reconnected