the-modem-distro/pinephone_modem_sdk #70: LTE signal loss sticky

SpookySkeletons, 2022-03-12:
Complete signal loss inside of deep buildings and rural areas can be fairly sticky so once you emerge again the modem might not reacquire signal in a reasonable time prompting me to restart the phone instead of a 5min wait.

Are there any tunables related to aggression of signal acquisition?

Biktorgj, 2022-03-13:
Not that I know of, there are settings to change the bands and priorities of each, or to lock the modem to certain ones, but "scan times" I haven't seen a way to configure them.

Next time it happens you might want to try disabling and re-enabling network (if using Phosh, go to Settings -> Mobile Network and switch the Mobile network toggle off, wait a few seconds, and turn it on again). That should issue a disconnect and reconnect to the modem

closed 2022-03-21 by SpookySkeletons

DylanVanAssche, 2022-03-21:
@SpookySkeletons Would you mind to add your findings here: https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/ADSP-CARRIERS.md ?