the-modem-distro/pinephone_modem_sdk #72: No audio on first call after boot for Pinephone Pro

ragreenburg, 2022-03-23:
After booting the device my first call always doesn't have any audio. I am running Phosh on Arch with firmware version 0.6.1 and ADSP 1.003 with Mint Mobile.

CodePhase, 2022-03-24:
I filed a duplicate issue, #75.  Here are the details I added to this problem:
> After updating my Pinephone Pro (mobian bookworm) to the 0.6.1 firmware (and rebooting the phone), I can make and receive calls and texts but there is no audio on my side or the other side of the call. Wys shows a couple errors:
> 
> ```
> Mar 24 09:21:07 mobian wys[1708]: wys_audio_ensure_loopback: assertion 'self->modem' failed
> Mar 24 09:21:07 mobian wys[1708]: wys_audio_ensure_loopback: assertion 'self->modem' failed
> Mar 24 09:21:14 mobian wys[1708]: wys_audio_ensure_no_loopback: assertion 'self->modem' failed
> Mar 24 09:21:14 mobian wys[1708]: wys_audio_ensure_no_loopback: assertion 'self->modem' failed
> ```
> 
> I tried restarting wys with `systemctl --user restart wys` but it didn't fix the problem
> 
> I was previously using 0.5.9 and the audio was working (albeit with some static in my local handset). When I upgraded to 0.6.1 I also upgraded the ADSP to 01.003.01.003 from 01.002.01.002, which is what I was using with the 0.5.9 firmware. I tried rolling back to 0.5.9 and 01.002.01.002 but the issue remained so I upgraded both again.

> I got the audio to work both ways if I restart the ModemManager.service. Restarting eg25-manager.service doesn't cut it.
I'll add to this that not only did I restart ModemManager.service but also the systemd-networkd.service and NetworkManager.service.  I do this so often that I created a script for it:

```
#!/bin/bash

sudo systemctl restart ModemManager
#sudo systemctl restart eg25-manager
if [ $? -eq 0 ]; then
  sleep 30
  sudo systemctl restart systemd-networkd; sudo systemctl restart NetworkManager
  pkill chatty
else
  echo "Failed to restart ModemManager"
#  echo "Failed to restart eg25-manager"
fi
```

ragreenburg, 2022-03-24:
Unfortunately the script above doesn't work for me on Arch. Or at least when testing it in the Settings. The mic works but the output doesn't.

CodePhase, 2022-03-24:
@ragreenburg if you go into Settings -> Sound -> Output section -> Output Device -> Test, do you get sound from the left and right side?  Like you said in the other bug, maybe it's an issue with the audio profiles in and out of calls.

ragreenburg, 2022-03-24:
@CodePhase I should have clarified, doing those steps is how I was testing the output. So when following those steps I don't get any sound output. Though with programs like `pavucontrol` I can see that it is reading there is output but I can't hear anything.

CodePhase, 2022-03-24:
Ok, I just tried a few things.  Turns out that if I set the _speaker_ 'Configuration' to Default I can perform the sound test successfully.  Changing the 'Configuration' to Phone call I get no sound on the test like you.

However, when I perform a phone call to my landline I can hear just fine on both sides of the call.  I'm not sure that the Settings Sound test is working for the 'Phone call' configuration

CodePhase, 2022-03-24:
Something I just thought about too, not sure if it's relevant since the Settings -> Sound stuff might not be working for phone calls but, the 'System Volume' 50% for my 'Default' configuration in the Output section, but changing to the 'Phone call' configuration in the output section changes the 'System Volume' to be 100%.

ragreenburg, 2022-03-24:
I also noticed the system volume was all the way down and took care of that. On my end the mic also sometimes is very low (20%). Though when I put my system in Phone Call configuration I can't hear music or podcasts either, with or without headphones. I've also tried calls (because I am currently in need of a pretty serious car fix) and those don't work either.

CodePhase, 2022-03-24:
can you try setting the volume in pavucontrol during a call?

Maybe the Settings app isn't working for phone call sound.  I think at least the microphone setting works because I have to set my Input sound volume to nearly 0 to not get nasty echos and feedback on the receiver side of the call.  Both the 'Default' and 'Phone call' profiles are set like that for my microphone.

CodePhase, 2022-03-24:
Not sure if Arch uses wys but, just for kicks, maybe restart the wys service?
`systemctl --user restart wys`

ragreenburg, 2022-03-24:
Unfortunately that also doesn't work. I can switch between the configurations and see the volume sliders for both being non-zero but when switching to Phone Call it loses audio.

carlosgonz0, 2022-03-24:
Calls IN OUT SMS is working good here with Firmodem 0.6.1 and Mobian-gnome-42

Biktorgj, 2022-03-25:
Ok let's go back a few steps here since I really don't understand what's going on exactly.

@CodePhase: There's two different ALSA profiles, one for `HiFi` use (system sounds, music, video etc.) and one specific for `VoiceCall`. The test function in Settings->Sound will not work when the `VoiceCall` profile is used. Up until recently there was some issue in Mobian where the first call wouldn't route audio (if you went, while in that call, to sound settings and switched profiles manually it would eventually work, second call would do it OK).

@ragreenburg ADSP version *shouldn't* affect audio configuration and routing inside the modem. All that is handled by the modem userspace. The only thing that makes sense to me in your scenario is that I left something broken in Circuit Switch calls and that there's some issue with ADSP 003 and your line connecting to VoLTE. 

If you try a loopback call, by telling the modem to call you, does that work? There the ADSP isn't involved at all

ragreenburg, 2022-03-28:
I unfortunately am not able to get the modem to call me as my SMS to the modem is currently not working (#73).

ragreenburg, 2022-03-28:
@neovalis I noticed a similar thing with `pavucontrol` but it didn't show up in the Sound Settings. So I think you might be right. I can play output and it shows there is output but it doesn't come out of the speakers.

ragreenburg, 2022-03-29:
On Arch we had a new `callaudio` package released yesterday(?) which seems to have fixed the missing call audio. I only had time to test it on one call but it was the first call that's worked since the update and I tried about 10-15 before last night. So that seems like it was routed incorrectly?

McUberStein, 2022-04-11:
I fixed this problem by installing the the alsa-firmware package on Arch.

Biktorgj, 2022-04-16:
There are three similar issues opened, so to avoid repeating, please, let's all go to issue #81 where there's more up to date info (even if this looks more like a distro issue than anything)

closed 2022-04-16 by Biktorgj