the-modem-distro/pinephone_modem_sdk #73: Unable to message modem

ragreenburg, 2022-03-23:
I was able to message my modem for a bit then after a while, I no longer receive messages from the modem. My messages go through (or at least that's what Chatty says) but I don't get any response. When using `call me in 1` it never actually makes the call.

I am using Phosh on Arch with firmware version 0.6.1 and ADSP 1.003 on Mint Mobile.

Biktorgj, 2022-03-25:
Does chatty show the "delivered" mark next to the message you sent?
If it doesn't something might have crashed in the modem, leaving it blocked until reboot. Might be good to check the logs to see what's going on (https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/TROUBLESHOOTING.md#getting-logs-to-help-with-an-issue)

Openqti.log should show if the message is being received or not (be careful with that log as it might have personal info in it)

ragreenburg, 2022-03-25:
Chatty shows that the message has been delivered. I am trying to get the openqti log but I keep getting a permissions error when running the command. I've tried running as root and still get the error.

stacyharper, 2022-03-28:
I also relate a similar bug on sxmo. Some sms are just shallowed. Some of them actually seems to works but take a long time (~1 minutes for a "call me"). But I never receive a response by sms. The new TTS while in call seems to works as the TTS voices the "help" reply (when it works and after a long time). (btw I also got timeouts errors while sending those sms (3/4 times) maybe unrelated ?)

Those issues may be not related to the modem firmware. I had them with 0.5.9 but it was working very smoothly at the release time. It started to sucks some time later. Maybe a eg25-manager or modem-manager issue ?

ragreenburg, 2022-03-28:
@Biktorgj below is the log file. 
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8364116/openqti.log)

Biktorgj, 2022-03-30:
> @Biktorgj below is the log file. [openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8364116/openqti.log)

@ragreenburg That log shows that the modem received an SMS but ModemManager never queried for the message, so it retried by sending a RING IN signal multiple times, but there's no mention of an internal messge pending delivery... Can you try to send the command `AT+SIMUSMS` and check if you get a Hello World at least?

ragreenburg, 2022-03-30:
I ran the command and get the response in the mdoem `response: ''` but no text. I did that about 15 minutes ago.

closed 2022-05-05 by ragreenburg