the-modem-distro/pinephone_modem_sdk #74: Memory garbage after default `name` and `username`

Ferenc-, 2022-03-24:
Modem SDK: `0.6.1`
OS: `Manjaro 22.03`
HW: `PP Beta convergence`

When the `name` and `username` commands are issued in a text message, before the default `Modem` and `Admin` names are replaced, the response has some sort of trailing memory garbage, but once the default name is replaced with `set name Foo`,
then consequent responses don't contain that anymore.

bircoph, 2022-04-09:
Same here: SDK-0.6.1, pinephone beta convergence, PostMarketOS-21.12 with latest updates . When names are set to custom values, the problem is gone as well.

Biktorgj, 2022-04-16:
The issue happens because the NAND block where the names are stored might have garbage before the first write, so when you write once, the problem goes away
Planning to move all that to the filesystem, but it will break all stored settings (so you'll need to re-set names, usb audio if configured etc.)
Will post back when I get a release running with all that fixed

Biktorgj, 2022-05-05:
Hi again!
[Version 0.6.4](github.com/biktorgj/pinephone_modem_sdk/releases/latest) fixes this bug by storing Openqti's configuration file in `/persist/openqti.conf`, so, unless someone went to the file and started putting garbage on purpose, this is fixed now :)

closed 2022-05-05 by Biktorgj