the-modem-distro/pinephone_modem_sdk #75: wys fails after 0.6.1 update

CodePhase, 2022-03-24:
After updating my Pinephone Pro (mobian bookworm) to the 0.6.1 firmware (and rebooting the phone), I can make and receive calls and texts but there is no audio on my side or the other side of the call.  Wys shows a couple errors:
```
Mar 24 09:21:07 mobian wys[1708]: wys_audio_ensure_loopback: assertion 'self->modem' failed
Mar 24 09:21:07 mobian wys[1708]: wys_audio_ensure_loopback: assertion 'self->modem' failed
Mar 24 09:21:14 mobian wys[1708]: wys_audio_ensure_no_loopback: assertion 'self->modem' failed
Mar 24 09:21:14 mobian wys[1708]: wys_audio_ensure_no_loopback: assertion 'self->modem' failed
```
I tried restarting wys with
`systemctl --user restart wys`
but it didn't fix the problem

I was previously using 0.5.9 and the audio was working (albeit with some static in my local handset).  When I upgraded to 0.6.1 I also upgraded the ADSP to 01.003.01.003 from 01.002.01.002, which is what I was using with the 0.5.9 firmware.  I tried rolling back to 0.5.9 and 01.002.01.002 but the issue remained so I upgraded both again.

CodePhase, 2022-03-24:
I got the audio to work both ways if I restart the ModemManager.service.  Restarting eg25-manager.service doesn't cut it.

ragreenburg, 2022-03-24:
I think we have the same problem, https://github.com/Biktorgj/pinephone_modem_sdk/issues/72. Mine originally said it was only the first call but I realized I was wrong. I've also noticed that in the settings, the microphone audio in Phone Call profile was set to about 20%.

CodePhase, 2022-03-24:
Yeah, I think my issue here is probably a duplicate of yours.  Guess I should have checked the issues list a little more carefully before creating this.  I'll close this one and we can add relevant information to your issue.

closed 2022-03-24 by CodePhase