the-modem-distro/pinephone_modem_sdk #77: Adjacent hardware compatibility

SpookySkeletons, 2022-03-24:
Does this firmware have any functionality if flashed to an EC25 chip?
If not does anything specific limit the build to EG25?

Biktorgj, 2022-03-25:
I think it *should* work, but I don't have an EC25 to try. The differences might be in the NAND chip which may or may not be supported by this version of the bootloader, the userspace is probably fine.

If you're going to try it, make sure you have a recovery firmware for the EC25, and access to the USB_BOOT pin so you can use `qfirehose` to flash it if it fails.

Don't flash EG25 ADSP firmware into the EC20/21/25 though, as those should be specific for the RF parts used in each model

SpookySkeletons, 2022-03-25:
Any documentation to verify the NAND info before potentially zapping the bootloader?

Biktorgj, 2022-04-16:
> Any documentation to verify the NAND info before potentially zapping the bootloader?

Hard to say, since there's 0 documentation, and the kernel might not even give the correct nand chip ID... but look for something like this in the kernel log:
```
[    1.894001] sps: BAM device 0x07984000 is not registered yet.
[    1.898942] sps:BAM 0x07984000 is registered.
[    1.903014] msm_nand_bam_init: msm_nand_bam_init: BAM device registered: bam_handle 0xce6d1200
[    1.911938] sps:BAM 0x07984000 (va:0xd0b20000) enabled: ver:0x18, number of pipes:7
[    1.921297] msm_nand_parse_smem_ptable: Parsing partition table info from SMEM
[    1.927514] msm_nand_parse_smem_ptable: SMEM partition table found: ver: 4 len: 17
[    1.935113] msm_nand_version_check: nand_major:1, nand_minor:5, qpic_major:1, qpic_minor:5
[    1.943361] msm_nand_flash_onfi_probe: Found a non ONFI device
[    1.949124] msm_nand_scan: NAND Id: 0x1590aa98 Buswidth: 8Bits Density: 256 MByte
[    1.956598] msm_nand_scan: pagesize: 2048 Erasesize: 131072 oobsize: 128 (in Bytes)
[    1.964233] msm_nand_scan: BCH ECC: 8 Bit
[    1.968199] msm_nand_scan: CFG0: 0x290408c0,           CFG1: 0x0804715c
[    1.968199]             RAWCFG0: 0x2b8400c0,        RAWCFG1: 0x0005055d
[    1.968199]           ECCBUFCFG: 0x00000203,      ECCBCHCFG: 0x42040d10
[    1.968199]           RAWECCCFG: 0x42000d11, BAD BLOCK BYTE: 0x000001c5
[    1.994622] Creating 17 MTD partitions on "7980000.nand":
```
Supported NAND chips in the bootloader are here: https://github.com/Biktorgj/quectel_lk/blob/0d061c60d2921510440890c769d2b9d73b05a195/platform/msm_shared/qpic_nand.c#L70

The EG25-G uses a NAND chip with ID 0x1590AAAD

DHowett, 2022-08-26:
Just to chip in a bit more info: my EC25-A has a NAND chip with ID `0x2690ac98`.

Here's `dmesg` from that region:

```
[    0.378682] msm_nand_bam_init: msm_nand_bam_init: BAM device registered: bam_handle 0xceb0b600
[    0.378903] sps:BAM 0x07984000 (va:0xd09a0000) enabled: ver:0x18, number of pipes:7
[    0.380950] msm_nand_parse_smem_ptable: Parsing partition table info from SMEM
[    0.380992] msm_nand_parse_smem_ptable: SMEM partition table found: ver: 4 len: 16
[    0.381020] msm_nand_version_check: nand_major:1, nand_minor:5, qpic_major:1, qpic_minor:5
[    0.381062] msm_nand_scan: NAND Id: 0x2690ac98 Buswidth: 8Bits Density: 512 MByte
[    0.381071] msm_nand_scan: pagesize: 4096 Erasesize: 262144 oobsize: 128 (in Bytes)
[    0.381079] msm_nand_scan: BCH ECC: 8 Bit
[    0.381093] msm_nand_scan: CFG0: 0x290409c0,           CFG1: 0x08045d5c
[    0.381093]             RAWCFG0: 0x2b8401c0,        RAWCFG1: 0x0005055d
[    0.381093]           ECCBUFCFG: 0x00000203,      ECCBCHCFG: 0x42040d10
[    0.381093]           RAWECCCFG: 0x42000d11, BAD BLOCK BYTE: 0x00000175
[    0.381114] Creating 16 MTD partitions on "7980000.nand":
[    0.381127] 0x000000000000-0x000000280000 : "sbl"
[    0.381839] 0x000000280000-0x000000500000 : "mibib"
[    0.382440] 0x000000500000-0x000001b00000 : "efs2"
[    0.383016] 0x000001b00000-0x000002500000 : "sys_rev"
[    0.383602] 0x000002500000-0x000002800000 : "tz"
[    0.384208] 0x000002800000-0x000002b00000 : "rpm"
[    0.384728] 0x000002b00000-0x000002e00000 : "aboot"
[    0.385261] 0x000002e00000-0x000003800000 : "boot"
[    0.385792] 0x000003800000-0x000004880000 : "scrub"
[    0.386350] 0x000004880000-0x0000098c0000 : "modem"
[    0.386926] 0x0000098c0000-0x000009bc0000 : "misc"
[    0.387489] 0x000009bc0000-0x00000a5c0000 : "recovery"
[    0.388066] 0x00000a5c0000-0x00000b9c0000 : "fota"
[    0.388679] 0x00000b9c0000-0x00000d7c0000 : "recoveryfs"
[    0.389239] 0x00000d7c0000-0x000013bc0000 : "sys_back"
[    0.389765] 0x000013bc0000-0x000020000000 : "system"
```

Curious to note that there's one fewer partition than on the EG25-G.

Biktorgj, 2022-08-31:
Interesting, is the system partition mounted read write or something? 
One of the things I've been working on this summer is a repartition tool that should be compatible with all these modules (probably anything using a mdm9x series at least) (https://github.com/Biktorgj/qcom_partition_tool), so if needed we should be able to slice that `system`partition in half and repurpose part of it.

About the bootloader, if anyone has:
- EDL test point access 
- A firmware update package for the modem
- Some patience, if something doesn't work at first

We can test the bootloader and do some tweaks to it if needed :)

DHowett, 2022-09-02:
> Interesting, is the system partition mounted read write or something?

```
ubi0:rootfs on / type ubifs (rw,relatime,bulk_read)
```

_Indeed!_

I've patched my firmware (and stripped the update to only include `system.img`) to offer `adb shell` access. Looking over my diff, it doesn't look like I'm the one who mounted the rootfs read-write. :smile:

I'm definitely interested in helping out. I can probably get access to the EDL test point as well. I've got a firmware update package versioned `EC25AFAR05A07M4G_01.001.01.001`.

Biktorgj, 2022-09-03:
If I had to guess... I'd say from those 6 testpoints exposed in the corner of the board near the connector, 4 could be the debug uart + VCC, and the other 2 could be usb_boot + 1.8v to bridge them:
[image]

If the pads are exposed, like in the photo, and you have good soldering skills, you should be able to solder 0.1mm wire  (https://uk.rs-online.com/web/p/copper-wire/3377088) to the usb_boot pin (115, bottom left of the picture below) directly (I have done that to some of my modules that didn't expose the pin to the board) 

[image]

DHowett, 2022-10-29:
Alright! I've finally had occasion to pull my EC25-A out of where I've been using it and hook it up to a PCIe WWAN USB adapter.

I'm no stranger to poking at pins--for this specific device, I've already flashed a custom firmware image onto it 🙂. It just happens to be one derived from the upstream EC25-A firmware that re-enables ADB and enables `root` login.

So, what do you need? How can I help?

rosbeef, 2024-01-10:
I have this modem in version EC25-AU.
@DHowett  did you manage to flash something ? i suspect the stock firmware to stuck insanely on 3G on 4g instable network, and dream to have the same "nearly" opensource firmware on my router modem as in my pinephone. ( without to trash and buy a new one)

UltraBlackLinux, 2024-07-10:
How's this going? I managed to brick my modem earlier today during a firmware update with the official tool (??).
What better time to try the modem distro. 
I have an EC25-E and I'd really like to know if this works.

To you all that have managed to get the modem into EDL mode - how exactly did you do that? connecting VDD_EXT to USB_BOOT didn't do anything. Do I need to somehow power-cycle the modem?