the-modem-distro/pinephone_modem_sdk #80: Receiving SMS Messages Delays

CodePhase, 2022-03-28:
This may not be an issue related to this project but I wanted to file an issue here just in case.

I am having a problem where SMS messages are received by my phone _sometimes_ after significant delays of between 5 and 10 minutes.  Sometimes even messages sent within seconds of each other will be received by my phone minutes apart.  The texting app, Chats in Phosh shows the correct timestamps of when the messages were sent.

I started noticing this problem when I upgraded my modem firmware from 0.5.9 to 0.6.1 and the ADSP from 01.002 to 01.003 but it's possible it was an issue before and I didn't notice.  The problem happens both while the phone is, and is not suspended.

I saw this bug on the Mobian bug tracker: https://gitlab.com/mobian1/devices/pinephone-support/-/issues/9 but I don't know if this is the same problem since I don't usually have a problem getting the SMS messages _eventually_

Pinephone Pro
Mobian Bookworm, Phosh (kernel 5.16.17+rockchip-14)
Mobile provider: Verizon, USA (4G LTE)

Biktorgj, 2022-03-30:
We need logs to debug that.
Easier way to check where we're at would be to leave the phone suspended for a minute or two, and then send an SMS to it. Even if the message doesn't arrive the screen should turn on, as the modem would signal there's some activity and try to wake it up. If that doesn't even happen, maybe the SMS is not reaching the modem at all.

But better to get some logs from the modem: https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/TROUBLESHOOTING.md#getting-logs-to-help-with-an-issue

Most important here would be ModemManager and Openqti logs. Take a look at them and redact if needed before posting (or feel free to DM them in Matrix) since they might have phone numbers, carrier data etc.

CodePhase, 2022-04-07:
I've been keeping an eye on this and I think it _might_ be a chatty issue.  I am still tinkering with it so once I get something a little more conclusive I'll report back and get you relevant logs.  Thank you for following up with me!

bfra2373, 2022-07-25:
Hi, 
I have the same issue where my message get delayed a lot. Sometime it goes 24h before receiving them. I've checked the modem clock with sudo mmcli -m any --command='AT+CCLK?' and after installing your firmware, the time is good but not the date (24/06/06) (yy/mm/dd). Anyway I ask the modem to log the openQTI and Kernell file from the +223344556677 number and would like to post them if I can find them. Any clue's about where to look for them? 
I use archPhosh on a PPpostmarket edition
Thx!

bfra2373, 2022-08-03:
Did a bit of digging and I have no openQTI log and every other log from [https://github.com/the-modem-distro/pinephone_modem_sdk/blob/honister/docs/TROUBLESHOOTING.md#getting-logs-to-help-with-an-issue] return empty file's...

bfra2373, 2022-08-04:
With the new released of arch Phosh thoose log came out but still have an issue with adb pull so I can't get openqti nor Modem's kernell (lib??.so)
[eg25.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9260565/eg25.log)
[mod.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9260566/mod.log)
[net.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/9260567/net.log)

CodePhase, 2022-09-18:
> With the new released of arch Phosh thoose log came out but still have an issue with adb pull so I can't get openqti nor Modem's kernell `adb: error while loading shared libraries: libprotobuf.so.30: cannot open shared object file: No such file or directory` wich is an open issue in arch repo...

In order to get the modems logs through adb I have to use sudo:

```
sudo sh -c "adb shell 'echo Collecting dmesg... > /dev/kmsg'"
sudo sh -c "adb shell dmesg > dmesg-modem.txt"
sudo sh -c "adb pull /var/log/openqti.log"
```

bfra2373, 2022-09-18:
I read that the library has been broken for some time now... So I'll wait for it to be fixed to get the other log from Phosh.

bfra2373, 2022-11-14:
I got some time to thinker a bit and now the log's are working!
I still had the same issue with text message getting in about 24h later but not all the time.
Here are some log's of when thing's go wrong (text message not getting in) you can see in the openqti.log that I send text message to myself at the end but they are not getting in. Should I change my modem log to mmcli -G debug to get more info?
Pinephone
Phosh ARCH (Linux Danctnix 6.0.2-3)
Mobile provider : FIZZ with ADSP Version 01.003.01.003 (v0.7.0)
[dmesg-modem.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003294/dmesg-modem.log)
[dmesg-pinephone.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003295/dmesg-pinephone.log)
[eg25-manager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003296/eg25-manager.log)
[ModemManager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003297/ModemManager.log)
[netwrokmanager.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003298/netwrokmanager.log)
[openqti.log](https://github.com/the-modem-distro/pinephone_modem_sdk/files/10003299/openqti.log)