the-modem-distro/pinephone_modem_sdk #82: OFono  is receiving the SMS twice.

carlosgonz0, 2022-04-06:
When someone send me a sms the phone receiving the message twice, when need be just one sms received.
Outcoming SMS it fine.

brennana, 2022-04-15:
I seem to be having this issue as well after flashing version 0.6.1 from stock. Also using a Pinephone Pro and Manjaro w/ Plasma Mobile (Spacebar SMS app).

Biktorgj, 2022-04-16:
I've seen oFono query, read and delete the same message multiple times, but Plasma is using ModemManager, just like Phosh / Chatty (whcih doesn't seem to do that). Most likely scenario I can imagine is the modem not getting some of the transaction messages... 
Can you get ModemManager / oFono logs? (make sure to redact them to avoid publishing phone numbers)

brennana, 2022-04-22:
When receiving an SMS, the first message comes in, without any logs generated from ModemManager. However, a couple seconds later, when the duplicated second message arrives, just two lines are output from ModemManager:

```
Apr 22 13:41:10 pinephonepro ModemManager[3645]: transaction 0x1d aborted, but message is not abortable
Apr 22 13:41:10 pinephonepro ModemManager[3645]: [/dev/cdc-wdm0] No transaction matched in received message
```

Biktorgj, 2022-05-04:
Could be caused by transaction timeouts when power/control is set to auto (the Pinephone will try to suspend the modem even when ModemManager/oFono are actively using it). Please, change the udev rules to set power/control to "on", which will prevent the modem's USB from being suspended while the phone is active (it will still put USB to sleep when the Pinephone suspends) and let me know if that fixes it (https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/SETTINGS.md#pinephone)

brennana, 2022-05-05:
It seems changing that udev rule has caused the duplicate messages to cease. I've not noticed any disconnection issues as described in the SETTINGS document (using a PPP, too), so I had not applied the `power/control` change before.

Seems like it's a good idea on the PPP as well as the original PP then. Thanks for the tip!

Biktorgj, 2022-05-13:
Added instructions in Settings for the Pinephone Pro. It's not always needed, but just in case better have it there (https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/SETTINGS.md#pinephone-pro)

closed 2022-05-13 by Biktorgj