the-modem-distro/pinephone_modem_sdk #88: Delay from switching to call audio if you are using "Custom alert tone" (affecting to outgoing calls only)

Alaraajavamma, 2022-04-18:
Hi! 

My carrier does not provide dialing tone / alert tone for outgoing calls so I have two options: use "Custom alert tone" or wait in silence. 

I discovered an issue when using "Custom alert tone" (AT+EN_CAT). 
Here are steps in chronological order:
1. Start outgoing call and "custom alert tone" starts beep sound
2. Recipient picks up and says "Hi" -> my Pinephone still plays "custom alert tone" for roughly 5 seconds so I do not know is the call picked up or not
3. After roughly 5 seconds "custom alert tone" stops and call audio starts -> this usually causes a funny situation, but when we both know that we can hear each other the call goes normally to the end.

If I disable "custom alert tone" (AT+DIS_CAT) I have to wait in pure silence but if recipient picks up I can hear him immediately. 

I tried if this was due to the fact that the "default custom alert tone" audio tape is 5 seconds long but the problem remained the same even though i switched to a one second tape. 

I am not an expert on this but trying to learn and trying to help community as I can. 
There is also similar / kind of similar issue in Gnome-Calls and I started to think can these have any connection.

Biktorgj, 2022-04-20:
Hi,
After debugging this, the problem came from the audio start-stop and alerting tone playback functions. For each operation, both of those functions would open the alsa mixer, enable or disable whatever they needed, then close it back. Each time any of the functions would open the mixer, the system would wait about a second (it not only opens the mixer but also reads all the available controls), so when it had to stop audio -> play tone -> close tone -> start call audio it could end up wasting about 5 seconds at the beginning of the call.

This change should fix that issue entirely: https://github.com/Biktorgj/meta-qcom/commit/1e666cec6a6a8dcb6abab6822fb1e797e1e3c6ab

But will keep this issue open until I release 0.6.4 in case there's some bug with these changes I haven't seen

Biktorgj, 2022-05-04:
Published 0.6.4, which contains a fix for this , so closing this now :)
https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.6.4

closed 2022-05-04 by Biktorgj