the-modem-distro/pinephone_modem_sdk #89: make root_fs fails in do_compile

jg110, 2022-04-20:
I'm getting an error when trying to run `make everything` with release 0.6.1, which I've traced down to `make_rootfs`:

```
ERROR: openqti-1.0-r8 do_compile: ExecutionError('/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/temp/run.do_compile.122233', 1, None, None)
ERROR: Logfile of failure stored in: /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/temp/log.do_compile.122233
Log data follows:
| DEBUG: Executing shell function do_compile
| src/config.c: In function 'do_sync_fs':
| src/config.c:39:5: warning: ignoring return value of 'system' declared with attribute 'warn_unused_result' [-Wunused-result]
|    39 |     system("sync");
|       |     ^~~~~~~~~~~~~~
| src/cell.c: In function 'parse_wcdma_neighbour_data':
| src/cell.c:442:60: warning: iteration 8 invokes undefined behavior [-Waggressive-loop-optimizations]
|   442 |         report_data.current_report.wcdma.neighbours[i - 1] =
|       |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
|   443 |             report_data.current_report.wcdma.neighbours[i];
|       |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| src/cell.c:441:21: note: within this loop
|   441 |       for (i = 1; i < 16; i++) {
|       |                   ~~^~~~
| src/cell.c: In function 'parse_gsm_neighbour_data':
| src/cell.c:521:58: warning: iteration 8 invokes undefined behavior [-Waggressive-loop-optimizations]
|   521 |         report_data.current_report.gsm.neighbours[i - 1] =
|       |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
|   522 |             report_data.current_report.gsm.neighbours[i];
|       |             ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| src/cell.c:520:21: note: within this loop
|   520 |       for (i = 1; i < 16; i++) {
|       |                   ~~^~~~
| src/openqti.c: In function 'main':
| src/openqti.c:74:14: warning: implicit declaration of function 'dump_audio_mixer' [-Wimplicit-function-declaration]
|    74 |       return dump_audio_mixer();
|       |              ^~~~~~~~~~~~~~~~
| /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/11.2.0/ld: /tmp/ccF1Q2cP.o: in function `main':
| openqti.c:(.text.startup+0x16e): undefined reference to `dump_audio_mixer'
| collect2: error: ld returned 1 exit status
| WARNING: /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/temp/run.do_compile.122233:145 exit 1 from 'arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a8 -fstack-protector-strong -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security -Wdate-time --sysroot=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/recipe-sysroot -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fmacro-prefix-map=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8=/usr/src/debug/openqti/1.0-r8 -fdebug-prefix-map=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8=/usr/src/debug/openqti/1.0-r8 -fdebug-prefix-map=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/recipe-sysroot= -fdebug-prefix-map=/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/recipe-sysroot-native= -Wl,-z,relro,-z,now -O2 src/config.c src/scheduler.c src/pico2aud.c src/qmi.c src/timesync.c src/cell.c src/call.c src/command.c src/proxy.c src/sms.c src/tracking.c src/helpers.c src/atfwd.c src/logger.c src/md5sum.c src/ipc.c src/audio.c src/mixer.c src/pcm.c src/openqti.c -o openqti -lpthread -lttspico'
| WARNING: Backtrace (BB generated script):
|       #1: do_compile, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/temp/run.do_compile.122233, line 145
|       #2: main, /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/build/tmp/work/cortexa8t2hf-neon-poky-linux-gnueabi/openqti/1.0-r8/temp/run.do_compile.122233, line 149
ERROR: Task (/home/jg110/Documents/repos/pinephone_modem_sdk/yocto/meta-qcom/recipes-modem/openqti/openqti.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1557 tasks of which 1477 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/jg110/Documents/repos/pinephone_modem_sdk/yocto/meta-qcom/recipes-modem/openqti/openqti.bb:do_compile
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
make: *** [Makefile:45: root_fs] Error 1
```

I've made sure to the best of my ability that I have all the development packages needed, although the recommended packages are for a Debian-based system and I'm on Fedora 34. From my admittedly limited C/C++ experience, this looks like a header file isn't being included where it should be, but as far as I can tell `dump_audio_mixer` isn't part of any system header file and should be part of what's provided by yocto.

Biktorgj, 2022-04-20:
Sorry!

I missed one file in previous commit... After some big modifications in the audio handling code, I removed dump_audio_mixer(), but I missed adding the changes to openqti.c in there.
Please go to `yoct/meta-qcom` and run `git pull`, then try to build it again :)

jg110, 2022-04-20:
Thanks! That fixed it.

closed 2022-04-20 by jg110