the-modem-distro/pinephone_modem_sdk #93: pinephonepro firmware v0.6.1 > v0..6.4

henrythemouse, 2022-05-05:
I recently installed Biktorgi fw v0.6.1 using gnome-firmware and have been having a few problems. The modem disappears often, the audio can be choppy and delayed to the point of hanging up. Sometimes there is a long delay when making a call, before I hear a ring on the other end. 

Anyway, instead of going back to the stock firmware, I'd like to give v0.6.4 a try as it seems to address many problems in v0.6.1. I'd also like to use gnome-firmware to do that upgrade. However, when I open it, it still has v0.6.1. Can anyone tell me what package contains this new version or is it in the  kernel? If the process will take weeks or months than I might try the command line, else I'll wait.

Thanks

DylanVanAssche, 2022-05-09:
v0.6.4 is still in LVFS testing, it has to move first to LVFS stable if you haven't enable LVFS testing repos.

henrythemouse, 2022-05-10:
I see. Makes sense. Thank you for the explanation.

closed 2022-05-10 by henrythemouse