the-modem-distro/pinephone_modem_sdk #94: SMS does not arrive

hooger, 2022-05-07:
Hello,

first of all thank you for your amazing job, I really appreciate that you take the time to create a FOSS driver. I'm confident, that it can work much better than any other mode developmentl because we can reach out and you can react openly to issues. And this is the reason I'm writting. :)

So calling and answering calls works, also sending SMS works, but receiving is unreliable, but mostly does not work. Setup:
- PinePhone, Manjaro CE. Magyar Telekom SIM card (Hunagary, basically Deutsche Telekom subsidiary).
- mobian mobian-pinephone-phosh-20220327.img, running from SD card, system upgrade and restart done before doing these tests.
- firmware: 0.6.4, ADSP: 4th version (installed both from a Manjaro system, from the internal storage)
- I did the following stuff:
  - 20:16 -> send SMS to the PP, does not arrive
  - 20:17 -> call PP, call rangs, I canceled it
  - 20:17 -> call from PP, rangs, I canceled it
  - 20:18 -> SMS sent from the PP. SMS arrives
  - 20:18 -> SMS sent to the PP, does not arrive

Logs attached created with the recommended script. Tell me what else should I test, it bothers me a lot, so I try to help as much as I can.
Best regards,

[dmesg-modem.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645754/dmesg-modem.txt)
[dmesg-pinephone.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645755/dmesg-pinephone.txt)
[eg25-manager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645756/eg25-manager.log)
[modemmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645757/modemmanager.log)
[networkmanager.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645758/networkmanager.log)
[openqti.log](https://github.com/Biktorgj/pinephone_modem_sdk/files/8645759/openqti.log)

Biktorgj, 2022-05-13:
I can see your message being sent, but there's not even an indication of any message being received after that.
The last thing I can see is this: `[457.7516]` the Pinephone tells the modem to send a message to a phone number. After that, at `[458.0176]`, the modem indicates the message has been passed to the network (`0x02 0x04 0x00 0x00 0x00 0x00 0x00` in that same line ), but there's absolutely nothing after that, the modem doesn't seem to be getting a notification from the network that there's a pending message

To discard other stuff, can you disable LTE (set the peferred network type to 3G/2G) and try again? (If your carrier still supports non-lte, that is). Most likely scenario that I can imagine is that the network is trying to send the message over LTE and the modem is not getting it, but can't really know without data

hooger, 2022-05-16:
Thanks for the reply. I will try it, when I have some time. Question in the meantime: should I install 0.6.5 or test with the 0.6.4 version? I definitely would like to roll with the newer, just not sure if it helps or hinders the triage.

hooger, 2022-05-29:
So it seems to be, that when I switched to 2G/3G (preferred) it does works. I will be testing it some more and see what happens.

hooger, 2022-06-02:
So first I set the modem to use 3G/2G and under Manjaro Plasma I received every call and every sms, which was nice. However the setup crashed on me regularly so I am trying Mobian/Phosh to see if that is more stable, yet it still has the problem of not receiving SMS once I sent one. Modem fw is 0.6.4 so I sent you the logs via Matrix.
I will reflash 0.6.5 so the logs will be obfuscated and everything can be checked here, and also reflash the 6th ADPS stuff to see if that helps (you suggested it for the 4G problem).
Thanks for looking into it!