the-modem-distro/pinephone_modem_sdk #96: PPP Modem 0.6.4 No sound/Has sound = Restart/Power off

henrythemouse, 2022-05-10:
I have sound (ringing and voice) after a poweroff,  no sound after a restart. I'm using firmware 0.6.4. I'm calling my voicemail number to test for sound. I get the  results if I use the chatty 'call me' function.

The steps I used to repeat the test:

```
POWER OFF/ON vs RESTART
    1) restart ppp
    2) modem icon shows at lock screen (1-2 secs)
        no sound dialing vm
    3) ppp unplugged
        power off
    4) power on
        modem icon not up at lock screen
    5) modem icon shows after 5-8 secs
        got sound dialing vm

```
Has anyone seen this or a similar condition? 

Currently, my workaround it to never use restart, always use poweroff/poweron. Also, if I use poweroff/poweron with the PPP plugged into a charger, the result is the same as restarting.

Two sets of logs are attached, one taken after a poweroff/poweron, and the other taken after a restart.
[modemLogs.zip](https://github.com/Biktorgj/pinephone_modem_sdk/files/8664906/modemLogs.zip)

bobjrsenior, 2022-05-12:
I'm using Arch with Phosh on a pinephonepro and I'm getting the same issue.

Biktorgj, 2022-05-13:
It seems I might have reintroduced a bug where the audio codec isn't always reset at boot.
Please flash this kernel and tell me if this fixes it for you (My Pinephone Pro's SIM socket is broken so I'm a bit limited on the things I can try right now)

SHA512: 703f1c25b5bc4b252e581834f32359ff3bd19a7e4781e9c9266a10af3bfc4488deef8d51d0d9f55b81f5a4c4e9938eed3e77703231178915b4b395c6f11d6421
[test_kernel.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/8684212/test_kernel.tar.gz)

How to flash:
Unpack the tar file, place it somewhere in the Pinephone, and as root, run the following commands:
`echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2 && fastboot oem stay && fastboot flash boot mdm9607-perf-boot.img && fastboot reboot`

bobjrsenior, 2022-05-13:
I think the kernel you provided fixed the issue.

Details:

It shouldn't make a difference but due to permissions on /dev/ttyUSB2, I ran
```
sudo sh -c 'echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2' && fastboot oem stay && fastboot flash boot mdm9607-perf-boot.img && fastboot reboot
```

I ran the test from henrythemouse before and after flashing the kernel you provided.

Before flashing, I still had the issue.

After flashing, the issue seems to be resolved.
I was able to:

- Restart twice without the call audio messing up.
- poweroff/poweron with and without the phone charging without the call audio messing up.

henrythemouse, 2022-05-13:
I can confirm, this issue is resolved after flashing the img provided.
Nice, thanks.

closed 2022-05-15 by henrythemouse