the-modem-distro/pinephone_modem_sdk #98: PPP Modem could be overheating?

henrythemouse, 2022-05-15:
using uptodate mobian on a PPP

I've had the modem icon disappear 4 times in the last two days. I originally thought it was random, but then I realized that the back of the phone was hot, the top half more than the bottom half. This heat is unusual and takes 30-40min to cool down.  It gets hot when watching video streams over the net. Which is what I was doing at least 3 of these four times. The modem seems to turn off, anywhere in the video. The phone looses it's connection to the carrier, and the playback ends.

Any one else having these heat issues, and a lost modem?

```
# accordeing to Usage/Thermal Zones:
cpu-thermal      72 C
gup-thermal      72 C
rk818-battery    19C

# If I shut it down and let it cool (modem comes back):
cpu-thermal      52 C
gup-thermal      51 C
rk818-battery    19C

```
openqti.log and dmeg-modem.txt are empty, I guess because the modem was not available
[modemLost.zip](https://github.com/Biktorgj/pinephone_modem_sdk/files/8694381/modemLost.zip)

Biktorgj, 2022-05-15:
The modem is configured to automatically shut down if temperature exceeds 75 degrees, so if the phone was overheating it makes sense that it shut down automatically (the Pinephone Pro mainboard sits on top of the phone).
If you want to do some further testing, the firmware automatically logs temperature data to /var/log/thermal.log (is lost on reboot, but you could run adb shell tail -f /var/log/thermal.log to watch the temperature in almost-real-time)

Notice it's probable that the heat generated by the main CPU (if pushing it hard, with the LCD at max brightness etc.) can be heating up the modem on the other side of the board. I guess an easy test is to do the same thing you were doing but making sure you're using wifi (maybe with data disabled in the modem), and see if it shuts down that way. There's some room in the thermal limits, so I could push it a bit higher, but probably a better fix would be to try to avoid getting the phone _that_ hot in the first case

henrythemouse, 2022-05-15:
> If you want to do some further testing, the firmware automatically logs temperature data to /var/log/thermal.log (is lost on reboot, but you could run adb shell tail -f /var/log/thermal.log to watch the temperature in almost-real-time)

I confess that I don't know how to use adb, I just copy/paste. So, I ran the command verbatim and got:
```
sudo adb shell tail -f /var/log/thermal.log
error: insufficient permissions for device: user in plugdev group; are your udev rules wrong?
```
I googled this issue. One recommendation was to stop the adb server and restart it is using sudo:
```
adb kill-server
sudo adb start-server
```
That worked. I could run the tail command and other adb commands. Of course, I wonder if this is normal or I've messed something up somewhere in the past. ;-) 
```
sudo adb shell tail -f /var/log/thermal.log
Z0: 63C Z1: 63C Z2: 62C Z3: 62C Z4: 62C Z5: 58C
Z0: 62C Z1: 63C Z2: 62C Z3: 62C Z4: 62C Z5: 58C
Z0: 63C Z1: 64C Z2: 62C Z3: 62C Z4: 62C Z5: 58C
Z0: 63C Z1: 64C Z2: 63C Z3: 63C Z4: 63C Z5: 58C
...
```
The modem is running on the relatively cool above. Although, from what I've read on the pinephone forum, some people consider 50C to be normal. Here the phone has been on for a several hours, but essentially idle.

After approximately an hour of video playback  (modem on, wifi on, screen on) the modem turns off.
```
...
Z0: 71C Z1: 71C Z2: 70C Z3: 71C Z4: 71C Z5: 66C
Z0: 71C Z1: 72C Z2: 70C Z3: 71C Z4: 71C Z5: 66C
Z0: 72C Z1: 72C Z2: 71C Z3: 71C Z4: 71C Z5: 66C
Z0: 73C Z1: 76C Z2: 75C Z3: 74C Z4: 73C Z5: 66C # last line received

```
More repeatable than I had imagined it would be. I wouldn't want to run it any hotter. I understand that there is a way to throttle the cpu to help managed the over heating. I might try that. Perhaps I should take a close look at the processes that are running at idle. If others are running their ppp at 50C I must have something running that is causing extra heat to be generated.

Anyway, thanks for the suggestion. It's nice to see some actual numbers.

Biktorgj, 2022-05-16:
Have to make a correction, the modem is reset at 78ºC, not 75 as I mentioned before
Seeing those numbers and the modem ending up powering off (that's where probably the thermal zone 1 hit 78C), I would say some thermal limits or throttling should be adjusted in the Pinephone Pro, so it doesn't end up overheating like that.
The Rockchip CPU uses the entire LCD as a heat spreader, but seeing your numbers make me wonder if the throttling trigger is too high (or is not setup correctly), so when everything is on all the time for a long period of time this ends up happening.

Happy to see the emergency shutdown working as expected though, it's doing its job well :)
Test procedure should then be:
- Backlight at max
- I assume charging?
- Video streaming through wifi with everything on?

henrythemouse, 2022-05-16:
> * Backlight at max
> * I assume charging?
> * Video streaming through wifi with everything on?

Actually, I don't know what the backlight was set at. I wasn't watching the stream, but it was bright.
Yes, charging.
I assume the stream was using wifi, data was also enabled.
And I was connected to the phone with ssh, tailing the log on my laptop.

> The Rockchip CPU uses the entire LCD as a heat spreader,

I didn't know that, even though it makes sense, the entire screen does get hot.
This brings up a possible issue that I hadn't considered before. The PPP comes with a screen protector installed, so I'm assuming that pine64 doesn't see that as a problem when it comes to heat dissipation.  Unfortunately, I wasn't smart enough to see that when I opened my phone. Since the package included a separate screen protector I installed it. Not seeing the one that was already installed. Now the screen has two protectors on it. I personally don't have a problem with that, but do you think it might have an effect on heat dissipation?

Biktorgj, 2022-05-17:
> I didn't know that, even though it makes sense, the entire screen does get hot. This brings up a possible issue that I hadn't considered before. The PPP comes with a screen protector installed, so [...]

No, the plastic film won't make a difference at all, don't worry about that. I went stress-testing yesterday loosely based on what you reported:
- Youtube with a 10h long video playing
- LCD brightness at max
- Charging
- Modem enabled (without SIM as my PPP board is currently broken)
- Ambient temperature at 24C
- Phone sitting on top of a cardboard box (didn't want to leave it on top of a surface that would help dissipate the heat easily)

After about 3 hours the modem reached 73C in thermal zone 1. If ambient temperature would have been a bit higher, I could imagine reaching that temperature would have been easier
The phone was really hot, to the point that it was uncomfortable to touch, though it didn't reach any critical temperature levels (+95.0°C for the CPU).

I could move the reset threshold higher, by default Quectel starts trying to mitigate (by lowering cpu freq) at [105C](https://github.com/Biktorgj/quectel_eg25_kernel/blob/82ed00908b3e8efc3ff0de27d2b5a7c0524ecd7f/arch/arm/boot/dts/qcom/mdm9607.dtsi#L1191), which is [5 degrees below the example from Qualcomms device tree docs](https://github.com/Biktorgj/quectel_eg25_kernel/blob/82ed00908b3e8efc3ff0de27d2b5a7c0524ecd7f/Documentation/devicetree/bindings/arm/msm/msm_thermal.txt#L255). We can't do that here as we're already at 1/8 of Quectel's mitigation frequency all the time, we can't really go lower than 100MHz without things falling apart

I can move `thermal-reset-temp` to something higher so it has a little more room (so this doesn't bite in zones with a high temperature in summer), but it will still end up shutting down if you keep pushing it :)

henrythemouse, 2022-05-17:
> I will also add SMS notifications for overheating, so we get at least some warnings before things shut down

OK, well you're the coder and it sounds like you have a good grasp of what is happening. I can tell by touch when the phone is getting too hot for me. I can take the time to let it cool down or find a way to help it cool down (maybe a fan or the AC when in the car). If it were me, I wouldn't allow the temperature to get higher, but that's entirely up to you.

I'm going to close this issue because the operating temperature range for the modem has been established and it's been proven that it is functioning as it should. Perhaps controlling the overall temperature of the phone needs be addressed at some point, but that's a separate issue.

Thanks again for you help.

henrythemouse, 2022-05-18:
firmware  0.6.5  on latest mobian 

Just a follow up on this issue (may be related to this issue). I brought my PPP to a temperature where the modem turned off,  (running a media stream). I began to wonder, if I let the phone cool down (stopped the media stream), would the  modem turn back on. I was hoping that it would, but now that I think of it, that could be difficult to implement in a reasonable way.

I tried restarting ModemManager, and eg25-manager. The Modem stayed off.
[modemDown.txt](https://github.com/Biktorgj/pinephone_modem_sdk/files/8712190/modemDown.txt)

In addition, when I restarted the PPP, the modem did not come back. I had to power off the ppp and reboot it to get the modem back.

I've done this test twice.

Has anyone else seen this with regard to 0.6.5?

Biktorgj, 2022-05-18:
Since it's an emergency shutdown, it makes sense that it stays off until power cycled, especially given that (depending on distro), some parts of the phone aren't completely reset on a reboot.

What _should_ work in whatever scenario, is to use the kill-switch to do the power cycle. Turn first kill switch off, then on, and then restart eg25-manager to issue a power on.

Can you give that a try?

henrythemouse, 2022-05-18:
> Can you give that a try?

I'll try it, may take a day of so.

henrythemouse, 2022-05-20:
`Can you give that a try?`

I streamed media on youtube (firefox) for about 1hr 10min before the modem shut off.
I allowed the phone to cool down some.
I cycled the the modem kill switch (1).
I then restarted eg25-manager.
The modem icon did not immediately appear,
but withing  seconds the modem was back on.

Logs
[mobiantesting.zip](https://github.com/Biktorgj/pinephone_modem_sdk/files/8735575/mobiantesting.zip)

Biktorgj, 2022-05-21:
Okay, that log shows the modem panicked, but not sure if it's related to heat or if it has to do with #90, or if maybe #90 is actually related to heat too, and when it resets it goes straight to emergency download mode sometimes...

Biktorgj, 2022-05-21:
Just finished testing, it does in fact reboot to emergency download mode, at least sometimes:
```
[ 3394.779673] usb 2-1: USB disconnect, device number 6
[ 3394.780852] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[ 3394.780961] option 2-1:1.0: device disconnected
[ 3394.781853] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[ 3394.781946] option 2-1:1.1: device disconnected
[ 3394.785128] option1 ttyUSB2: GSM modem (1-port) converter now disconnected from ttyUSB2
[ 3394.785220] option 2-1:1.2: device disconnected
[ 3394.786071] option1 ttyUSB3: GSM modem (1-port) converter now disconnected from ttyUSB3
[ 3394.786171] option 2-1:1.3: device disconnected
[ 3394.788670] qmi_wwan 2-1:1.4 wwan0: unregister 'qmi_wwan' usb-1c1b000.usb-1, WWAN/QMI device
[ 3402.294732] usb 2-1: new high-speed USB device number 7 using ehci-platform
[ 3402.453481] option 2-1:1.0: GSM modem (1-port) converter detected
[ 3402.454152] usb 2-1: GSM modem (1-port) converter now attached to ttyUSB0
[ 3402.460271] udevd[7794]: error opening ATTR{/sys/devices/platform/soc/1c1b000.usb/usb2/2-1/power/wakeup} for writing: Permission denied
pine64-pinephone:/home/user# 
```
The thermal monitoring and reporting functions are set up already. I'm going to push those limits higher, and if the modem reaches about 85ºC I will try to gracefully shut it down so it doesn't look like a crash. We'll then retest #90 too as I wasn't able to reproduce but it seems to me that it wasn't a kernel panic but an overheating issue too

tiol11, 2022-05-21:
Thank you for quoting this issue in #90 . Today modem disappeared twice on my Pinephone 3GB, was about to upload logs but I am glad to see that troubleshooting has already gone deeper in finding root causes.

 @Biktorgj , are you going to prepare a (pre)release with higher thermal limit and SMS notification, in order to check the overheating hypothesis?

henrythemouse, 2022-05-21:
What  I have to report may have nothing specifically to do with the modem overheating, still it may be of interest.
Of course, maybe I'm not understanding the issues at hand and this test is not relevant.

All along I've been testing by streaming media online, but what happens when I play local media?
Online streaming is using wifi (if on) or mobile data. Neither is used for local media, so they are not contributing to the overall temperature in this test.

I set up my ppp to play a movie, directly from storage, offline.
Screen dimming off, screen blanking off, automatic suspend off.
Wifi off, bluetooth off.
Charging.
Mobile connection and data on.

Using the 'Usage' app I check the thermals at start:  51C.
I played the movie for about 30 min, before the mobile connection disappeared.
I noted that the thermals reported by 'Usage' were now at 72C.
The phone was hot to the touch, yet not as hot as in previous tests.
The 'heat zone' seemed be smaller as well.

What logs were available are attached.
[2022-05-21.zip](https://github.com/Biktorgj/pinephone_modem_sdk/files/8748200/2022-05-21.zip)

Biktorgj, 2022-05-23:
Probably not, but this might fix your issue anyway. When the modem hits a temperature limit, the cpu core resets itself forcing itself into emergency recovery mode. Doesn't matter if it's doing something, if idle, in recovery or without a SIM.

Here's a test build (meant to be flashed on top of what you're already running), that removes the core resetting, and adds thermal monitoring and reporting via SMS. I've set it up so it still will shut down, but at 85C instead of 78 as it was doing before, and will repeatedly notify you if things are getting too hot :)

It has three notification thresholds (rate limited so you don't get a message every five seconds):
At > 70C You will receive a warning telling it's heating up (one / 5min)
At > 80C comes the second warning, informing it will shutdown soon if it keeps heating up (one / 5min)
At > 85C last warning, it will shutdown no matter what you do. Will log the event, wait 30 seconds, then poweroff

At 85C it will do a normal poweroff, so it won't reboot into emergency recovery mode anymore, allowing you to just restart eg25-manager to start up again (if it's too hot when you restart it it will shut down again)

Let's see how this works! 

ebafac6342aa30b234ccf5733d8616c22615218e156068c59a3963d14e81d5fe338ddec9802043103c17e21e49d2a3a5c0b846a25feac04f15228435f3506094
[package-rootfsonly.tar.gz](https://github.com/Biktorgj/pinephone_modem_sdk/files/8751353/package-rootfsonly.tar.gz)

tiol11, 2022-05-23:
Thank you @Biktorgj !

Had to remove the two recovery lines inside flashall script, then it flashed fine.

> I'm at version 0.6.6-b0
> ADSP Version is EG25GGBR07A08M2G_01.003.01.003

Let's see how many temperature notification SMSs I will get in the next days... :)

henrythemouse, 2022-05-27:
Sorry about the delay. 

I flashed this file (also had to comment out the recovery lines). I restarted the phone.
I started a youtube podcast using firefox as before.
I wasn't using mobile data (had to switch my sim card to another phone).
I put in an old 'data only' card just for testing.

The sms reports came through as promised. I wasn't able to tail the log as I normally do, it didn't exists.
When the temp got to 86C I got the shutdown message. The mobile icon took a few seconds to disappear.
The playback of the podcast was not interrupted, 

I waited a few minutes just to let the temp go down a bit, then restarted eg25-manager.
The mobile icon took a few seconds to come back, but it did. 
As far as I could tell, everything was back to normal. 
I started getting sms messages again as the temp got to 70C.

The logic seems sound. Perhaps the force-off and restart could be automated. Then again I know when it's getting hot way before it gets to 86C, I should let it rest then. I'm thinking it should not get above 70C, which at this point in time does not seem practical. I can't recall ever feeling my Samsung get this hot to the touch. Maybe it's just destined to be a hot running phone.

Thanks for the opportunity to test this out.

Biktorgj, 2022-06-22:
@henrythemouse Temperature logging is now done in openqti, and follows the persistent logging configuration, so your temperature logs should be either in /var/log/openqti.log or in /persist/openqti.log depending if you have persistent logging enabled or not. For next release I will reduce the number of entries about it in the log as its flooding it a bit with a message every 5 seconds...

From your description everything is working as expected then. If it overheats it notifies until temperature is too high, then powers down itself, and leaves you in charge of turning it on again. Poweron should be about 14 seconds + 5 of waiting in fastboot

henrythemouse, 2022-07-01:
OK, looks like I can close this issue.

closed 2022-07-01 by henrythemouse