the-modem-distro/pinephone_modem_sdk #99: android-adp

pratriciap, 2022-05-17:
Hi:  This may not be the best place for my question but I am new to git hub and dont know my way around yet.
I have been trying to update the modem for a p pro using biktorgis flashall,  runing both mobian and manjaro from minicom.

  fllashall failed, as well as adp --help with a can not load shared libraries error. I tried two versions of adp, one a snap, and the other a manjaro package. Manjaro was missing libprotobuf.so.30, and the snap was missing libcrypto.so.1.0.0. Both of these libraries are antiquated and I could not find them anywhere.  Neither am I aware of any other available arm version of adp. 

 So far as I know adp must consist of an arm phone resident version. Could someone please tell me where I can get a version of android-adb which works on either manjaro or mobian,  how to get traction on this problem, or what I am missing.  What do you guys do that I dont know about ?

Biktorgj, 2022-05-18:
Hi!

If you're using Mobian, fastest way to get both `adb` and `fastboot` will be to simply run `apt install adb fastboot`
In Arch / Manjaro, this should install both applications too: `pacman -S android-tools`

Those will install all dependencies and the binaries you might need, no need to do anything else

Let me know if it works for you!

pratriciap, 2022-05-21:
Hi,  I tried both suggestions, manjara/pacman and mobian/apt early on and neither worked, but this time
      apt did install adb correctly in mobian,  many thanks!   Manjaro however, is still broken with  
root@manjaro-arm bin]#  adb --help

adb: error while loading shared libraries: libprotobuf.so.30: cannot open 
shared 
object file: No such file or directory

Name            : android-tools
Version         : 31.0.3-3

Mobian is less sastifactory to work on using minicom, because it dumps system messages to the screen. whereas manjaro does not.   By chance, do you know the switch to turn this off ?

Biktorgj, 2022-05-23:
Manjaro currently has some issue with the adb package, apparently an incorrect version if a library is installed with adb so it fails. We'll have to wait for them to fix that

About the serial port, I assume you are running `minicom` over the Pinephone serial port? By default most distros dump the console log there, so it will be constantly nagging. You can remove `console=...` parameter from `/boot/extlinux/extlinux.conf` and reboot to disable console logging, but maybe easier and more confortable to just start SSH and run minicom over ssh :)

pratriciap, 2022-05-29:
Modem flash attempt
 
 5-28-22
 
  Having obtained a good version of android-tools, adb, I attempted to flash it, but I think the 
modem got lunched from an earlier attempt that resulted  from trying to use flashall on manjaro, with a broken 
adb package. (I wish somebody would have told me)  

 Assumming a modem boot failure I went to your TROUBLE SHOOTING section and did
 the following:
 
 - disable hibernation, and remove back of phone.
 - fastboot oem stay & 
 - turn off, wait and turn on modem kill switch
 - systemctl restart eg25-manager
 
 The modem was now supposed to have been started in fastboot mode,
So, as I understand, follow through with MANUALLY FLASHING steps 6 ..8 

 At this point nothing I would expect worked as advertised.
 
 - lsusb  -> the Quctel listing,  not a ~xaimoi~ description as advertised
 - fastboot kept giving me  waiting for device messages

 - adb from kkeijezers script  while collecting journal log data ->
     * daemon started successfully
       error: no devices/emulators found
       error: no devices/emulators found
       adb: error: failed to get feature set: no devices/emulators found

I went through this a couple of times, and only invoked adb about the third time 
through to get journal data when nothing else made any sense

  Finally, so far as I know, there exists a valid ting card in the sim 
slot, and as I remember, there was a modem or mobile entry in the settings 
menu. Now I see no such thing, no way to get the imei num, nothing.
       
 The modem gets recognized as a USB device and the adb deamon starts, but
adb and fastboot fail to communicate with it. modem-manager gui fails
to indicate any devices.

 Am I missing something, or do I need to go down to the bottom of the sewer
with EDB and qfirehose ?  I am not sure what state of the universe I am in.

 I have attached a description for some of the journal messages, if per chance
you can make some sense out of them.
 
 I would really like to see the PPro through, but my time is limited. We are 
building a home on Orcas Island, and I only have about one day a week
available at home to do anything with it, So if I delay in getting 
back with you, it is not for lack of interest.

*************************************************************************************************************************

MODEM LOG INFO PINEPHONE PRO FLASH ATTEMPT
5-28-22 
 
these are taken from kkeijzer's script 
 
mobian/modem/messages.sh  (kkeijzers script)
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found
error: no devices/emulators found
adb: error: failed to get feature set: no devices/emulators found

 
 systemctl status eg25-manager
 
● eg25-manager.service - Quectel EG25 modem
[... 157 lines cut ...]
grab port ttyUSB3: Cannot add e
May 28 21:44:53 mobian ModemManager[638]: <warn>  [plugin/quectel] could not 
grab port ttyUSB2: Cannot add e
May 28 21:44:53 mobian ModemManager[638]: <warn>  [base-manager] couldn't 
create 
modem for device '/sys/devt
root@mobian:/home/mobian/modem# [ 1205.021096] rk818-bat: changed: dsoc=19, 
rsoc=50, v=3548, ov=3812 c=-1180
[ 1205.032958] rk818-bat: dl=48, rl=64, v=3733, halt=0, halt_n=2, max=0, 
init=0, 
sw=0, calib=0, below0=0, f0

sudo dmesg | grep modem                                         
                           
[    1.141474] cdc_acm: USB Abstract Control Model driver for USB modems and 
ISDN adapters
[   30.401717] usbserial: USB Serial support registered for GSM modem (1-port)
[   30.402791] option 1-1:1.0: GSM modem (1-port) converter detected
[   30.404010] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB0
[   30.405116] option 1-1:1.1: GSM modem (1-port) converter detected
[   30.409880] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB1
[   30.410999] option 1-1:1.2: GSM modem (1-port) converter detected
[   30.412138] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB2
[   30.413189] option 1-1:1.3: GSM modem (1-port) converter detected
[   30.419600] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB3

Biktorgj, 2022-06-01:
So, you never managed to get into fastboot mode, and I'm not really sure what's the state of the modem right now.

Are you still running stock? If you are, the `fastboot oem stay &` trick is not going to work (that is a custom firmware trick that the stock firmware doesn't have)

There are only two ways to enter fastboot mode from the stock firmware. One would be from ADB, but you need to unlock it and enable it, and the other, much faster way, is with an AT command.

From a root terminal, try to run this: `echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2`

If this works, after running `lsusb` you should see the fastboot device listed.
The modem doesn't connect anymore, probably because of the timeouts it's showing in ModemManager, but since I'm unsure if you managed to flash anything I'm unsure on which advice to give... If you never really managed to flash anything onto it, you can try to set power/control to `on` in the udev rules file as explained here: https://github.com/Biktorgj/pinephone_modem_sdk/blob/kirkstone/docs/SETTINGS.md#pinephone-pro

NOTE: If you're running stock, *only* change the `power/control` line, don't set `persist` to `1` as this breaks modem functionality on suspend

Hope it helps!

pratriciap, 2022-06-04:
OK,  so heres what I tried to do:

- power up on minicom, login as root
- dont logon through user display
- set 80-modem-eg25.rules   ...{power/control}="on"
                            ...{power/persist}="0"
- systemctl mask ... all suspend,  hibernation modes ..
- echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2
- lslusb  shows Bus 001 Device 002: ID 2c7c:0125 Quectel Wireless Solutions 
                Co.,Ltd. EC25 LTE modem  (no fastboot here)
                
  The eg25-manager log still shows  ... mobian eg25-manager[562]: 
Rescheduling upload since modem isn't online yet, in 30s     messages.

  modemmanager log stil shows:  mobian ModemManager[xxx]: transaction 0x28 
aborted, but message is not abortable         messages

- try  adb reboot bootloader
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found

- try adb shell reboot; fastboot oem stay
error: no devices/emulators found
< waiting for any device >  (requires ctrl-c abort)

 In-other-words, there were no changes since my last correspondence. The
 80-modem-eg25.rules setting changes seems to have had no effect.
 
 I cant tell what state the modem is in and if anything was flashed or 
not, but clearly, fastboot mode is not accessable.

 You did mention adb unlock and enable, is this something other than what I
have already done ? 

 Something got fumbled during my first attempt at flashing via your flashall
script, due either to a bungled adb package from manjaro or some combination
of the phone going into hibernation, crash or ctrl-c abort necessitating a
reboot while I was trying to figure things out.

 Any suggestions ??
 
BTW  no NON-HLOS.ubi file was included in your 6.5 package as there was with
     earlier ones.

pratriciap, 2022-06-04:
Ok, so here's what I tried to do:

- power up on minicom, login as root
- dont logon through user display
- set 80-modem-eg25.rules   ...{power/control}="on"
                            ...{power/persist}="0"
- systemctl mask ... all suspend,  hibernation modes ..
- echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2
- lslusb  shows Bus 001 Device 002: ID 2c7c:0125 Quectel Wireless Solutions 
                Co.,Ltd. EC25 LTE modem  (no fastboot here)
                
  The eg25-manager log still shows  ... mobian eg25-manager[562]: 
Rescheduling upload since modem isn't online yet, in 30s     messages.

  modemmanager log stil shows:  mobian ModemManager[xxx]: transaction 0x28 
aborted, but message is not abortable         messages

- try  adb reboot bootloader
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found

- try adb shell reboot; fastboot oem stay
error: no devices/emulators found
< waiting for any device >  (requires ctrl-c abort)

 In-other-words, there were no changes since my last correspondence. The
 80-modem-eg25.rules setting changes seems to have had no effect.
 
 I cant tell what state the modem is in and if anything was flashed or 
not, but clearly, fastboot mode is not accessable.

 You did mention adb unlock and enable, is this something other than what I
have already done ? 

 Something got fumbled during my first attempt at flashing via your flashall
script, due either to a bungled adb package from manjaro or some combination
of the phone going into hibernation, crash or ctrl-c abort necessitating a
reboot while I was trying to figure things out.

 Any suggestions ??
 
BTW  no NON-HLOS.ubi file was included in your 6.5 package as there was with
     earlier ones.

pratriciap, 2022-06-04:
Correction:
  sorry about the two entries:

NON-HLOS.ubi file was not included in package 6.5

fabertawe, 2022-06-25:
@pratriciap - did you get any further resolving this?

@Biktorgj - I'm in the same situation as pratriciap. I think I'm at the point of trying the "recovery" but how exactly do I short the two points? With something like a paperclip?

Biktorgj, 2022-06-26:
As I don't know in which situation each one of you are anymore, and I feel @pratriciap for some things mixed up, let's start from the beginning so I can understand where each of you are at:
1. Are you still running stock, or are you already using the custom firmware?
2. The modem still boots?

If any of the answers is true, you _shouldn't need_ to use the testpoint method to reflash anything.
 
Also, are you using a Pinephone or a Pinephone Pro?
If you open a root terminal, and you run `echo -ne "AT+QFASTBOOT\r\n" > /dev/ttyUSB2`, does the modem icon disappear from the topbar, at least for a few seconds?

fabertawe, 2022-06-26:
The reason I didn't start a new issue is because my logs were, as far as I could tell, identical to pratriciap! I have a Pinephone 1.2 (very recent purchase) running stock (.001). I only wanted to flash as the PP wiki suggests this.

I did read as much as I could regarding this before attempting it but I think I may have done things in the wrong order... because of the warning about having a recent enough baseband version before installing the firmware, I tried updating the ADSP first, to .003.

**echo -ne "AT+QFASTBOOT\r" > /dev/ttyUSB2** worked but **fastboot flash modem NON-HLOS.ubi && fastboot reboot** didn't. Apologies, I should have noted the exact error message but suffice to say, any attempt now at entering fastboot just says "waiting for device". The modem does not boot (no icon/greyed). I have tried all troubleshooting I can find, including reading all open and closed issues here. This is on Arch Phosh btw.

**lsusb**
`Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 2c7c:0125 Quectel Wireless Solutions Co., Ltd. EC25 LTE modem
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
**ls -la /dev**
`total 4
drwxr-xr-x  19 root  root       4320 Jun 26 12:46 .
drwxr-xr-x  17 root  root       4096 May  2 12:15 ..
lrwxrwxrwx   1 root  root         26 Jun 26 12:46 anbox-binder -> /dev/binderfs/anbox-binder
lrwxrwxrwx   1 root  root         28 Jun 26 12:46 anbox-hwbinder -> /dev/binderfs/anbox-hwbinder
lrwxrwxrwx   1 root  root         29 Jun 26 12:46 anbox-vndbinder -> /dev/binderfs/anbox-vndbinder
crw-r--r--   1 root  root    10, 235 Jun 26 12:46 autofs
lrwxrwxrwx   1 root  root         20 Jun 26 12:46 binder -> /dev/binderfs/binder
lrwxrwxrwx   1 root  root         28 Jun 26 12:46 binder-control -> /dev/binderfs/binder-control
drwxr-xr-x   3 root  root          0 Jun 26 12:46 binderfs
drwxr-xr-x   2 root  root        320 Jun 26 12:46 block
crw-------   1 root  root    10, 234 Jun 26 12:46 btrfs-control
drwxr-xr-x   3 root  root         60 Jun 26 12:45 bus
crw-------   1 root  root   180, 176 Jun 26 12:46 cdc-wdm0
drwxr-xr-x   2 root  root       3760 Jun 26 12:46 char
crw-------   1 root  root     5,   1 Jun 26 12:46 console
crw-------   1 root  root    10, 127 Jun 26 12:46 cpu_dma_latency
crw-------   1 root  root    10, 203 Jun 26 12:46 cuse
drwxr-xr-x   8 root  root        160 Jun 26 12:46 disk
drwxr-xr-x   3 root  root        120 Jun 26 12:46 dri
crw-rw----   1 root  video   29,   0 Jun 26 12:46 fb0
lrwxrwxrwx   1 root  root         13 Jun 26 12:46 fd -> /proc/self/fd
lrwxrwxrwx   1 root  root         22 Jun 26 12:46 features -> /dev/binderfs/features
crw-rw-rw-   1 root  root     1,   7 Jun 26 12:46 full
crw-rw-rw-   1 root  root    10, 229 Jun 26 12:46 fuse
crw-------   1 root  root   254,   0 Jun 26 12:46 gpiochip0
crw-------   1 root  root   254,   1 Jun 26 12:46 gpiochip1
drwxr-xr-x   2 root  root          0 Jun 26 12:46 hugepages
lrwxrwxrwx   1 root  root         22 Jun 26 12:46 hwbinder -> /dev/binderfs/hwbinder
crw-------   1 root  root    10, 183 Jun 26 12:46 hwrng
crw-------   1 root  root    89,   0 Jun 26 12:46 i2c-0
crw-------   1 root  root    89,   1 Jun 26 12:46 i2c-1
crw-------   1 root  root    89,   2 Jun 26 12:46 i2c-2
crw-------   1 root  root    89,   3 Jun 26 12:46 i2c-3
crw-------   1 root  root    89,   4 Jun 26 12:46 i2c-4
crw-------   1 root  root    89,   5 Jun 26 12:46 i2c-5
crw-------   1 root  root   250,   1 Jun 26 12:46 iio:device1
crw-------   1 root  root   250,   3 Jun 26 12:46 iio:device3
drwxr-xr-x   3 root  root        180 Jun 26 12:46 input
crw-r--r--   1 root  root     1,  11 Jun 26 12:46 kmsg
crw-rw-rw-   1 root  kvm     10, 232 Jun 26 12:46 kvm
lrwxrwxrwx   1 root  root         28 Jun 26 12:46 log -> /run/systemd/journal/dev-log
brw-rw----   1 root  disk     7,   0 Jun 26 12:46 loop0
brw-rw----   1 root  disk     7,   1 Jun 26 12:46 loop1
brw-rw----   1 root  disk     7,   2 Jun 26 12:46 loop2
brw-rw----   1 root  disk     7,   3 Jun 26 12:46 loop3
brw-rw----   1 root  disk     7,   4 Jun 26 12:46 loop4
brw-rw----   1 root  disk     7,   5 Jun 26 12:46 loop5
brw-rw----   1 root  disk     7,   6 Jun 26 12:46 loop6
brw-rw----   1 root  disk     7,   7 Jun 26 12:46 loop7
crw-rw----   1 root  disk    10, 237 Jun 26 12:46 loop-control
drwxr-xr-x   2 root  root         60 Jun 26 12:45 mapper
crw-rw----+  1 root  video  251,   0 Jun 26 12:46 media0
crw-rw----+  1 root  video  251,   1 Jun 26 12:46 media1
crw-r-----   1 root  kmem     1,   1 Jun 26 12:46 mem
brw-rw----   1 root  disk   179,   0 Jun 26 12:46 mmcblk2
[... 186 lines cut ...]

**systemctl status eg25-manager**
`● eg25-manager.service - Quectel EG25 modem
     Loaded: loaded (]8;;file://pine64/usr/lib/systemd/system/eg25-manager.service/usr/lib/systemd/system/eg25-manager.service]8;;; enabled; vendor preset: disabled)
     Active: active (running) since Sun 2022-06-26 12:46:05 BST; 2min 13s ago
   Main PID: 2921 (eg25-manager)
      Tasks: 3 (limit: 3273)
     Memory: 7.8M
        CPU: 202ms
     CGroup: /system.slice/eg25-manager.service
             └─2921 /usr/bin/eg25-manager

Jun 26 12:46:05 pine64 eg25-manager[2921]: Starting modem...
Jun 26 12:46:06 pine64 eg25-manager[2921]: Executed power-on/off sequence
Jun 26 12:46:06 pine64 eg25-manager[2921]: taking systemd sleep inhibitor
[... cut ...]
Jun 26 12:46:07 pine64 eg25-manager[2921]: ModemManager appeared on D-Bus
Jun 26 12:46:35 pine64 eg25-manager[2921]: Rescheduling upload since modem isn't online yet, in 30s
Jun 26 12:47:04 pine64 eg25-manager[2921]: Rescheduling upload since modem isn't online yet, in 30s
Jun 26 12:47:34 pine64 eg25-manager[2921]: Rescheduling upload since modem isn't online yet, in 30s
Jun 26 12:48:04 pine64 eg25-manager[2921]: Rescheduling upload since modem isn't online yet, in 30s
`

If I can supply anything else then please advise, thanks.

fabertawe, 2022-06-28:
I went ahead and shorted the pins and carried out the recovery. I successfully booted into EDL mode (lsusb showing "(QDL mode)") on my first attempt (phew!) but firehosing failed a few times...

```
[000.000]: QFirehose Version: Quectel_LTE&5G_QFirehose_Linux&Android_V1.2.3
[000.000]: Builded: Oct 21 2020 19:13:31
[000.000]: Find md5 check file <./md5.txt>
[000.001]: md5 checking: ./contents.xml pass
[000.014]: md5 checking: ./update/appsboot.mbn pass
[000.017]: md5 checking: ./update/ENPRG9x07.mbn pass
[000.018]: md5 checking: ./update/firehose/partition_complete_p2K_b128K.mbn pass
[000.018]: md5 checking: ./update/firehose/patch_p2K_b128K.xml pass
[000.022]: md5 checking: ./update/firehose/prog_nand_firehose_9x07.mbn pass
[000.022]: md5 checking: ./update/firehose/rawprogram_nand_p2K_b128K_update.xml pass
[000.198]: md5 checking: ./update/mdm9607-boot.img pass
[000.488]: md5 checking: ./update/mdm9607-recovery.ubi pass
[002.056]: md5 checking: ./update/mdm9607-sysfs.ubi pass
[003.213]: md5 checking: ./update/NON-HLOS.ubi pass
[003.216]: md5 checking: ./update/NPRG9x07.mbn pass
[003.217]: md5 checking: ./update/partition.mbn pass
[003.217]: md5 checking: ./update/partition_nand.xml pass
[003.221]: md5 checking: ./update/rpm.mbn pass
[003.228]: md5 checking: ./update/sbl1.mbn pass
[003.243]: md5 checking: ./update/tz.mbn pass
[003.303]: md5 checking: ./update/usrdata.ubi pass
[003.303]: Totals checking 18 files md5 value, 0 file fail!
[003.303]: [1] /sys/bus/usb/devices/2-1 5c6/9008/0
[003.304]: P: /dev/bus/usb/002/002 idVendor=05c6 idProduct=9008
[003.304]: C: /dev/bus/usb/002/002 bNumInterfaces: 1
[003.304]: I: If#= 0 Alt= 0 #EPs= 2 Cls=ff Sub=ff Prot=ff
[003.304]: E: Ad=81 Atr=02 MxPS= 512 Ivl=0ms
[003.304]: E: Ad=01 Atr=02 MxPS= 512 Ivl=0ms
[003.304]: usbfs_is_kernel_driver_alive ioctl USBDEVFS_GETDRIVER failed, kernel driver may be inactive
[003.304]: dir=./update/firehose
[003.304]: d_name=prog_nand_firehose_9x07.mbn
[003.304]: prog_nand_firehose_filename = prog_nand_firehose_9x07.mbn
[003.304]: STATE <-- SAHARA_WAIT_HELLO
[003.308]: inf[0] ep_in -1/4096, errno = 71 (Protocol error)
[003.309]: qusb_noblock_read read=-1, errno: 71 (Protocol error)
[003.309]: qusb_noblock_read cur=0, min_size=1
[003.313]: qusb_noblock_write write=-1, errno: 71 (Protocol error)
[003.313]: qusb_noblock_write cur=0, min_size=1
[003.317]: inf[0] ep_in -1/4096, errno = 71 (Protocol error)
[003.317]: qusb_noblock_read read=-1, errno: 71 (Protocol error)
[003.317]: qusb_noblock_read cur=0, min_size=1
[003.317]: Sahara protocol error
[003.317]: Upgrade module failed.
```

But after toggling the modem hardware switch and restarting eg25-manager.service, the update succeeded!

I'm now running .002. Thanks.