the-modem-distro/pinephone_modem_sdk #101: Upgraded to 0.6.5, resulting in loss of service

domrc, 2022-05-21:
When I upgraded from 0.6.4 to 0.6.5 I lost cell service (Calling, SMS, and Data would not work). I tried seeing if different ADSP versions would affect service, but they did not (side not, I made sure that the modem was using this firmware with `AT+QGMR` & `AT+GETADSPVER`). Reverting to the stable release according to LVFS, 0.6.1, resolved this issue. Just want to know if anyone else experiences this issue.

Biktorgj, 2022-05-21:
First report in that. I've seen sometimes ModemManager getting confused after the first reboot when applying a firmware update (any firmware update) and needing sometimes a service restart, but not the issue you're mentioning.

As far as I know nothing _that_ important changed between 0.6.4 and 0.6.5 to cause something like that to happen.
Any chance you can update again to 0.6.5 and get some logs?

domrc, 2022-05-21:
Okay, I tried upgrading from 0.6.1 to 0.6.4 and got the same result. Unfortunately, Manjaro ARM made it so I couldn't use adb to get the logs, as adb on Arch ARM variants apparently [isn't supported](https://xnux.eu/devices/feature/modem-pp.html#toc-unlock-adb-access).

So instead, due to this and other unrelated issues I've been having with Manjaro ARM, I switched to Mobian. Before switching to Mobian, I reverted to 0.6.1 to test and make sure Mobian would work with the modem. After installation and setting up the recommended settings for the modem, I installed `fastboot` and `adb` and preceded to upgrade to 0.6.5. Switching OS's had no affect, but I found out that I cannot access adb from the modem, after issuing the `AT+ADBON` command and with a working adb binary. I looked at what `AT+QGMR` to confirm that I installed the correct version. It reported back "EG25GGBR07A08M2G_01.003.00.000".

I do remember upgrading to the ADSP to 01.003 before upgrading the modem firmware to 0.6.5, for the first time. At first it did work, and I was able to test out the new service number +22 33 44 55 66 77. But then I decide to try the `set user name` command via SMS to customize what the modem says on call. Not long after that was when the modem would just stop giving any sort of service, while still being visible (you can see the cell signal icon on Phosh, but SMS and calling doesn't work)

hooger, 2022-05-22:
I've hit the same issue. The problem is that the current version of android-tools depends on libprotobuf 3.19, while currently 3.20 is installed. There is some compilation issue that is why android-tools is behind. It is an Arch ARM issue. That is what Strit said to me in the #manjaro-arm-public:matrix.org matrix room.
Links:
https://gitlab.manjaro.org/manjaro-arm/packages/temp/android-tools/-/jobs/8645
https://github.com/nmeum/android-tools/pull/61

Biktorgj, 2022-05-23:
> 

> But then I decide to try the `set user name` command via SMS to customize what the modem says on call. Not long after that was when the modem would just stop giving any sort of service, while still being visible (you can see the cell signal icon on Phosh, but SMS and calling doesn't work)

Unfortunately, I'd need to see the log to be able to figure out what's going on... I can imagine some error while parsing the config file maybe, but can't know for sure (maybe delete `/persist/openqti.conf` and reboot?), 
In theory, openqti should recreate the config file if it's unable to parse part of its contents, but maybe certain characters don't make it happy...