the-modem-distro/pinephone_modem_sdk #112: Make open firmware identifyable in udev

MartijnBraam, 2022-06-14:
For adding the recommended settings udev rules: https://github.com/Biktorgj/pinephone_modem_sdk/blob/honister/docs/SETTINGS.md it would be nice to have a identifier that works there so that rule only gets executed when this firmware is installed.

For example setting the usb serial number so that can be matched on

Biktorgj, 2022-06-14:
That's a great idea!
Vendor and product IDs I can't change without breaking ModemManager and fwupd, but the serial number I could set whatever we need and it's literally a one line change in a script

Is it possible to set udev rules for a vendor+product+serial combination?

MartijnBraam, 2022-06-14:
Yes you can in theory match on anything exposed as an USB data field

Biktorgj, 2022-06-15:
Much better :)
```
udevadm info --attribute-walk /sys/bus/usb-serial/devices/ttyUSB0 |grep -i serial
    SUBSYSTEM=="usb-serial"
    ATTRS{serial}=="c0ffee7ab1e"
    ATTRS{serial}=="fe3c0000.usb"
```

Biktorgj, 2022-06-16:
Okay:
https://github.com/Biktorgj/meta-qcom/commit/2a52f9d41f5a283cf458e5d6d32d5c3e5fafa0e3
From 0.6.7 onwards, the modem will report `community_fw` as a serial number.

If I find some time today I will post a 0.6.7 release with the changes baked in so we can try it :)

Biktorgj, 2022-06-16:
New release with the change included: https://github.com/Biktorgj/pinephone_modem_sdk/releases/tag/0.6.7

MartijnBraam, 2022-06-19:
Nice,  this is supported in postmarketOS now :)

closed 2022-06-19 by MartijnBraam